bb-qq/r8152 #92: DS920+ success

imtv, 2021-01-27:
DS920+ ，
version: DSM 6.2.3-25426 Update 3
Equipment: UNITEK usb 2.5G U316A
Driver: r8152-geminilake-2.14.0-1.spk
After installing spk, there is no need to restart, first disable and then enable the driver, you can see it under the network.
He cannot use DHCP to get the ip address, so I assign it manually.

Joeygu79121, 2021-01-28:
Have you ever try reboot? Is it still there after reboot?

imtv, 2021-01-29:
After restarting, you still need to disable the driver, and then restart the driver.
You can see it under the network.

bb-qq, 2021-02-23:
Could you try to increase the delay of loading the driver that is written in /var/packages/r8152/scripts/start-stop-status?
"sleep 5" is it.

Joeygu79121, 2021-02-23:
> Could you try to increase the delay of loading the driver that is written in /var/packages/r8152/scripts/start-stop-status?
> "sleep 5" is it.

Thank you!

imtv, 2021-06-02:
1.use r8152-geminilake-2.14.0-3.spk
2.written in /var/packages/r8152/scripts/start-stop-status ，change "sleep 5" to "sleep 20"
 Can be used perfectly

closed 2021-06-02 by imtv