bb-qq/r8152 #93: DS918+ was nowork，whenplug in the Ethernet cable，Will missing the card

lamyanchau, 2021-01-27:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2021-02-23:
Please fill the template. Due to the lack of information, I am unable to investigate the problem.

bb-qq, 2021-05-29:
I close this issue.

closed 2021-05-29 by bb-qq