bb-qq/r8152 #96: 420j is good to work with minor issues, manual assigned IP after reboot

James-Zhu-CA, 2021-02-02:
### Description of the problem
I have install it in 420j with the latest firmware 6.3 update3. With one problem that I need to use ssh to login the NAS and assigned a IP to it with ifconfig. But every time I reboot the machine, I need to assigned the IP again, can any one teach me how to let it run the command automatically? Thanks

> Replace this line with your answer.

bb-qq, 2021-02-23:
Did you try to re-install the driver?
If re-install does not work for you, please try to increase the delay of loading driver that is written in /var/packages/r8152/scripts/start-stop-status. "sleep 5" is it.

bb-qq, 2021-05-29:
DHCP bug was fixed in release 2.14.0-3. Please feel free to re-open the issue if you still have the issue.

closed 2021-05-29 by bb-qq