bb-qq/r8152 #97: Remove the network card and dmesg report trace

akikaede6, 2021-02-03:
### Description of the problem

I install this driver with my kernel, and cp it to /lib/modules/4.4.15-deepin-wutip-421/kernel/drivers/net/usb/, so that I can use it after reboot, but I found that when i remove this network card, dmesg will report trace, but I don't know what will it cause, maybe it will cause something bad?
PS. network card can be used normally, and this message was also reported when I didn't connect to cable.

### Description of your products
kernel version: 4.4.15
USB device: biaze RTS8153
chip: RTL8153
Network port type: RJ45
### Description of your environment
PC->USB to network
NIC name:
```
06:00.2 Ethernet controller: Intel Corporation I350 Gigabit Network Connection (rev 01)
```
and this is USB controller:
```
09:00.0 USB controller: Renesas Technology Corp. uPD720201 USB 3.0 Host Controller (rev 03)
```
### Output of ``dmesg`` command
this is dmesg after I insert and remove network card
```
[ 4028.960000] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[ 4029.104000] usb 2-1: reset SuperSpeed USB device number 4 using xhci_hcd
[ 4029.164000] r8152 2-1:1.0 eth2: v2.13.0 (2020/04/20)
[ 4029.164000] r8152 2-1:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.
               
[ 4029.296000] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[ 4029.324000] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[ 4030.896000] r8152 2-1:1.0 eth2: Stop submitting intr, status -85
[ 4031.200000] usb 2-1: USB disconnect, device number 4
[ 4031.200000] ------------[ cut here ]------------
[ 4031.200000] WARNING: CPU: 2 PID: 13294 at fs/sysfs/group.c:237 sysfs_remove_group+0x104/0x130()
[ 4031.200000] sysfs group ffffffe000bd2e80 not found for kobject '2-1:1.0'
[ 4031.200000] Modules linked in: fuse snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_hwdep snd_pcm r8152(O) snd_timer snd soundcore mii bq27xxx_battery binfmt_misc ip_tables x_tables ipv6 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx
[ 4031.200000] CPU: 2 PID: 13294 Comm: kworker/2:0 Tainted: G        W  O    4.4.15-deepin-wutip-421 #655
[ 4031.200000] Workqueue: usb_hub_wq hub_event
[ 4031.200000] fffffc01e78b3a28 ffffffff8171b44c ffffffff81713fdd fffffc01e78b3aa8 
                      ffffffff8093e7c4 fffffc01e78b3aa8 0000000000000009 ffffffff81725b40 
                      ffffffff8093e870 0000000000000000 ffffffe000bd2e80 fffffc01e7c79c40 
                      fffffc01d7cd2490 fffffc01e7c79c30 ffffffff80b0c204 0000000080000001 
                      ffffffff81725bb2 fffffc01e78b3af8 fffffc0100000018 fffffc00da600c30 
                      ffffffff80b05c5c ffffffff81968ed8 fffffc01f96b6348 ffffffe000bd2e80 
[ 4031.200000] Trace:
[ 4031.200000] [<ffffffff8093e7c4>] warn_slowpath_common+0xc4/0x120
[ 4031.200000] [<ffffffff8093e870>] warn_slowpath_fmt+0x50/0x70
[ 4031.200000] [<ffffffff80b0c204>] sysfs_remove_group+0x104/0x130
[ 4031.200000] [<ffffffff80b05c5c>] kernfs_find_and_get_ns+0x5c/0xb0
[ 4031.200000] [<ffffffff80b054d8>] kernfs_put+0x158/0x290
[ 4031.200000] [<ffffffff80b0c140>] sysfs_remove_group+0x40/0x130
[ 4031.200000] [<ffffffff80b054d8>] kernfs_put+0x158/0x290
[ 4031.200000] [<ffffffe000b8e154>] rtl8152_disconnect+0x34/0x100 [r8152]
[ 4031.200000] [<ffffffff811fb24c>] usb_unbind_interface+0x7c/0x280
[ 4031.200000] [<ffffffff81059f08>] __device_release_driver+0xb8/0x160
[ 4031.200000] [<ffffffff81059ff0>] device_release_driver+0x40/0x70
[ 4031.200000] [<ffffffff810593a4>] bus_remove_device+0x154/0x1d0
[ 4031.200000] [<ffffffff810544f8>] device_del+0x178/0x310
[ 4031.200000] [<ffffffff811f8d80>] usb_disable_device+0xd0/0x2d0
[ 4031.200000] [<ffffffff811ee088>] usb_disconnect+0xf8/0x350
[ 4031.200000] [<ffffffff811ef280>] hub_event+0x470/0x1470
[ 4031.200000] [<ffffffff8095c55c>] process_one_work+0x24c/0x540
[ 4031.200000] [<ffffffff8095c9bc>] worker_thread+0x16c/0x6b0
[ 4031.200000] [<ffffffff8095ca78>] worker_thread+0x228/0x6b0
[ 4031.200000] [<ffffffff80964d38>] kthread+0x148/0x160
[ 4031.200000] [<ffffffff8095c850>] worker_thread+0x0/0x6b0
[ 4031.200000] [<ffffffff80911508>] ret_from_kernel_thread+0x18/0x20
[ 4031.200000] [<ffffffff80964bf0>] kthread+0x0/0x160
[ 4031.200000] [<ffffffff80964bf0>] kthread+0x0/0x160

[ 4031.200000] ---[ end trace 3ccbda04e487488f ]---
```

### Output of ``lsusb -t`` command
**Bus 002 Device 007: ID 0bda:8153 Realtek Semiconductor Corp.**  is my network card
````
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
**Bus 002 Device 007: ID 0bda:8153 Realtek Semiconductor Corp.** 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID ff00:00ff  
Bus 001 Device 002: ID 0d8c:0012 C-Media Electronics, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
````

### Output of ``ifconfig -a`` command
eth2 is my network card
````
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.2.11.106  netmask 255.255.255.0  broadcast 10.2.11.255
        inet6 fe80::5ab4:1ab5:fec8:57c9  prefixlen 64  scopeid 0x20<link>
        ether 6c:92:bf:07:5f:c8  txqueuelen 1000  (Ethernet)
        RX packets 118370  bytes 7597925 (7.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 125684  bytes 171762298 (163.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0xc0e0300000-c0e031ffff  

eth1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 6c:92:bf:07:5f:c9  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0xc0e0320000-c0e033ffff  

eth2: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:e0:4c:68:01:49  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 278  bytes 28124 (27.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 278  bytes 28124 (27.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
````

akikaede6, 2021-02-03:
I used driver 2.14.0 in https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software and there was no trace when I remove network card, maybe it fixed in 2.14.0?

bb-qq, 2021-02-20:
I don't know why this message is being output. As far as I can tell from searching the web, there is nothing wrong with ignoring it, and if it is fixed with the latest driver, then I don't think you need to worry about it.

However, Synology's system may not expect the network interface to be detached while it's running, so I think it's safer to power off and detach it.

akikaede6, 2021-02-21:
Thank you for your reply！I'm glad to slove this issue，I'll close it because I have already find the new driver。

closed 2021-02-21 by akikaede6