bb-qq/r8152 #98: DS918+ failed to copy data to Raid5 disk, but working fine to copy to basic disk

rickmjli, 2021-02-05:
When I copy my file (about 400gb data) from my iMac to my NAS (DS918+), 
-- if the destination folder is in a basic drive, it working well. 
-- if the destination folder is in a Raid5 drive, it will failed after copied about 10gb data. message showed "The operation can't be completed because an unexpected error occurred (error code -50).  I have tried many times, always failed with same msg.

Note: my DS918+ is using 4 disks:
-- 3 disks are set as raid 5 with data protection
-- 1 disk is set as basic, without data protection

### Description of your products

> DS 918+, using latest driver r8152-apollolake-2.14.0-2.spk
>  DSM 6.2.3-25426 update 3
>  ugreen USB3.0 to 2.5g

### Description of your environment

> - iMac and DS918+ are connected to TP-LINK 2.5G switch (TL-SH1008)
> - PC: iMac (macOS 10.15.7), with ugreen usb-c to 2.5g ethernet adapter
> - cable  Cat.7

this is the screenshot which copied file to basic disk
[image]

this is the screenshot which copied file to raid5 disk
[image]

bb-qq, 2021-02-20:
* Can you use SSH to get the output of dmesg when the problem occurs?
* Does the same problem occur if you copy files from another PC such as Windows?

bb-qq, 2021-05-16:
Also using the front ports may fix the error.

I would like to close this issue since I can not get a response. Please feel free to re-open the issue if you wish to continue conversations.

closed 2021-05-16 by bb-qq