bb-qq/r8152 #101: Success ! Confirmed working on Synology DS620slim with CableCreation 2.5gbe USB-C

jjjleung, 2021-02-14:
Hi, 

I got 2x CableCreation 2.5gbe USB-C CD0674 dongle, hooked up MacbookAir M1, and Synology DS620slim, both are working fine, R/W speed is >230MB/s.

One little thing to note, the dongle is USB-C, so I used an adapter to connect to USB-A at back of the NAS. It turns out not all adapters are equal, I had to try a few to finally get it working with the one came with Sandisk Extreme SSD. Also, the dongle has to be connected to NAS during startup for the driver(v.2.14.0-2) to load properly.

Thanks for the great work, my NAS speed has doubled now. If I get a chance to try the CableCreation USB-A dongle, will update.

Dongle used : 
https://www.amazon.com/CableCreation-Gigabit-Ethernet-Compatible-Chromebook/dp/B07VQJ468M/ref=sr_1_4?dchild=1&keywords=cablecreation+usb+c+2.5gbe&qid=1613281846&s=electronics&sr=1-4

Sjekke, 2021-02-14:
I'm using the https://plugable.com/products/usbc-e2500 the USB-A to USB-C adapter is included

jjjleung, 2021-02-15:
> I'm using the https://plugable.com/products/usbc-e2500 the USB-A to USB-C adapter is included

Thanks for the tip, I switched to the Plugable one as well, works the 1st time I plugged it in. NAS, M1, even my old 2012 Mac Mini(did install the Realtek driver though). The CableCreation one is unstable, sometimes dropping link, even only negotiate to 1G LAN. Plugable one is much better.

closed 2021-02-15 by jjjleung