bb-qq/r8152 #103: DS216play with DSM 6.2.3 update 3 didn't always work

Jacky71wu, 2021-02-21:
### Description of the problem
Plug 2.5gbe to DS216play read USB 3.0 port (DSM 6.2.3-25426 Update 3) sometimes DSM will show 2.5gbe network. sometime won't
> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
Linux SWW-NAS 3.10.105 #25426 SMP Tue Nov 24 17:15:05 CST 2020 armv7l GNU/Linux synology_monaco_ds216play
>     - DSM version x.x
DSM 6.2.3-25426 Update 3
> - the product name of the ethernet dongle
USB 10/100/1G/2.5G LAN - Realtek Semiconductor Corp.
### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
Windows 10
>     - Hub: model name
QSW-1105-5T
> - cable category (Cat.5e or Cat.6 or Cat.7)
CAT 5E
### Output of ``dmesg`` command
````
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 3.10.105 (root@build1) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25426 SMP Tue Nov 24 17:15:05 CST 2020
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: STi SoC with Flattened Device Tree, model: STiH410 B2120
[    0.000000] Ignoring memory below PHYS_OFFSET: 0x40000000-0x43000000
[    0.000000] Ignoring memory below PHYS_OFFSET: 0x42000000-0x43000000
[    0.000000] cma: CMA: reserved 16 MiB at 5f000000
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 118784
[    0.000000] free_area_init_node: node 0, pgdat 806e0000, node_mem_map 8073d000
[    0.000000]   Normal zone: 928 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 118784 pages, LIFO batch:31
[    0.000000] PERCPU: Embedded 7 pages/cpu @80b2b000 s6976 r8192 d13504 u32768
[    0.000000] pcpu-alloc: s6976 r8192 d13504 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 117856
[    0.000000] Kernel command line: console=ttyS0,115200 CONSOLE=/dev/ttyS0 ip=off mtdparts=stm-nand-bch.0:0x500000(boot),-(rfs) bpa2parts=BPA2_Region|aud-coded|aud-transcoded|vid-coded|vid-transcoded|vid-raw-input|vid-output-0|vid-decimated-1|vid-copied-1|vid-macroblock-0|vid-extra-data-0|vid-extra-data-1|v4l2-coded-video-buffers|vid-enc-data-0|vid-enc-scaled-0|vid-encoded-0|v4l2-clut|v4l2-grab|gfx-memory|aud-codec-data|vid-codec-data|aud-output-0|aud-output-1|vid-output-1|vid-decimated-0|vid-copied-0|vid-macroblock-1|blitter|vid-enc-data-1|vid-enc-scaled-1|vid-encoded-1|coredisplay-video|multicom|hades-l3|BPA2_Region0|BPA2_Region1:480M:0x62000000,v4l2-vb2|v4l2-stmc8jpg|v4l2-stmvout|v4l2-compo:24M:0x60800000,coprocs:32M:0x40000000,modules|audio:8M:0x60000000,bigphysarea|te-buffers|m2m-ca3:48M:0x48000000,STMCReserve:32768:0x47FF8000 vmalloc=775m mem=503316480@0x42000000 root=/dev/md0 rw syno_hw_version=DS216play hd_power_on_seq=2 ihd_num=2 netif_num=1 flash_size=8 phys_memsize=1024 syno_me
[    0.000000] bpa2: partition 'BPA2_Region' created at 0x62000000 (high), size 491520 kB (0x1e000000 B)
[    0.000000] bpa2: partition 'v4l2-vb2' created at 0x60800000 (high), size 24576 kB (0x01800000 B)
[    0.000000] bpa2: partition 'coprocs' created at 0x40000000 (high), size 32768 kB (0x02000000 B)
[    0.000000] bpa2: partition 'modules' created at 0x60000000 (high), size 8192 kB (0x00800000 B)
[    0.000000] bpa2: partition 'bigphysarea' created at 0x48000000 (low), size 49152 kB (0x03000000 B)
[    0.000000] bpa2: partition 'STMCReserve' created at 0x47ff8000 (low), size 32 kB (0x00008000 B)
[    0.000000] Synology Hardware Version: DS216play-j
[    0.000000] Internal HD num: 2
[    0.000000] Internal netif num: 1
[    0.000000] SYNO Transcoding Memory Mode: 3
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 464MB = 464MB total
[    0.000000] Memory: 393560k/393560k available, 81576k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
                   vmalloc : 0x9d800000 - 0xff000000   (1560 MB)
                   lowmem  : 0x80000000 - 0x9d000000   ( 464 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80667be4   (6527 kB)
                     .init : 0x80668000 - 0x806a2b40   ( 235 kB)
                     .data : 0x806a4000 - 0x806e0ac0   ( 243 kB)
                      .bss : 0x806e0ac0 - 0x8073cd24   ( 369 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[... 971 lines cut ...]
  |__4-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)

````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:54:3E:37
          inet addr:192.168.1.102  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:151067 errors:0 dropped:0 overruns:0 frame:0
          TX packets:142222 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:72814177 (69.4 MiB)  TX bytes:33114755 (31.5 MiB)
          Interrupt:130

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3582 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3582 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:484391 (473.0 KiB)  TX bytes:484391 (473.0 KiB)
````

bb-qq, 2021-02-23:
Unfortunately, this model does not have enough memory to load the driver.

This error indicates that an allocation of 256kb contiguous physical memory was failed due to fragmentation or other reason.
```
[  314.825187] ifconfig: page allocation failure: order:4, mode:0x40d0
```

Uninstalling or stopping some applications/services might help to resolve the issue.

bb-qq, 2021-05-29:
I close this issue. Please feel free to re-open the issue if you have another question.

closed 2021-05-29 by bb-qq