bb-qq/r8152 #104: Sabrent NT-S25G Unstable on DS416Play

deckard2123, 2021-03-01:
### Description of the problem

When I have sustained data transfers going (10+ minutes in duration) the connection seems to drop and I'm unable to reconnect to the 2.5G dongle without rebooting the DS416Play.  When the connection to the 2.5G dongle drops, I can still connect to the DS416Play through the built in 1G NIC, but I can no longer connect to the Sabrent NT-S25G USB NIC.  The outputs listed below were obtained by connecting to the 1G NIC after the 2.5 G dongle appeared to have crashed during a long file transfer.  Please help me identify what is causing the connection drop/crash.  I've noticed the dongle gets quite hot to the touch and I wonder if its a heat dissipation issue.

### Description of your products

NAS Model: DS416Play
output of ``uname -a`` command
Linux DS416Play 3.10.105 #25426 SMP Mon Dec 14 18:45:24 CST 2020 x86_64 GNU/Linux synology_braswell_416play
DSM version: DSM 6.2.3-25426 Update 3 
Dongle: Sabrent USB to 2.5 Gigabit Ethernet Adapter [10/100/1000/2500] (NT-S25G)

### Description of your environment

The dongle is connected to one of the rear USB ports on the DS416Play.  I conducted this test over WiFi to test the stability of the Sabrent dongle attached to the NAS.  Not trying to achieve 2.5 Gbps speeds here.  Previous tests run using iperf3 over the wired network achieved expected speeds.

### Output of ``dmesg`` command

````
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.105 (root@build4) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25426 SMP Mon Dec 14 18:45:24 CST 2020
[    0.000000] Command line: root=/dev/md0 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 acpi_enforce_resources=lax libata.force=5:3.0G syno_hw_version=DS416play vender_format_version=2 syno_hdd_detect=56,59,63,57 syno_hdd_enable=61,60,71,58 syno_usb_vbus_gpio=93@0000:00:14.0@2 sn=1771NMN185200 macs=00113279cd69,00113279cd6a
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007a4befff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x7bffe014  SMBIOS=0x7a850000 
[    0.000000] efi: mem00: type=3, attr=0xf, range=[0x0000000000000000-0x0000000000001000) (0MB)
[    0.000000] efi: mem01: type=7, attr=0xf, range=[0x0000000000001000-0x000000000006f000) (0MB)
[    0.000000] efi: mem02: type=10, attr=0xf, range=[0x000000000006f000-0x0000000000070000) (0MB)
[    0.000000] efi: mem03: type=7, attr=0xf, range=[0x0000000000070000-0x0000000000086000) (0MB)
[    0.000000] efi: mem04: type=0, attr=0xf, range=[0x0000000000086000-0x0000000000088000) (0MB)
[    0.000000] efi: mem05: type=6, attr=0x800000000000000f, range=[0x0000000000088000-0x00000000000a0000) (0MB)
[    0.000000] efi: mem06: type=2, attr=0xf, range=[0x0000000000100000-0x0000000000e60000) (13MB)
[    0.000000] efi: mem07: type=7, attr=0xf, range=[0x0000000000e60000-0x00000000374ce000) (870MB)
[    0.000000] efi: mem08: type=2, attr=0xf, range=[0x00000000374ce000-0x0000000037a5f000) (5MB)
[    0.000000] efi: mem09: type=7, attr=0xf, range=[0x0000000037a5f000-0x00000000533fc000) (441MB)
[    0.000000] efi: mem10: type=2, attr=0xf, range=[0x00000000533fc000-0x000000007127e000) (478MB)
[    0.000000] efi: mem11: type=4, attr=0xf, range=[0x000000007127e000-0x000000007129e000) (0MB)
[    0.000000] efi: mem12: type=7, attr=0xf, range=[0x000000007129e000-0x0000000074256000) (47MB)
[    0.000000] efi: mem13: type=4, attr=0xf, range=[0x0000000074256000-0x00000000752b0000) (16MB)
[    0.000000] efi: mem14: type=7, attr=0xf, range=[0x00000000752b0000-0x000000007541e000) (1MB)
[    0.000000] efi: mem15: type=1, attr=0xf, range=[0x000000007541e000-0x00000000754bf000) (0MB)
[    0.000000] efi: mem16: type=7, attr=0xf, range=[0x00000000754bf000-0x0000000077fb0000) (42MB)
[    0.000000] efi: mem17: type=4, attr=0xf, range=[0x0000000077fb0000-0x0000000077fb1000) (0MB)
[    0.000000] efi: mem18: type=7, attr=0xf, range=[0x0000000077fb1000-0x0000000078050000) (0MB)
[    0.000000] efi: mem19: type=4, attr=0xf, range=[0x0000000078050000-0x00000000794bf000) (20MB)
[    0.000000] efi: mem20: type=7, attr=0xf, range=[0x00000000794bf000-0x000000007a092000) (11MB)
[    0.000000] efi: mem21: type=2, attr=0xf, range=[0x000000007a092000-0x000000007a09b000) (0MB)
[    0.000000] efi: mem22: type=3, attr=0xf, range=[0x000000007a09b000-0x000000007a4bf000) (4MB)
[    0.000000] efi: mem23: type=5, attr=0x800000000000000f, range=[0x000000007a4bf000-0x000000007a6bf000) (2MB)
[    0.000000] efi: mem24: type=6, attr=0x800000000000000f, range=[0x000000007a6bf000-0x000000007aabf000) (4MB)
[    0.000000] efi: mem25: type=0, attr=0xf, range=[0x000000007aabf000-0x000000007aebf000) (4MB)
[    0.000000] efi: mem26: type=10, attr=0xf, range=[0x000000007aebf000-0x000000007bfbf000) (17MB)
[... 1665 lines cut ...]
          inet addr:192.168.2.2  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe79:cd69/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8598338 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2464028 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:9841609419 (9.1 GiB)  TX bytes:36410319791 (33.9 GiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:79:CD:6A  
          inet addr:169.254.88.211  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::211:32ff:fe79:cd6a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11252 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:2377408 (2.2 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

jjjleung, 2021-03-02:
Did you try stop/re-run the driver from DSM ? Would it work ?

bb-qq, 2021-03-05:
Did you try the front port? Some users reported stability issues when they use rear ports.

deckard2123, 2021-03-07:
Moving the adapter to the front USB port, seems to have solved the stability issue.  Thanks!

closed 2021-03-07 by deckard2123