bb-qq/r8152 #106: Sabrent 2.5 Gbps Adapter Inaccessible on Boot

deckard2123, 2021-03-07:
### Description of the problem

Whenever the DS416Play boots up, the Sabrent 2.5 Gbps Adapter is listed as one of the network interfaces, however it can't be reached via the network or pinged until I switch the adapter's IPv4 setting in DSM from "Use manual configuration" to "Get network configuration automatically (DHCP) and then back to "Use manual configuration.

On boot, the fields for Gateway and DNS Server are are always blank when "Use manual configuration" is selected prior to a reboot.  If I fill these fields in manually without clicking "Get network configuration automatically (DHCP)" the Sabrent adapter is still unreachable on the network.  It doesn't become reachable until I select "Get network configuration automatically (DHCP)" click ok, and then switch it back to "use manual configuration" and click ok that the network adapter becomes reachable.  When switching the adapter to "Get network configuration automatically (DHCP)" DSM fills in the Gateway and DNS Server fields with default IP addresses that work for my network.

Keeping the adapter configured to "Get network configuration automatically (DHCP)" when rebooting doesn't solve the issue.  When rebooting with this configuration, the network adapter remains unresponsive on the network following the reboot.

Is there a workaround for this or a startup script could be created to have DSM automatically cycle through these settings (DHCP off--> DHCP on--> DHCP off) on startup?

Note: dmesg, lsusb, and ifconfig commands were done immediately after rebooting the DS416Play and connecting via SSH through the built in 1 Gbps ethernet port. 

### Description of your products

NAS Model: DS416Play
output of uname -a command
Linux DS416Play 3.10.105 #25426 SMP Mon Dec 14 18:45:24 CST 2020 x86_64 GNU/Linux synology_braswell_416play
DSM version: DSM 6.2.3-25426 Update 3
Dongle: Sabrent USB to 2.5 Gigabit Ethernet Adapter [10/100/1000/2500] (NT-S25G)

### Description of your environment

The dongle is connected to the front USB port on the DS416Play.
- PC: Intel I225-V built onto motherboard
- Hub: QNAP QSW-1105-5T
- Cable category: Cat 6

### Output of ``dmesg`` command

````
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.105 (root@build4) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25426 SMP Mon Dec 14 18:45:24 CST 2020
[    0.000000] Command line: root=/dev/md0 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 acpi_enforce_resources=lax libata.force=5:3.0G syno_hw_version=DS416play vender_format_version=2 syno_hdd_detect=56,59,63,57 syno_hdd_enable=61,60,71,58 syno_usb_vbus_gpio=93@0000:00:14.0@2 KPTI_on sn=1771NMN185200 macs=00113279cd69,00113279cd6a
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007a4befff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x7bffe014  SMBIOS=0x7a850000
[    0.000000] efi: mem00: type=3, attr=0xf, range=[0x0000000000000000-0x0000000000001000) (0MB)
[    0.000000] efi: mem01: type=7, attr=0xf, range=[0x0000000000001000-0x000000000006f000) (0MB)
[    0.000000] efi: mem02: type=10, attr=0xf, range=[0x000000000006f000-0x0000000000070000) (0MB)
[    0.000000] efi: mem03: type=7, attr=0xf, range=[0x0000000000070000-0x0000000000086000) (0MB)
[    0.000000] efi: mem04: type=0, attr=0xf, range=[0x0000000000086000-0x0000000000088000) (0MB)
[    0.000000] efi: mem05: type=6, attr=0x800000000000000f, range=[0x0000000000088000-0x00000000000a0000) (0MB)
[    0.000000] efi: mem06: type=2, attr=0xf, range=[0x0000000000100000-0x0000000000e60000) (13MB)
[    0.000000] efi: mem07: type=7, attr=0xf, range=[0x0000000000e60000-0x00000000374ce000) (870MB)
[    0.000000] efi: mem08: type=2, attr=0xf, range=[0x00000000374ce000-0x0000000037a5f000) (5MB)
[    0.000000] efi: mem09: type=7, attr=0xf, range=[0x0000000037a5f000-0x00000000533fc000) (441MB)
[    0.000000] efi: mem10: type=2, attr=0xf, range=[0x00000000533fc000-0x000000007127e000) (478MB)
[    0.000000] efi: mem11: type=4, attr=0xf, range=[0x000000007127e000-0x000000007129e000) (0MB)
[    0.000000] efi: mem12: type=7, attr=0xf, range=[0x000000007129e000-0x0000000074256000) (47MB)
[    0.000000] efi: mem13: type=4, attr=0xf, range=[0x0000000074256000-0x00000000752b0000) (16MB)
[    0.000000] efi: mem14: type=7, attr=0xf, range=[0x00000000752b0000-0x000000007541e000) (1MB)
[    0.000000] efi: mem15: type=1, attr=0xf, range=[0x000000007541e000-0x00000000754bf000) (0MB)
[... 916 lines cut ...]

eth2      Link encap:Ethernet  HWaddr 00:24:27:88:0C:01
          inet6 addr: fe80::224:27ff:fe88:c01/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:153 errors:0 dropped:0 overruns:0 frame:0
          TX packets:41 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:19547 (19.0 KiB)  TX bytes:12292 (12.0 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:652 errors:0 dropped:0 overruns:0 frame:0
          TX packets:652 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:86371 (84.3 KiB)  TX bytes:86371 (84.3 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

chenlei630, 2021-03-11:
This situation happens also on my NAS 918 with ugreen 2.5G. I muss change the ip address of the nas, then it could be connected with my computer.

bb-qq, 2021-03-21:
Could you try this workaround?
https://github.com/bb-qq/r8152/issues/92#issuecomment-783863386

deckard2123, 2021-03-21:
> Could you try this workaround?
> [#92 (comment)](https://github.com/bb-qq/r8152/issues/92#issuecomment-783863386)

I tried changing "sleep 5" to sleep 7, 10, and 20.  Each time following reboot, the 2.5 Gbps adapter was still inaccessible until I switched it to DHCP and back to manual in DSM.

Sjekke, 2021-03-21:
On my 916+ I use a fixed IP and via Task Scheduler following commands 

ifconfig eth2 up
synonet –dhcp eth2

Everything works fine with this after startup...

deckard2123, 2021-03-21:
> On my 916+ I use a fixed IP and via Task Scheduler following commands
> 
> ifconfig eth2 up
> synonet –dhcp eth2
> 
> Everything works fine with this after startup...

Is this how you have it configured in Task scheduler @Sjekke?  Screenshots are below.  I tried rebooting the DS416Play with this enabled and the 2.5 Gbps adapter set to a static ip address.  I still couldn't access the 2.5 Gbps adapter until switching DHCP on and then back off. 

[image]

[image]

Sjekke, 2021-03-21:
Hi @deckard2123 indeed like that. But I use a Plugable 2.5 adapter.

Gensplejs, 2021-05-13:
I submitted a pull request that should fix the dhcp issue.

bb-qq, 2021-05-16:
I created the 2.14-3 including the fix of that PR.

Gensplejs, 2021-05-21:
2.14-3 Works fine for me on my DS918+. But then again it would be nice if someone else could verify that it fixes the problem.

Garfield999, 2021-06-12:
By the way: if you want to assign a static ip, add this in synology taskmanager - without it, the ip in syno settings was not assigned for me with ds918+ after reboot:

taskmanager (user root, boot-up)

synonet --set_ip -4 eth2 add 192.168.178.3/24

bb-qq, 2021-07-11:
I close this issue. If you are still having problems, please feel free to reopen this issue.

closed 2021-07-11 by bb-qq