bb-qq/r8152 #109: DS1821+ not working

joejohnson1701, 2021-03-12:
ds1821 
Cannot start after installation

Sjekke, 2021-03-12:
> ds1821 
> 
> Cannot start after installation

You hope to have reply? 

What cannot start? You ethernet card, your NAS, your driver? 
What???

joejohnson1701, 2021-03-12:
> > ds1821
> > 安装后无法启动
> 
> 您希望得到答复？
> 
> 什么不能开始？您的以太网卡，您的NAS，您的驱动程序？
> 什么？？？

r8152-v1000-2.14.0-2a.spk
After installing the package center, the r8152 driver cannot be started

joejohnson1701, 2021-03-12:
> > ds1821
> > Cannot start after installation
> 
> You hope to have a reply?
> 
> What cannot start? You ethernet card, your NAS, your driver?
> What???

r8152-v1000-2.14.0-2a.spk
After installing the package center, the r8152 driver cannot be started

bertybassett, 2021-04-10:
the applications is running on my 1821+ however I am still awaiting delivery of my card.

bb-qq, 2021-05-16:
This loding issues was fixed in the 2.14.0-2b release. Please create a new issue if that has another problem.

closed 2021-05-16 by bb-qq