bb-qq/r8152 #111: DS1817USB LAN card cannot be used in VMM

a4771107, 2021-03-20:
[image]

Is there any way I can use it in VMM?

bb-qq, 2021-03-21:
This is a limitation of the driver currently. The workaround is written here.
https://github.com/bb-qq/r8152/issues/90#issue-794013966

a4771107, 2021-03-21:
I solved it, but it still doesn't work on reboot.
But I am satisfied enough
Many Korean friends are getting help
Thank you to the faceless hero

a4771107, 2021-03-21:
https://cafe.naver.com/synologynas
Korea exchanges information here
For many members here you are hope

bb-qq, 2022-01-22:
The latest driver does support the Open vSwitch(VMM) feature. Please try it. If you have any issues, create a new issue.

closed 2022-01-22 by bb-qq