bb-qq/r8152 #112: DS216+ instable with Club3d 2.5Gbit driver.

EvertKlevert, 2021-04-05:
### Description of the problem

I got the driver and 2.5Gb usb interface installed and working, but it's not stable. 
After a few MB of file transfer, data rate goes to 0. 
DSM sometimes removes the interface. 
Can't figure out what's the problem.

### Description of your products

DS216+
upgraded tot 8GB ram
Intel Celeron N3050
Club3d 2.5Gbit USB Club 3D USB 3.2 Gen1 adapter type A naar 2,5 Gigabit Ethernet ST./BU. zwart, **CAC-1420** (Amazon)
DSM 6.2.3-25426 update #68 
used driver: bb-qq 2.14.0-2   r8152-braswell-2.14.0-2.spk from https://github.com/bb-qq/r8152/releases

### Description of your environment
both NAS and PC connected through a 2.5Gbit QNAP QSW-1105-5T
Cat 5e
PC with Realteak 2.5Gb gaming Family controller

### Output of ``dmesg`` command
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.105 (root@build4) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25426 SMP Mon Dec 14 18:45:24 CST 2020
[    0.000000] Command line: root=/dev/md0 ihd_num=2 netif_num=1 HddHotplug=1 SataPortMap=21 DiskIdxMap=0002 acpi_enforce_resources=lax syno_castrated_xhc=0000:00:14.0@0x0c syno_hw_version=DS216+ vender_format_version=2 syno_hdd_detect=56,59,63,57 syno_hdd_enable=61,60,71,58 sn=1620NUN811304 macs=00113258afd4
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007a4befff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x7bffe014  SMBIOS=0x7a850000
[    0.000000] efi: mem00: type=3, attr=0xf, range=[0x0000000000000000-0x0000000000001000) (0MB)
[    0.000000] efi: mem01: type=7, attr=0xf, range=[0x0000000000001000-0x000000000006f000) (0MB)
[    0.000000] efi: mem02: type=10, attr=0xf, range=[0x000000000006f000-0x0000000000070000) (0MB)
[    0.000000] efi: mem03: type=7, attr=0xf, range=[0x0000000000070000-0x0000000000086000) (0MB)
[    0.000000] efi: mem04: type=0, attr=0xf, range=[0x0000000000086000-0x0000000000088000) (0MB)
[    0.000000] efi: mem05: type=6, attr=0x800000000000000f, range=[0x0000000000088000-0x00000000000a0000) (0MB)
[    0.000000] efi: mem06: type=2, attr=0xf, range=[0x0000000000100000-0x0000000000e60000) (13MB)
[    0.000000] efi: mem07: type=7, attr=0xf, range=[0x0000000000e60000-0x00000000374d0000) (870MB)
[    0.000000] efi: mem08: type=2, attr=0xf, range=[0x00000000374d0000-0x0000000037a60000) (5MB)
[    0.000000] efi: mem09: type=7, attr=0xf, range=[0x0000000037a60000-0x00000000533d6000) (441MB)
[    0.000000] efi: mem10: type=2, attr=0xf, range=[0x00000000533d6000-0x000000007127e000) (478MB)
[    0.000000] efi: mem11: type=4, attr=0xf, range=[0x000000007127e000-0x000000007129e000) (0MB)
[    0.000000] efi: mem12: type=7, attr=0xf, range=[0x000000007129e000-0x0000000074256000) (47MB)
[    0.000000] efi: mem13: type=4, attr=0xf, range=[0x0000000074256000-0x00000000752b0000) (16MB)
[    0.000000] efi: mem14: type=7, attr=0xf, range=[0x00000000752b0000-0x0000000075427000) (1MB)
[    0.000000] efi: mem15: type=1, attr=0xf, range=[0x0000000075427000-0x00000000754bf000) (0MB)
[    0.000000] efi: mem16: type=7, attr=0xf, range=[0x00000000754bf000-0x00000000780dc000) (44MB)
[    0.000000] efi: mem17: type=4, attr=0xf, range=[0x00000000780dc000-0x00000000794bf000) (19MB)
[    0.000000] efi: mem18: type=7, attr=0xf, range=[0x00000000794bf000-0x000000007a092000) (11MB)
[    0.000000] efi: mem19: type=2, attr=0xf, range=[0x000000007a092000-0x000000007a09b000) (0MB)
[    0.000000] efi: mem20: type=3, attr=0xf, range=[0x000000007a09b000-0x000000007a4bf000) (4MB)
[    0.000000] efi: mem21: type=5, attr=0x800000000000000f, range=[0x000000007a4bf000-0x000000007a6bf000) (2MB)
[    0.000000] efi: mem22: type=6, attr=0x800000000000000f, range=[0x000000007a6bf000-0x000000007aabf000) (4MB)
[    0.000000] efi: mem23: type=0, attr=0xf, range=[0x000000007aabf000-0x000000007aebf000) (4MB)
[... 1067 lines cut ...]

eth1      Link encap:Ethernet  HWaddr A0:CE:C8:E1:E3:AF
          inet addr:192.168.192.123  Bcast:192.168.192.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fee1:e3af/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:114158 errors:0 dropped:0 overruns:0 frame:0
          TX packets:171726 errors:73 dropped:33 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:74504429 (71.0 MiB)  TX bytes:236763953 (225.7 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:30829 errors:0 dropped:0 overruns:0 frame:0
          TX packets:30829 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8130097 (7.7 MiB)  TX bytes:8130097 (7.7 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Sjekke, 2021-04-05:
It seems that Club 3D doesn't function very well. I use the Plugable 2.5gb on a DS916+ ... it works flawlessly and it transmitting continuously because I use a camera with Surveillance Station. Can't you swap the adapter?

EvertKlevert, 2021-04-05:
Sorry, don't quite understand you. The Club3d _is_ plugable (usb). 

What kind of adapter do you mean?

Sjekke, 2021-04-12:
Inderdaad 😃

Raychan87, 2021-04-27:
ince you have a similar problem like me and also a similar CPU try the patch "r8152-braswell-2.12.0-3.spk" with this it runs first. But it must still make long-term test whether it works.

Raychan87, 2021-04-28:
In case you still have the problem: https://github.com/bb-qq/r8152/issues/114#issuecomment-828359849

Knoddel, 2022-10-16:
With my Ds216+ with 8GB Ram, the Club 3D USB 3.2 Gen1 Adapter Type-A to 2.5 Gigabit Ethernet works without any problems. The data throughput remains constant.

bb-qq, 2022-12-04:
@EvertKlevert, sorry I haven't been able to reply.

The problem seems to be caused by your environment, as there are other reports of it working. From your logs, it seems that you have many USB devices attached, so disconnecting all USB devices except the ethernet dongle may improve the symptoms.

bb-qq, 2024-06-24:
I would like to close this issue as I have not heard back from you in a long time.
If you have any questions, feel free to re-open.

closed 2024-06-24 by bb-qq