bb-qq/r8152 #114: DS916+ with Club 3D 2.5GB with r8152-braswell-2.14.0-2.spk is instabil

Raychan87, 2021-04-26:
### Description of the problem

I installed the driver "r8152-braswell-2.14.0-2.spk" for the Club 3D 2.5Gbit module, but it runs very unstable.
When writing several 60MB files the transfer rate goes to 0.
When reading data everything is fine.

### Description of your products

DS916+ (DSM 6.2.4-25556)
- 8GB Ram
- Intel Pentium N3710
- Club3d 2.5Gbit USB Club 3D USB 3.2 Gen1 adapter type A naar 2,5 Gigabit Ethernet ST./BU. zwart, CAC-1420 (Amazon)
- r8152-braswell-2.14.0-2.spk
- MTU 1500

Rechner:
MSI MEG Unify X570
- Realtek PCIe 2.5GbE

### Description of your environment

I have a MSI MEG Unify X570 withRealtek PCIe 2.5GbE connected directly to the Club 3D 2.5GB USB3.2 Gen1 via a Cat7 cable.

### Output of ``dmesg`` command

gnal
[100890.140058] init: iscsi_pluginengined main process (18437) killed by TERM si                                                                                                gnal
[100890.188988] init: scsi_plugin_server main process (18457) killed by TERM sig                                                                                                nal
[100893.190189] init: iscsi_pluginserverd main process (19252) killed by TERM si                                                                                                gnal
[100893.201089] init: iscsi_pluginengined main process (19250) killed by TERM si                                                                                                gnal
[100893.251642] init: scsi_plugin_server main process (19259) killed by TERM sig                                                                                                nal
[100932.357707] r8152 4-2:1.0 eth2: carrier off
[100933.938303] init: scsi_plugin_server main process (19978) killed by TERM sig                                                                                                nal
[101881.770293] r8152 4-2:1.0 eth2: carrier on
[101883.410098] init: scsi_plugin_server main process (20342) killed by TERM sig                                                                                                nal
[101885.368940] init: iscsi_pluginserverd main process (22115) killed by TERM si                                                                                                gnal
[101885.393012] init: iscsi_pluginengined main process (22106) killed by TERM si                                                                                                gnal
[101885.458143] init: scsi_plugin_server main process (22142) killed by TERM sig                                                                                                nal
[101888.414410] init: iscsi_pluginserverd main process (22883) killed by TERM si                                                                                                gnal
[101888.426760] init: iscsi_pluginengined main process (22879) killed by TERM si                                                                                                gnal
[101888.470928] init: scsi_plugin_server main process (22897) killed by TERM sig                                                                                                nal
[101897.132277] r8152 4-2:1.0 eth2: carrier off
[101898.724374] init: scsi_plugin_server main process (23345) killed by TERM sig                                                                                                nal
[101900.200730] r8152 4-2:1.0 eth2: carrier on
[101901.840764] init: scsi_plugin_server main process (23898) killed by TERM sig                                                                                                nal
[101904.001815] init: iscsi_pluginserverd main process (24276) killed by TERM si                                                                                                gnal
[101904.013784] init: iscsi_pluginengined main process (24268) killed by TERM si                                                                                                gnal
[101904.081492] init: scsi_plugin_server main process (24296) killed by TERM sig                                                                                                nal
[101904.590479] init: nmbd main process (25099) killed by TERM signal
[101906.964662] init: iscsi_pluginserverd main process (25092) killed by TERM si                                                                                                gnal
[101906.975510] init: iscsi_pluginengined main process (25089) killed by TERM si                                                                                                gnal
[101907.011630] init: scsi_plugin_server main process (25102) killed by TERM sig                                                                                                nal
[102265.863393] r8152 4-2:1.0 eth2: get_registers -71
[102278.877055] r8152 4-2:1.0 eth2: get_registers -71
[102291.888670] r8152 4-2:1.0 eth2: get_registers -71
[102332.923378] r8152 4-2:1.0 eth2: get_registers -71
[102346.935787] r8152 4-2:1.0 eth2: get_registers -71
[102373.957502] r8152 4-2:1.0 eth2: get_registers -71
[102413.992901] r8152 4-2:1.0 eth2: get_registers -71
[102441.015812] r8152 4-2:1.0 eth2: get_registers -71
[102455.027072] r8152 4-2:1.0 eth2: get_registers -71
[102468.276295] r8152 4-2:1.0 eth2: get_registers -71
[102590.143870] r8152 4-2:1.0 eth2: get_registers -71
[102644.195667] r8152 4-2:1.0 eth2: get_registers -71
[102711.256149] r8152 4-2:1.0 eth2: get_registers -71
[102725.269185] r8152 4-2:1.0 eth2: get_registers -71
[102752.291130] r8152 4-2:1.0 eth2: get_registers -71
[102766.303381] r8152 4-2:1.0 eth2: get_registers -71
[102778.313952] r8152 4-2:1.0 eth2: get_registers -71
[102806.337632] r8152 4-2:1.0 eth2: get_registers -71
[... 1960 lines cut ...]
          RX bytes:12495932589 (11.6 GiB)  TX bytes:2860856473 (2.6 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:383191 errors:0 dropped:0 overruns:0 frame:0
          TX packets:383191 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:246032611 (234.6 MiB)  TX bytes:246032611 (234.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.10.10.1  P-t-P:10.10.10.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Sjekke, 2021-04-26:
I have the Plugable Usb 2.5gbit adapter on 916+. I have no issues.

Raychan87, 2021-04-26:
Which DSM version do you have? 
What driver version do you have? 
Do you also have MTU on 1500?

Sjekke, 2021-04-26:
I'm running DSM DSM 6.2.4-25556
Driver 2.14.0-2
MTU 1500

Sjekke, 2021-04-26:
https://plugable.com/products/usbc-e2500

Raychan87, 2021-04-26:
Then I don't understand what's wrong with my DS916
I have a lot of Tx errors and dropped on the net wed card. :( 
But without this 2.5gb adapter, it runs flawlessly. Is the adapter defective?

Sjekke, 2021-04-26:
You are the first with problems with this adapter https://github.com/bb-qq/r8152/issues/112

Defective... I can't say ... incompatible... maybe. Can you return it?

Raychan87, 2021-04-27:
I have now tested a lot and found that it works with an old driver "r8152-braswell-2.12.0-3.spk". Maybe there is a bug in the new version?

Sjekke, 2021-04-27:
> I have now tested a lot and found that it works with an old driver "r8152-braswell-2.12.0-3.spk". 
> 
> And I have set the MTU to 9000, as well as enabled the jumbo frame (9014 bytes) 

Good news!

Raychan87, 2021-04-28:
I have now done a major test with 1500 Raw photos imported into Lightroom and converted to DNG and it works.

eth2      Link encap:Ethernet  HWaddr A0:CE:C8:E1:E3:55
          inet addr:192.168.170.1  Bcast:192.168.170.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fee1:e355/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:43004940 errors:0 dropped:0 overruns:0 frame:0
          TX packets:23604721 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:342367038420 (318.8 GiB)  TX bytes:99465515654 (92.6 GiB)

Important:
1. install the firmware "r8152-braswell-2.12.0-3.spk".
2. plug in Club3d USB-RJ45 2.5GB adapter (front USB port) and connect with cat 7 cable to PC
3. set IP and MTU to 9000
4. reboot DS916+

Also set MTU 9000 and jumbo frame (9014 bytes) on the computer network card and reboot the computer.

bb-qq, 2021-05-16:
I released 2.14-3 that includes the capability of showing the chip revision of the device.
I appreciate it if you paste dmesg logs with this version using working devices and not working devices.

deveth0, 2021-06-03:
hey, thanks for posting this bug, saved me a lot of debugging.
I ran into the same issue on a 918+ with the Club3 thing, I had the last release installed.

now with 2.15.0-1 sadly the transfer still crashes after a few seconds, weird enough I don#t see any of the stack traces menthoned above.

dmesg:
```
[  120.011923] r8152 2-1:1.0 eth2: v2.15.0 (2021/04/15)
[  120.020610] r8152 2-1:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[  120.062169] r8152 2-1:1.0 eth2: chip rev 12
```

bb-qq, 2021-07-11:
Thanks for posting the info on the chip revision of the Club 3D device. As the generation of DS918 can run xHCI devices stable　if front ports are used, so the different chip revisions may be causing problems.

Can someone tell me the chip revision of the Pluggable one?

Raychan87, 2021-07-11:
I wanted to report after a long time.

I have since 28.04.21 no disconnections or other problems. But still use the old driver version "r8152-braswell-2.12.0-3.spk". 
The new driver I will only test on DSM 7.0, if the old driver makes problems. Never chance a running system :)

About the chip revision. How can I find out the chip revision?

Raychan87, 2021-07-11:
I have not had any disconnects or other problems with the driver version "r8152-braswell-2.12.0-3.spk" since 28.04.

I have now installed the latest version "r8152-braswell-2.15.0-1.spk" for testing to tell you the chip revision.

dmesg:
`
[622815.278916] netif_napi_add() called with weight 256 on device eth%d
[622815.293328] r8152 2-2:1.0 eth2: v2.15.0 (2021/04/15)
[622815.299054] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.

[622815.316759] r8152 2-2:1.0 eth2: chip rev 12
[622815.321733] usbcore: registered new interface driver r8152
[622825.377683] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[622825.602048] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[622825.611243] r8152 2-2:1.0 eth2: carrier on
`

Raychan87, 2021-07-11:
The current driver "r8152-braswell-2.15.0-1.spk" is not suitable for the DS916+. I switched back to the old driver "r8152-braswell-2.12.0-3.spk", it runs perfectly.

- write speed does not go higher than 60MB/s (read speed up to 300MB/s)
- Network connection freezes after a few hours. (No log entries)

I don't know what the problem is.

Sjekke, 2021-07-11:
Are you on DSM 7?

Raychan87, 2021-07-11:
no. I have DSM 6.2.4-25556

Sjekke, 2021-07-11:
Ah Ok. I already moved to DSM 7.0 and use the DSM 7.0 drivers with a Plugable 2.5Gbit USD dongle. Till know works perfectly

Raychan87, 2021-07-11:
Also ist der Treiber "r8152-braswell-2.15.0-1.spk" nur für DSM 7 gedacht? Das wusste ich nicht. 

Ich werde erst auf DSM 7 wechseln, wenn DSM 7 final ist.

Sjekke, 2021-07-11:
> So the driver "r8152-braswell-2.15.0-1.spk" is only for DSM 7? I did not know that. 
> 
> 
> 
> I will only switch to DSM 7 when DSM 7 is final.

DSM 7.0 IS final.

Raychan87, 2021-07-11:
> DSM 7.0 IS final.
> 
> https://www.synology.com/nl-nl/DSM70

This is the first version. I always wait for 2-3 patches. I have a productive system. Stability is important to me.

Sjekke, 2021-07-11:
> > DSM 7.0 IS final.
> 
> > 
> 
> > https://www.synology.com/nl-nl/DSM70
> 
> 
> 
> This is the first version. I always wait for 2-3 patches. I have a productive system. Stability is important to me.

I understand... (but it works ... ) 🤙