bb-qq/r8152 #115: DriverGenius CUG025 VID:0bda PID:8156 REV:31.00 working on DS218+ on the back port

jglathe, 2021-05-02:
### Description of the problem

The adapter works fine on an ioSafe N2 which has been retrofitted with a DS218+ PCB. The only "issue" I have is, as it seems, that the the adapter won't come up at boot time. Since the ioSafe is running 24/7, this is not too often. To get the interface up you need to do a stop/start the driver in the package manager. 

Debuging this a bit, I could see these interfaces immediately after boot:

jglathe@iosafeN2:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr 00:11:32:A2:54:6E
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6336 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5989 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1791738 (1.7 MiB)  TX bytes:1604278 (1.5 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6440 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6440 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1247672 (1.1 MiB)  TX bytes:1247672 (1.1 MiB)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:A2:54:6E
          inet addr:192.168.0.14  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fea2:546e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5975 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5494 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1684296 (1.6 MiB)  TX bytes:1574312 (1.5 MiB)

The r8152 driver is loaded, however:

root@iosafeN2:~# lsmod|grep r8152
r8152                 193442  0
usbcore               198691  12 btusb,etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci

And a device created, too.

root@iosafeN2:~# ls /sys/class/net
eth0  eth1  lo  ovs_eth0  ovs-system  sit0  syno_ovs_bonds

Looking for the ifcfg-eth1 file, this is not where you would need it for the `ifconfig eth1 up` command. 

Thinking about this a little, it looks that the start command for the package is not (?) executed at startup, or that the stop command is executed at shutdown (why? because package, probably). 

I'm still investigating this out of curiosity, the adapter is working fine, though. Thank you for the great work.

zhuanyi, 2021-07-30:
@jglathe  May I know are you on DSM 6 or 7 and what's the version of the driver you've used? I have bought one recently (LLANO brand but same VID and PID and Rev), I tried to install and uninstall the driver and reboot the whole diskstation after install and uninstall and had no luck getting it to work, eth3 for me was simply showing "disconnected". I am on DS412+ which unfortunately is not supported by DSM 7 so I am wondering if that's the cause. Thanks!

jglathe, 2021-07-31:
@zhuanyi hi, my DS218+ is on DSM 6.2.4-25556 currently. I think I need to have a look at these init routines again. Did you try the stop package / start package trick after rebooting? This brought mine up. Just had to do this after a run- in with the UPS :)

zhuanyi, 2021-07-31:
Thanks @jglathe, would be great if you could see if there is any other steps you've done. I have not only stop/start it but also tried to delete it altogether, reboot and then install it again and had no luck getting it to work. :(

jglathe, 2021-08-01:
@zhuanyi well then there is some debugging in order. First, is the device recognized as plugged in the info center?
[image]
Second, is the driver module loaded? `lsmod|grep r8152` should give 2 lines at least, also `dmesg|grep r8152` should give something. like: 
`jglathe@iosafeN2:~$ dmesg|grep r8152
[   84.781736] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   84.824926] usbcore: registered new interface driver r8152
[   96.566670] r8152 2-1:1.0 eth1: v2.14.0 (2020/09/24)
[   96.573642] r8152 2-1:1.0 eth1: This product is covered by one or more of the following patents:
`
If this is the case, it is more a question of bringing the eth interface up. If not, the dongle is not really recognized.

zhuanyi, 2021-08-01:
Wow, thanks so much for going this far out to help me! Really appreciate it, I have filed a separate issue here (https://github.com/bb-qq/r8152/issues/142) and this has a bit more detailed as required when submitting an issue, to answer your questions:

1. Yes it is showing in the Info Centre in DSM. interestingly I had a Club3D dongle before (one of the models that approves to work) but when I removed the Club3D device it stays in that list of USB devices but I can see both devices in list right now. One in the red box below is the new one that I am having trouble with.

[image]

2. Output of the relevant commands:
```
lsmod|grep r8152
r8152                 196503  0
usbcore               176590  9 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd
```

and
```
dmesg|grep r8152
[  187.260071] r8152: module verification failed: signature and/or required key missing - tainting kernel
[  187.284384] r8152 8-2:2.0: Unknown version 0x0000
[  187.298594] usbcore: registered new interface driver r8152
[  188.614107] r8152 8-1.2:1.0 eth2: v2.15.0 (2021/04/15)
[  188.619345] r8152 8-1.2:1.0 eth2: This product is covered by one or more of the following patents:
[  188.635472] r8152 8-1.2:1.0 eth2: chip rev 6
[20804.398560] r8152 8-1.2:1.0 eth2: get_registers -71
[20940.012040] r8152 8-1.2:1.0 eth2: get_registers -71
[22356.444802] r8152 8-1.2:1.0 eth2: get_registers -71
[23909.051140] r8152 8-1.2:1.0 eth2: get_registers -71
[27612.347010] r8152 8-1.2:1.0 eth2: get_registers -71
[29301.337535] r8152 8-1.2:1.0 eth2: get_registers -71
[107438.655727] r8152 8-1.2:1.0 eth2: get_registers -71
[107602.938697] r8152 8-1.2:1.0 eth2: get_registers -71
[108229.052956] r8152 8-1.2:1.0 eth2: get_registers -71
[108882.255323] r8152 8-1.2:1.0 eth2: get_registers -71
[111360.636443] r8152 8-1.2:1.0 eth2: get_registers -71
[112121.774205] r8152 8-1.2:1.0 eth2: get_registers -71
[112204.051267] r8152 8-1.2:1.0 eth2: get_registers -71
[112230.094508] r8152 8-1.2:1.0 eth2: get_registers -71
[112802.093087] r8152 8-1.2:1.0 eth2: get_registers -71
[114843.499719] r8152 8-1.2:1.0 eth2: get_registers -71
```

So overall I think the device is getting recognized but just for some reason it is not getting the IP and all that (I use DHCP and also on the router side I can not see the device connected at all, the other ethernet is connected to a separate router so there is no device name conflict as well). Also when I disabled the driver package in Package Centre I can see LAN3 is gone from the Network page in Control Panel and when I re-enabled LAN3 comes back so it seems the package is doing the right job.

If it helps this is the exact package file I downloaded (one in red box):
[image]

Thanks again for all your help!

jglathe, 2021-08-01:
Just an idea. The dmesg looks like some new code is there... ? Chip rev was not in my dmesg, and I'm on 2.14.0. And the get_registers error (-71) doesn't sound like anything good, it apparrently can't handle this version. And there is also a R8153 (10/100/1000 USB LAN) on the hub, or maybe it is the hub, not sure (my 8153 "hubs" here hava a LAN port and three USB3 slots, and are generally a PITA). Difficult to assign which is which now, lsusb could help (if installed). 
To debug further with this data, I would do:

1. isolate the device, remove the 8153 port (same driver)
2. go back to release 2.14 (latest) and see if it might work then.