bb-qq/r8152 #116: 16K Jumbo Frame option?

Ngakt, 2021-05-04:
Is it possible to add 16K Jumbo Frame option for 2.5GbEs in the future? I can only enable 9K Jumbo Frame in the option.

bb-qq, 2021-05-16:
Synology platforms support up to 9K MTU officially. Using the command line `ip link set ethX mtu 16334` may work, but it is unconfirmed. And I guess that increase MTU size larger than 9k does not gain throughput.

Ngakt, 2021-05-18:
Thanks a lot, I'll give it a try.

Ngakt, 2021-05-25:
It seems that the Realtek driver cannot recognize mtu that is larger than 9000 and `ip link set ethX mtu 16334` does not work(the driver returns an invalid argument error in the command line). Maybe it is okey to stick with 9000 mtu option. Thanks a lot.

closed 2021-05-25 by Ngakt