bb-qq/r8152 #119: DS918+ Cable Matters USB to 2.5G Ethernet Adapter Reboot issue and speed test.

tinyspider, 2021-05-11:
### Description of the problem
DS918+ no 2.5G connection after reboot. In 'network interface' in DSM it shows the 2.5gbe is connected, but using the assigned IP, I am not able to reach it. I tried to increase the sleep time from 5 to 120. It did not work. 
The workaround I found is I need to change the static IP to a different one, i.e. 192.168.1.240 and the network adapter will work properly. I can then change it back to the desired IP 192.168.1.230.
When doing speed test using librespeed/speedtest docker image, although I am accessing the docker image via 192.168.1.230 (2.5gbe adapter), the traffic from NAS to PC is actually handled by LAN1, the internal gigbyte port, and the traffic from PC to NAS is handled by LAN3.
Please let me know if there is additional information I can provide.
Thanks for your time.

### Description of your products

> DS918+ 
>     Linux jnas 4.4.59+ #25556 SMP PREEMPT Thu Mar 18 13:00:35 CST 2021 x86_64 GNU/Linux synology_apollolake_918+
>    DSM 6.2.4-25556
> Cable Matters USB to 2.5G Ethernet Adapter Supporting 2.5 Gigabit Ethernet Network - USB C Adapter Included for USB-C and Thunderbolt 3

### Description of your environment

> - connect to a qnap qsw-m408-2c 10gbe switch
>     Marvell AQtion 10Gbit Network Adapter AQC107
>    qsw-m408-2c 
>    CAT 6 
### Output of ``dmesg`` command

````

[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build3) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25556 SMP PREEMPT Thu Mar 18 13:00:35 CST 2021
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 syno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syno_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 sn=1970PDN869100 macs=001132b455eb,001132b455ec
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000067c17fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c18000-0x0000000067c18fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000067c19000-0x0000000067c42fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000067c43000-0x0000000067c7dfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c7e000-0x0000000068c7dfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000068c7e000-0x000000007822efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007822f000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007afbf000-0x000000007affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d370afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] bootconsole [earlyser0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x01d646c8-0x01d646d7] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001d646c7] usable
[... 1185 lines cut ...]
ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:B4:55:EC
          inet addr:192.168.1.231  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2141 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:295582 (288.6 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.0.0.1  P-t-P:10.0.0.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:0 (0.0 B)  TX bytes:306 (306.0 B)

````

Garfield999, 2021-05-13:
https://github.com/bb-qq/r8152/issues/106

bb-qq, 2021-05-16:
Thank you for creating the PR. Indeed I had not confirmed DHCP configuration. 
I merged the PR and create a new release. Please check it.

yaoshuangshl, 2021-07-03:
Has the problem of having to switch DHCP to use the adapter after reboot been solved? The description of the problem is exactly the same as mine.
Similarly, I also had problems with download speeds compared to other adapters.

Garfield999, 2021-07-03:
I solved it with:

if you want to assign a static ip, add this in synology taskmanager.
(without it, the ip in syno settings was not assigned for me with ds918+ after reboot)

taskmanager (user root, boot-up)

synonet --set_ip -4 eth2 add 192.168.178.3/24 

or for getting ip from dhcp:
synonet –dhcp eth2

Change ip And/or subnet to your needs. 24 means 255.255.255.0

tinyspider, 2021-07-08:
> Has the problem of having to switch DHCP to use the adapter after reboot been solved? The description of the problem is exactly the same as mine.
> Similarly, I also had problems with download speeds compared to other adapters.
I have since given up on the 2.5g adapter solution. The cause is that I can not solve the speed issue. I have 2.5gbps upload while only 1gbps download from PC, which I believe the upload traffic is only going through the USB adapter. In addition, I updated DSM 7 and it does not work with this driver.

bb-qq, 2021-07-11:
If you still experience DHCP problems, please post the version of the driver you are using and the `dmesg` output.

And if you are experiencing the problem of asymmetric throughput, it is possible that traffic is not going through the new NIC properly due to a routing problem. Try assigning a different subnet to the new NIC. You can check if the problem is caused by routing by looking at the increase in `RX bytes`/`TX bytes` in `ifconfig`.

Garfield999, 2021-07-11:
Thebonly way forn fixing ip/(dhcp address problem is the way I posted above. Raising the sleep timer does not work.

It's no problem, because both (dhcp and static) ip works after rebooting after adding the custom script to syno taskmanager.

bb-qq, 2022-12-31:
Synology has changed the IP address configuration process in 7.1.1, and the driver has also undergone significant changes to track these changes.

On the other hand, this Issue refers to the old driver,  so I would like to close it for now.

@thedatabaseme, I guess your problem seems to be another issue, please create a new, independent Issue and describe the detailed network configuration information there.

closed 2022-12-31 by bb-qq