bb-qq/r8152 #122: Not working on rtd1296-based DS220j

marcosscriven, 2021-05-18:
### Description of the problem

Installed package for rtd1296 on my DS220j. Upon inserting device, I see several processes die.

### Description of your products

DS220j
rtd1296
DSM 6.2.4-25556
[Sabrent NT-S25G ](https://www.sabrent.com/product/NT-S25G/usb-type-a-or-type-c-to-2-5-gigabit-ethernet-adapter-10-100-1000-2500-nt-s25g/#description)

### Description of your environment

Direct to PC
Another NT-S25G
Short cat6 cable

### Output of ``dmesg`` command

````
[  190.151887] usb 5-1: new SuperSpeed USB device number 2 using xhci-hcd
[  190.387630] usb 5-1: reset SuperSpeed USB device number 2 using xhci-hcd
[  190.502040] netif_napi_add() called with weight 256 on device eth%d
[  190.541820] r8152 5-1:1.0 eth1: v2.14.0 (2020/09/24)
[  190.546913] r8152 5-1:1.0 eth1: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[  193.554448] Bluetooth: Core ver 2.21
[  193.558197] NET: Registered protocol family 31
[  193.562810] Bluetooth: HCI device and connection manager initialized
[  193.569500] Bluetooth: HCI socket layer initialized
[  193.574590] Bluetooth: L2CAP socket layer initialized
[  193.579817] Bluetooth: SCO socket layer initialized
[  193.596915] Bluetooth: RFCOMM TTY layer initialized
[  193.602072] Bluetooth: RFCOMM socket layer initialized
[  193.607417] Bluetooth: RFCOMM ver 1.11
[  193.641079] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  193.647272] Bluetooth: HIDP socket layer initialized
[  193.838857] usbcore: registered new interface driver btusb
[  195.501889] init: btacd main process (9978) killed by TERM signal
[  197.065687] init: nmbd main process (10288) killed by TERM signal
[  201.662438] init: bluetoothd main process (9939) killed by KILL signal
````

### Output of ``lsusb`` command

````
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
  |__5-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:E4:92:A9                       
          inet addr:192.168.1.204  Bcast:192.168.1.255  Mask:255.255.255.0    
          inet6 addr: fe80::211:32ff:fee4:92a9/64 Scope:Link                  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1                  
          RX packets:11925 errors:0 dropped:0 overruns:0 frame:0              
          TX packets:9230 errors:0 dropped:0 overruns:0 carrier:0             
          collisions:0 txqueuelen:1000                                        
          RX bytes:2880220 (2.7 MiB)  TX bytes:6164369 (5.8 MiB)              
          Interrupt:11                                                        
                                                                              
eth1      Link encap:Ethernet  HWaddr 00:24:27:88:0A:B0                       
          BROADCAST MULTICAST  MTU:1500  Metric:1                             
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0                  
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0                
          collisions:0 txqueuelen:1000                                        
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)                              
                                                                              
lo        Link encap:Local Loopback                                           
          inet addr:127.0.0.1  Mask:255.0.0.0                                 
          inet6 addr: ::1/128 Scope:Host                                      
          UP LOOPBACK RUNNING  MTU:65536  Metric:1                            
          RX packets:28 errors:0 dropped:0 overruns:0 frame:0                 
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0               
          collisions:0 txqueuelen:1                                           
          RX bytes:3363 (3.2 KiB)  TX bytes:3363 (3.2 KiB)                    
                                                                              
sit0      Link encap:IPv6-in-IPv4                                             
          NOARP  MTU:1480  Metric:1                                           
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0                  
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0                
          collisions:0 txqueuelen:1                                           
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)                              
                                                                              
````

marcosscriven, 2021-05-18:
Hmm - scratch that, after trying to unplug/plug a few times, I do manage to get a connection. 

Not sure why, but sometimes I have to run `ifconfig eth1 up`, and sometimes not.

iperf3 result:

```
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.2.1, port 51497
[  5] local 192.168.2.2 port 5201 connected to 192.168.2.1 port 51498
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   255 MBytes  2.14 Gbits/sec
[  5]   1.00-2.00   sec   272 MBytes  2.28 Gbits/sec
[  5]   2.00-3.00   sec   269 MBytes  2.25 Gbits/sec
[  5]   3.00-4.00   sec   273 MBytes  2.29 Gbits/sec
[  5]   4.00-5.00   sec   270 MBytes  2.26 Gbits/sec
[  5]   5.00-6.00   sec   272 MBytes  2.28 Gbits/sec
[  5]   6.00-7.00   sec   271 MBytes  2.28 Gbits/sec
[  5]   7.00-8.00   sec   269 MBytes  2.26 Gbits/sec
[  5]   8.00-9.00   sec   274 MBytes  2.30 Gbits/sec
[  5]   9.00-10.00  sec   271 MBytes  2.27 Gbits/sec
[  5]  10.00-10.04  sec  10.8 MBytes  2.25 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.04  sec  2.64 GBytes  2.26 Gbits/sec                  receiver
-----------------------------------------------------------
```

marcosscriven, 2021-05-18:
Arghh... it's unstable. After working a few minutes, iperf3 stops and I see:

```
[ 1608.463064] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.471358] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.479642] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.487926] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.496210] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.504494] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.512777] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.521061] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.529344] xhci-hcd xhci-hcd.5.auto: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 1608.537651] xhci-hcd xhci-hcd.5.auto: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 3 comp_code 1
[ 1608.548426] xhci-hcd xhci-hcd.5.auto: Looking for event-dma 000000001e165130 trb-start 000000001e165040 trb-end 000000001e165040 seg-start 000000001e165000 seg-end 000000001e165ff0
[ 1608.564889] xhci-hcd xhci-hcd.5.auto: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 3 comp_code 1
[ 1608.575662] xhci-hcd xhci-hcd.5.auto: Looking for event-dma 000000001e165140 trb-start 000000001e165040 trb-end 000000001e165040 seg-start 000000001e165000 seg-end 000000001e165ff0
[ 1634.009627] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
[ 1634.016264] ------------[ cut here ]------------
[ 1634.020990] WARNING: at net/sched/sch_generic.c:306
[ 1634.025983] Modules linked in: btusb btrtl btintel btbcm hidp rfcomm bluetooth r8152(OE) cifs udf isofs loop tcm_loop(O) iscsi_target_mod(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) nfsd exportfs rpcsec_gss_krb5 syno_hddmon(P) hid_generic usbhid hid usblp usb_storage uhci_hcd ds220j_synobios(PO) exfat(O) btrfs synoacl_vfs(PO) zlib_deflate hfsplus md4 hmac r8168(O) i2c_algo_bit vxlan ip6_udp_tunnel udp_tunnel fuse vfat fat aes_neon_blk aes_ce_blk gf128mul ablk_helper arc4 cryptd ecryptfs sha256_generic ecb authenc des_generic ansi_cprng cts md5 cbc dm_snapshot dm_bufio crc_itu_t crc_ccitt quota_v2 quota_tree psnap p8022 llc sit tunnel4 ip_tunnel ipv6 sg dwc3_rtk xhci_pci xhci_plat_hcd xhci_hcd ehci_pci
[ 1634.098998]  ehci_hcd dwc3 usbcore usb_common phy_rtk_usb3 phy_rtk_usb2 phy_rtk_rle0599 [last unloaded: ds220j_synobios]

[ 1634.110415] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE   4.4.59+ #25556
[ 1634.118163] Hardware name: Realtek_RTD1296 (DT)
[ 1634.122800] task: ffffffc000a4b2c0 ti: ffffffc000a40000 task.ti: ffffffc000a40000
[ 1634.130468] PC is at dev_watchdog+0x24c/0x25c
[ 1634.134929] LR is at dev_watchdog+0x24c/0x25c
[ 1634.139388] pc : [<ffffffc0007315ec>] lr : [<ffffffc0007315ec>] pstate: 60000145
[ 1634.146958] sp : ffffffc000a43c00
[ 1634.150350] x29: ffffffc000a43c00 x28: 00000000000000a0
[ 1634.155810] x27: 0000000000000008 x26: dead000000000200
[ 1634.161269] x25: ffffffc01df829b8 x24: ffffffc01df827b8
[ 1634.166729] x23: 00000000ffffffff x22: ffffffc000a91000
[ 1634.172189] x21: ffffffc000a46000 x20: ffffffc00675c000
[ 1634.177650] x19: 0000000000000000 x18: fffffffffffffffb
[ 1634.183110] x17: 0000007fa69bfe48 x16: ffffffc0002eb658
[ 1634.188571] x15: 003b9aca00000000 x14: 0fffffffffffffff
[ 1634.194031] x13: 0000000000000028 x12: 0101010101010101
[ 1634.199491] x11: ffffffc000a55000 x10: 00000000000002b3
[ 1634.204952] x9 : ffffffc000a43960 x8 : 756575712074696d
[ 1634.210412] x7 : 00000000000002b4 x6 : 0000000000000049
[ 1634.215872] x5 : 0000000000000000 x4 : 0000000000000000
[ 1634.221332] x3 : 0000000000000000 x2 : ffffffc01df83228
[ 1634.226790] x1 : ffffffc01df81ae0 x0 : 0000000000000039

[ 1634.233775] Call trace:
[ 1634.236282] [<ffffffc0007315ec>] dev_watchdog+0x24c/0x25c
[ 1634.241812] [<ffffffc0002ea310>] call_timer_fn.isra.29+0x28/0x88
[ 1634.247962] [<ffffffc0002ea4f0>] run_timer_softirq+0x180/0x1f0
[ 1634.253936] [<ffffffc00029c8d0>] __do_softirq+0xf8/0x218
[ 1634.259375] [<ffffffc00029cc0c>] irq_exit+0x64/0x90
[ 1634.264371] [<ffffffc0002db424>] __handle_domain_irq+0x60/0xb0
[ 1634.270343] [<ffffffc000280cf8>] gic_handle_irq+0x60/0xb8
[ 1634.275869] Exception stack(0xffffffc000a43db0 to 0xffffffc000a43ee0)
[ 1634.282463] 3da0:                                   ffffffc000a46000 0000008000000000
[ 1634.290480] 3dc0: ffffffc000a43f10 ffffffc000284e70 0000000060000145 ffffffc0007ea000
[ 1634.298496] 3de0: 0000000000000000 0000000000000000 0000000000000001 ffffffc000a47000
[ 1634.306512] 3e00: 0000000000000001 0000000000000000 0000000000048540 000000010002091c
[ 1634.314529] 3e20: ffffffc000a4bb30 ffffffc000a43e80 0000000000000810 0000000000000030
[ 1634.322545] 3e40: 0000000000000000 ffff000000000000 282039302f32302f 003b9aca00000000
[ 1634.330561] 3e60: ffffffc0002eb658 0000007fa69bfe48 fffffffffffffffb ffffffc000a46000
[ 1634.338577] 3e80: ffffffc000a46a48 ffffffc000a2bac0 ffffffc000a2f048 ffffffc000a43f30
[ 1634.346594] 3ea0: ffffffc0007ea000 ffffffc000a46000 ffffffc000a40000 0000000000000000
[ 1634.354610] 3ec0: 0000000000000000 ffffffc000a43f10 ffffffc000284e6c ffffffc000a43f10
[ 1634.362626] [<ffffffc000283f00>] el1_irq+0x80/0x100
[... cut ...]
[ 1634.414208] r8152 5-1:1.0 eth1: Tx status -2
[ 1634.418601] r8152 5-1:1.0 eth1: Tx status -2
[ 1636.424898] r8152 5-1:1.0 eth1: get_registers -108
[ 1636.429882] r8152 5-1:1.0 eth1: get_registers -71
<--- repeats dozens of times --->
[ 1636.572539] r8152 5-1:1.0 eth1: get_registers -71
[ 1636.601036] xhci-hcd xhci-hcd.5.auto: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 1636.610735] r8152 5-1:1.0 eth1: get_registers -108
[ 1636.610791] usb 5-1: reset SuperSpeed USB device number 2 using xhci-hcd
```

marcosscriven, 2021-05-18:
Hmm  - I see @Raychan87 used the older driver https://github.com/bb-qq/r8152/issues/114#issuecomment-827853571

Should we be using that one?

marcosscriven, 2021-05-18:
@bb-qq - is there any way to get an rtd2196 build of 2.12.0-3.spk please (or let me know how to build for it).

Per prior comment, someone has found that worked for their 8156-based adapter, and it looks like there's a huge number of changes in the Realtek driver since.

bb-qq, 2021-05-29:
I created [r8152-rtd1296-2.12.0-3.spk](https://github.com/bb-qq/r8152/releases/download/2.12.0-3/r8152-rtd1296-2.12.0-3.spk). If you wish to create it by yourself, please see [this page](https://help.synology.com/developer-guide/compile_applications/compile_kernel_modules.html).

bb-qq, 2021-05-29:
FYI: Today I released the new release [2.15.0-1](https://github.com/bb-qq/r8152/releases/tag/2.15.0-1) that includes many changes from Realtek. You might find worth in this release...

deveth0, 2021-06-03:
hey, thanks for posting this bug, saved me a lot of debugging. 
I ran into the same issue on a 918+ with the Club3 thing (same revision), will give 2.15.0-1 a try now!

bb-qq, 2021-07-11:
It's been a while since my last post, so I'm closing this issue for now. If you still have any problems, please feel free to reopen this issue.

closed 2021-07-11 by bb-qq