bb-qq/r8152 #123: Need support for Biaze KZ13

xuqingkuang, 2021-05-26:
### Description of the problem
The network adapter doesn't appear in the control panel.

### Description of your products

- HP MicroServer Gen8
  - Linux microserver 3.10.105 #25426 SMP Mon Dec 14 18:47:46 CST 2020 x86_64 GNU/Linux synology_broadwell_3617xs
  - DSM 6.2.3-25426 Update 3
- Biaze KZ13: https://www.banggood.com/BIAZE-KZ13-USB-External-2_5G-Network-Adapter-USB-to-RJ45-Converter-RJ45-Network-Port-Hub-USB-Gigabit-Wired-Network-Card-for-Macbook-or-Surface-or-Lenovo-or-ASUS-Computers-p-1815264.html

### Output of ``dmesg`` command

````
[ 2425.128112] usb 5-1: USB disconnect, device number 2
[ 2430.328214] usb 5-2: new SuperSpeed USB device number 3 using xhci_hcd
[ 2430.340431] usb 5-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 2430.348660] r8152 5-2:2.0: Unknown version 0x7410
````

### Output of ``lsusb`` command

````
|__usb5          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:04:00.0) hub
  |__5-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 001000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx
          inet addr:169.254.145.210  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:16 

eth1      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx  
          inet addr:192.168.88.221  Bcast:192.168.255.255  Mask:255.255.0.0
          inet6 addr: fe80::b25a:daff:fe87:e459/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5418 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9184 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1740232 (1.6 MiB)  TX bytes:3818043 (3.6 MiB)
          Interrupt:17 

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:467 errors:0 dropped:0 overruns:0 frame:0
          TX packets:467 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:42061 (41.0 KiB)  TX bytes:42061 (41.0 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2021-05-29:
Today I released the new release [2.15.0-1](https://github.com/bb-qq/r8152/releases/tag/2.15.0-1) that includes support of a new chip revision. Hopefully, this release supports your device.

xuqingkuang, 2021-05-31:
> Today I released the new release [2.15.0-1](https://github.com/bb-qq/r8152/releases/tag/2.15.0-1) that includes support of a new chip revision. Hopefully, this release supports your device.

The problem still exists after update.
 
I search the keyword by `Unknown version 0x7410` found there are some related issues & PR, but not sure the solution

https://github.com/bb-qq/r8152/issues/61
https://github.com/bb-qq/r8152/pull/26
https://github.com/bb-qq/r8152/issues/29

bb-qq, 2021-07-11:
Hmm... Recent drivers should work with 0x7410(RTL_VER_13) devices. That log is strange.

Eiranya, 2021-07-13:
> Hmm... Recent drivers should work with 0x7410(RTL_VER_13) devices. That log is strange.

I'm tryed KZ13 can work fine at 920+ with 2.15.0-2

bb-qq, 2021-07-15:
Thanks for the information. I am closing this issue for now. If you still have problems, please feel free to reopen the issue.

closed 2021-07-15 by bb-qq

xuqingkuang, 2021-09-05:
Hi, I just success solved the probelm from the tips - https://github.com/bb-qq/r8152/issues/7#issuecomment-578098899

After restart the service, the network interface is appear.

```
sudo synoservice --restart pkgctl-r8152
sudo ifconfig eth2 up
``

With upon comments, the `dmesg` output is correct.

```
[  927.463678] usb 5-2: USB disconnect, device number 2
[  933.014325] usb 5-2: new SuperSpeed USB device number 3 using xhci_hcd
[  933.026494] usb 5-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[  933.034783] r8152 5-2:2.0: Unknown version 0x7410
[ 1935.317487] usbcore: deregistering interface driver r8152
[ 1935.588297] usbcore: registered new interface driver r8152
[ 1935.717034] usb 5-2: reset SuperSpeed USB device number 3 using xhci_hcd
[ 1935.730014] usb 5-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 1935.731964] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803f49da980
[ 1935.731968] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803f49da9c0
[ 1935.731971] xhci_hcd 0000:04:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8803f49daa00
[ 1946.432820] netif_napi_add() called with weight 256 on device eth%d
[ 1946.442363] r8152 5-2:1.0 eth2: v2.15.0 (2021/04/15)
[ 1946.442373] r8152 5-2:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.
[ 1946.442375] r8152 5-2:1.0 eth2: chip rev 14
[ 1947.611641] init: scsi_plugin_server main process (20855) killed by TERM signal
[ 1947.789636] init: nmbd main process (4106) killed by TERM signal
[ 2041.133981] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[ 2041.208305] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[ 2041.210223] r8152 5-2:1.0 eth2: carrier on
```