bb-qq/r8152 #125: Request for DS414slim / Armada 370

rwesterflan, 2021-06-05:
### Description of the problem

> I recently acquired a DS414slim, filled it with SSD's and I want to use it as a iSCSI backend for VMWare. A 2500mbps adapter would be ideal for this.

### Description of your products

> - DS414slim
>     - Linux thor 3.2.40 #25556 Thu Mar 4 17:56:47 CST 2021 armv7l GNU/Linux synology_armada370_414slim
>    - CPU MARVELL Armada 370 88F6707
>     - DSM version 6.2.4-25556
> - 'generic white Aliexpress unit', 0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)

### Description of your environment

> - Direct connected to Dell Optiplex 3070, with the same adapter
> - Cat 6 cable (0.5m)

Thanks in advance!

bb-qq, 2021-07-10:
It seems that the kernel version of armada370 is too old in order to use xHCI devices...
I will continue to investigate this issue, but there is little hope that it will work.

rwesterflan, 2021-08-06:
Hey, no worries, I already sold the unit and replaced it something else that has support :) Thank you!

closed 2021-08-06 by rwesterflan