bb-qq/r8152 #126: Add alpine4k support (e.g. 416)

deveth0, 2021-06-07:
Hey, is there any chance to get builds for alpine4k architecture (e.g. DS416)?

sftt, 2021-06-27:
DS215+ also apline4k, Thanks.
Linux mynas 3.10.105 #25556 SMP Thu Mar 4 17:49:58 CST 2021 armv7l GNU/Linux synology_alpine4k_ds215+
Linux version 3.10.105 (root@build1) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25556 SMP Thu Mar 4 17:49:58 CST 2021

helsinkii, 2021-07-09:
DS416 apline4k, thanks

bb-qq, 2021-07-10:
I added the support of alpine4k from the release for DSM 7.0. Is it enough?

deveth0, 2021-07-10:
Awesome, thanks! Good reason to upgrade :)

helsinkii, 2021-07-14:
can't install the driver on DSM7 from the SSH terminal
install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory

bb-qq, 2021-07-15:
> install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory

It seems that spk has not been installed.
Did you proceed to the point where the Package Manager shows an error?

helsinkii, 2021-07-15:
> > install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory
> 
> It seems that spk has not been installed.
> Did you proceed to the point where the Package Manager shows an error?

Sorry, it seems that I mixed up alpine with alpine4k.  Have the driver for alpine4k been released? I can't find It.

bb-qq, 2021-07-18:
Build outputs produced by SDK of alpine4k has the same architecture name `alpine` in spk. So I believe that -alpine.spk works for your model. If not, please let me know.

helsinkii, 2021-07-19:
> > install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory
> 
> It seems that spk has not been installed.
> Did you proceed to the point where the Package Manager shows an error?

DS416, DSM 7.0, manually uploaded "r8152-alpine-2.15.0-3.spk" in package center ,  pop-up message "the package is not supported on the platform of Diskstation or is incompattible with the current DSM version."

bb-qq, 2021-07-20:
I created two versions of the package to investigate this issue. Could you try both?
https://github.com/bb-qq/r8152/releases/download/2.15.0-3/r8152-alpine4k-2.15.0-3.spk
https://github.com/bb-qq/r8152/releases/download/2.15.0-3/r8152-alpine4k_test-2.15.0-3.spk

helsinkii, 2021-07-27:
> I created two versions of the package to investigate this issue. Could you try both?
> https://github.com/bb-qq/r8152/releases/download/2.15.0-3/r8152-alpine4k-2.15.0-3.spk
> https://github.com/bb-qq/r8152/releases/download/2.15.0-3/r8152-alpine4k_test-2.15.0-3.spk

I tried the first one, it worked. Will try the second one this week.

bb-qq, 2021-11-21:
I am adopting the first one as the official release. If you have any problems, please create a new issue.

closed 2021-11-21 by bb-qq