bb-qq/r8152 #128: Update for Support in DSM 7

plangpl, 2021-06-29:
it is necessary to create a new spk for users with DSM 7. The new DSM does not have support for packages which want root privileges. Thanks

19Fidel84, 2021-06-29:
Oh yes, I would be very interested today the final version DSM7 came out and you don't want to miss your 2.5Gbit

Sjekke, 2021-06-29:
Same here ... I'm already on DSM 7 beta and it's much quicker ... but I also want the 2.5gbit driver ... 

🙏 🙏 🙏

Gensplejs, 2021-07-02:
Yeah... so DSM 7 has been released now. Hoping there is some way to allow me to use 2.5g with dsm7

lai-zai, 2021-07-05:
DSM7 has matured and is now mainstream

BiffaZ, 2021-07-08:
Might be SOL folks. Looks like Synology has removed support for USB devices. The update readme for all models shows:

> **DSM**
> 
> USB devices (Wi-Fi dongle, Bluetooth dongle, 3G/4G dongle, USB DAC/speaker, and DTV dongle) are no longer supported. If your Synology NAS is currently connected via a wireless dongle, it will be disconnected after the update.

hhessel, 2021-07-08:
@BiffaZ So you're linking to Reddit which links back to this repo and another issue here: #88 . Can we close this issue here so we only have one?

BiffaZ, 2021-07-09:
Its not my issue, but close if you wish :)

bb-qq, 2021-07-10:
I have created and uploaded drivers for DSM 7.0. Try it, please.

prismon, 2021-07-10:
Hey @bb-qq - I am working on testing it, but do not see the new DSM 7.0 drivers in the release or downloads folder.

Gensplejs, 2021-07-10:
Awsone!!!!
Got it working perfectly on ds918+!

stepman0, 2021-07-10:
@bb-qq Where did you upload the new drivers?

stepman0, 2021-07-10:
Just found it. They are placed second on the releases page.

If somebody else is searching like me: https://github.com/bb-qq/r8152/releases/tag/2.15.0-2

Gensplejs, 2021-07-10:
So if anyone has problems figuring out the excat steps..
Install the new spk...
The install fails... (do not uninstall)
connect via ssh and run "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su"
install package again and everything it works.

stepman0, 2021-07-10:
First test on DS918+ looks very promising. Driver is installed, Test copy via SMB from Windows PC achieved ~270MB/s - same as DSM6
@bb-qq 👍

prismon, 2021-07-10:
Ditto - working on a 918+. Two issues I have found:
1) Using MTU above 1500 seems to crash the driver at sustained traffic. Unplugging seems to resolve it, as does setting the MTU to 1500. 
2) Getting 150MB/s on write, but only 100mb/s read. That seems strange. 

All in all, it's awesome to have some performance back. It's a 50% boost. 

(Tested on two seperate USB dongles).

Sjekke, 2021-07-10:
Testing on a 916+ till now everything ok.

wkupe50hon, 2021-07-10:
Works on DS920+. Restarted several times and still good to go. Created account here to say thank you. Thank You for your time. You guys are cool people.

mrjohnpoz, 2021-07-11:
This was one of the thing holding back from moving to 7.. Sweet thanks!

stepman0, 2021-07-12:
I had a sudden reboot of my DS918+ (never had this before). The PC connected via 2,5G was running, but no data transfer. I will watch this...

@bb-qq: Any logfile I can have a look at for investigation?

ibizastyler, 2021-07-12:
thank you so far! 

With DSM 6.2, the old driver was working like a charm with the USB-ENET adapter "CLUB 3D CAC-1420" on my DS918+ and DS716+.

I'm testing your new drivers (braswell) now with the RC/stable DSM 7 version on my ds716+ (backup).

The first installation didn't work, but with your provided command line...

"sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su"

... the second try was successful.

Unfortunately, I was just able to get the USB-ENET working, but not with the same speed of about 220-260 MB/s I had with DSM 6.2. The copy process starts with ~ 250 MB/s, but fells down to ~ 120 MB/s - tested with different file sizes.

I restartet my 2.5 switch and also the DS716+, but nothing changed.

Sjekke, 2021-07-12:
Are there any 5gbit adapters that work with this driver?

19Fidel84, 2021-07-13:
CLUB 3D CAC-1420 works well now on my DS 1019+ 
big thx 4 the fix

mrjohnpoz, 2021-07-13:
So finally pulled the trigger on dsm 7 update.. Went smooth - got my usb nic working with it thanks to this driver update.  Thanks again.. DS918+ with the Club 3D CAC-1420 USB 3.2 Gen1 Type A to RJ45 2.5Gbps Ethernet Adapter

bb-qq, 2021-07-15:
@Sjekke , as far as I know, Realtek has not released a USB adapter that supports 5Gbps. But you can use Aquantia's devices instead.
https://github.com/bb-qq/aqc111

bb-qq, 2021-07-15:
@stepman0 ,I don't have any ideas to investigate the issue...  There is no information left under /var/log, is there?

stepman0, 2021-07-15:
I have not seen anything useful in /var/log. Until now the issue did not occur again. I will keep you updated if it happens again...

zorglub74, 2021-07-15:
The last release work perfectly fine on my DS918+ a huge thank you to the contributors that made it happens.

jvonseoul, 2021-07-23:
> So if anyone has problems figuring out the excat steps..
> Install the new spk...
> The install fails... (do not uninstall)
> connect via ssh and run "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su"
> install package again and everything it works.

Currently using 918+ and followed the steps above. However I receive the message
"install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory"

What may I be doing wrong here?

k-vq, 2021-07-24:
Hi @bb-qq , is there any chance you can add support for Evansport architect with DSM7?

bb-qq, 2021-09-24:
I uploaded [2.15.0-4 release](https://github.com/bb-qq/r8152/releases/tag/2.15.0-4) which includes the evansport support.

bb-qq, 2021-09-24:
I think work for DSM 7.0 is complete. I close this issue.

I would really appreciate it if you report to #135 or a new issue whether it works and its stability. It is very helpful for new comers and useful for me to investigate problems. It doesn't matter if the reports are the same as those of other users. The number of reports is also an important piece of information.

closed 2021-09-24 by bb-qq

axd1152, 2025-03-23:
Has anyone tried on DS220+?

eswagar, 2025-03-23:
I have DS220+, 7.2.2-7208U3 with Cable Creation 2.5 USB dongle with the RTL8156 chip.  No issue.

axd1152, 2025-03-23:
> I have DS220+, 7.2.2-7208U3 with Cable Creation 2.5 USB dongle with the RTL8156 chip. No issue.

Thanks, do you have it attached to the front of the nas or rear side?

eswagar, 2025-03-23:
I have 3 Synology's, all of them are attached to the front and all are running 2.19.2-1.

axd1152, 2025-03-23:
> 2.19.2-1.

Is that the driver version? 
So, if you have them connected to the front, that means, Synology USB-COPY app will does not work anymore, as it works only with the front port?

axd1152, 2025-03-23:
> I have 3 Synology's, all of them are attached to the front

I guess I could have USB hub attached to the front and have the dongle attached to the hub, and then have my USB-COPY devices attached to the HUB? Should probably work that way?

eswagar, 2025-03-23:
I don't know if the USB-COPY app works or not.  I've never used it.

axd1152, 2025-03-24:
> I don't know if the USB-COPY app works or not. I've never used it.

Got my ugreen 2.5gbps to usb-c adaptor today. Tested on my DS220+ on DSM 7.2. Both front and back usb work fine so far. I also testing with ugreen usb hub having the rj45 adaptor and external usb box with w 3.5 hard drives connecte at the same time to the hub, and it WORKED!

There is one question I have though. When I installed driver, I downloaded driver for DSM 7.2, now say DSM updates to 7.3, how do I update the driver? Do I have to uninstall the old one from package center and the install the new one using ssh again?