bb-qq/r8152 #129: Only 1000 MBit/s on DS620slim

Chrno21, 2021-07-01:
### Description of the problem
Only 1000 MBit/s, Vollduplex, MTU 1500

### Description of your products

> - Synology DS620slim
>     - Linux Syno 4.4.59+ #25556 SMP PREEMPT Thu Mar 18 13:00:35 CST 2021 x86_64 GNU/Linux synology_apollolake_620slim
>     - DSM 6.2.4-25556
> - ASUSTOR AS-U2.5G

### Description of your environment

> - connect directly to a PC Windows NAS
>     - PC: ASUSTOR AS-U2.5G
> - Cat.7

2,5 Gb/s Works between Laptop(Windows) and old NAS(Windows)

### Output of ``dmesg`` command

<details>
  <summary>dmesg output</summary>

````
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build3) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25556 SMP PREEMPT Thu Mar 18 13:00:35 CST 2021
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=6 netif_num=2 HddHotplug=1 syno_hw_version=DS620slim vender_format_version=2 syno_hdd_detect=18,179,176,175,177,171 syno_hdd_enable=21,20,19,9,173,172 syno_usb_vbus_gpio=11@0000:00:15.0@1,10@0000:00:15.0@2,10@0000:00:15.0@3 syno_spinup_group=6 syno_spinup_group_delay=0 synoboot2 sn=2120R2RA3AY4S macs=001132eeeeb3,001132eeeeb4
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000067c27fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c28000-0x0000000067c28fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000067c29000-0x0000000067c42fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000067c43000-0x0000000067c7dfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c7e000-0x0000000068c7dfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000068c7e000-0x000000007822efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007822f000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007afbf000-0x000000007affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d370afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000001ffffffff] usable
[    0.000000] bootconsole [earlyser0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x01d646c8-0x01d646d7] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001d646c7] usable
[... 1035 lines cut ...]
          inet addr:192.168.1.20  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::224:27ff:fe88:484/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:67 errors:0 dropped:0 overruns:0 frame:0
          TX packets:184 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12594 (12.2 KiB)  TX bytes:34125 (33.3 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:36 errors:0 dropped:0 overruns:0 frame:0
          TX packets:36 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:4422 (4.3 KiB)  TX bytes:4422 (4.3 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````
</details>

bb-qq, 2021-07-11:
> 2,5 Gb/s Works between Laptop(Windows) and old NAS(Windows)

I have a feeling that this problem is caused by the routing configuration. Can you please provide more details about the network configuration that is not working properly?

Chrno21, 2021-07-12:
Oh I did not want to create confusion.
That was only as an example that the connection works without the Synology.

The connection I need and where only 1 GBit/s works is:
DS620slim > ASUSTOR AS-U2.5G > Cat.7 Lan Cable > ASUSTOR AS-U2.5G > Old Windows based NAS

I think the problem is this:
`  |__1-1         0bda:8156:3000 00  2.10  480MBit/s 500mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)`
and
`[   42.660855] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   42.781525] usbcore: registered new interface driver r8152
[   42.911676] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   43.071594] usb 1-1: not running at top speed
[   43.196220] init: btacd main process (6893) killed by TERM signal
[   43.271926] init: iscsi_pluginserverd main process (10931) terminated with status 255
[   43.296889] init: iscsi_pluginserverd main process ended, respawning
[   43.445498] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   43.487180] usbcore: deregistering interface driver btusb
[   43.593568] usb 1-1: not running at top speed
[   43.968492] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   44.128541] usb 1-1: not running at top speed
[   44.503433] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   44.654511] usb 1-1: not running at top speed
[   45.029452] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   45.195499] usb 1-1: not running at top speed
[   45.571305] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   45.727295] usb 1-1: not running at top speed
[   45.822451] netif_napi_add() called with weight 256 on device eth%d
[   45.845127] r8152 1-1:1.0 eth2: v2.15.0 (2021/04/15)
[   45.873117] r8152 1-1:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[   45.983064] r8152 1-1:1.0 eth2: chip rev 12`

bb-qq, 2021-07-13:
Indeed, that must be the cause.

Could you try following?
* connect to another port
* use another xHCI device (eg. mass storage devices) to isolate the problem

But, since whether use SuperSpeed mode is determined by the Host Controller, there is nothing to do by the Class Driver.

Chrno21, 2021-07-13:
This was the first thing I tested.
But the same on both ports.

The USB Drives/HDDs are all recognized as USB 3.2 on both Ports.

Chrno21, 2021-07-14:
I baught a Club 3D CAC-1420 and 2,5GBit/s are "working".

PC > ASUSTOR AS-U2.5G > Cat.7 Lan Cable > Club 3D CAC-1420 > DS620slim:
```
Connecting to host 192.168.1.19, port 5201
[  5] local 192.168.1.20 port 50590 connected to 192.168.1.19 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   1.00-2.00   sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   2.00-3.00   sec   239 MBytes  2.00 Gbits/sec    0    212 KBytes
[  5]   3.00-4.00   sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   4.00-5.00   sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   5.00-6.00   sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   6.00-7.00   sec   238 MBytes  2.00 Gbits/sec    0    212 KBytes
[  5]   7.00-8.00   sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   8.00-9.00   sec   240 MBytes  2.01 Gbits/sec    0    212 KBytes
[  5]   9.00-10.00  sec   239 MBytes  2.01 Gbits/sec    0    212 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.34 GBytes  2.01 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  2.33 GBytes  2.01 Gbits/sec                  receiver
```
can we improve this? it seems he only use 2 GBit/s.
A matrix for compatibility would be really handy.

bb-qq, 2021-07-15:
Hmm, the signal quality may be degraded due to the C to A conversion adapter included in ASUSTOR AS-U2.5G. I'm interested to see what happens when you connect the dongle via a USB hub. 

> can we improve this? it seems he only use 2 GBit/s.

You may have already tried this, but increasing the MTU to 9k will improve the throughput somewhat.

Chrno21, 2021-07-15:
> Hmm, the signal quality may be degraded due to the C to A conversion adapter included in ASUSTOR AS-U2.5G. I'm interested to see what happens when you connect the dongle via a USB hub.
I can test it with my Lenovo P52 directly.
> You may have already tried this, but increasing the MTU to 9k will improve the throughput somewhat.
No, good point.

PMu1, 2021-07-29:
I have just tried the same Asustor AS-U2.5G2 on my DS720+ and I get the same as you: only connected to USB2 speed. 

I can confirm that it's the USC-C to USB-A adapter that limits it to 480 MBit/s (tested with and without the adapter on my laptop: with adapter -> 480 MBit/s, without adapter -> 5000 MBit/s).

bb-qq, 2021-08-22:
Thank you for the information. It is very helpful.  I update the readme to describe this.
I'm closing this issue. Please feel free to re-open the issue If there is anything else I should be doing,

closed 2021-08-22 by bb-qq

Akoi89, 2021-09-24:
> I have just tried the same Asustor AS-U2.5G2 on my DS720+ and I get the same as you: only connected to USB2 speed.
> 
> I can confirm that it's the USC-C to USB-A adapter that limits it to 480 MBit/s (tested with and without the adapter on my laptop: with adapter -> 480 MBit/s, without adapter -> 5000 MBit/s).
> 
> The issue is that the adapter seems to be directional: there's a mark on one side of the USB-C connector that must be on the same side as the USB logo on the adapter. Once connected thus, the NIC is back to USB3 speeds.

How's the AS-U2.5G2 working for you? I am eyeing this adapter since it uses the RTL8156B controller and it seems to be one of the few that use it.

PMu1, 2021-09-24:
> How's the AS-U2.5G2 working for you? I am eyeing this adapter since it uses the RTL8156B controller and it seems to be one of the few that use it.

It's been running flawlessly for a month now and no issue with the speed.