bb-qq/r8152 #132: DSM 7.0 DS220J: Package don't start, manually stopped.

techsolo12, 2021-07-16:
### Description of the problem

I installed the DSM 7.0 version of the driver and failed.
Then i tried `install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su` after this, i can do the installation successfully.
But the package will stop from it self and i got the error: Manually stopped.
Than i started the package manually and after 5 seconds i got the same error.

### Description of your products

Synology DS220j
Linux node-backup 4.4.180+ #41890 SMP Fri Jun 25 02:40:26 CST 2021 aarch64 GNU/Linux synology_rtd1296_ds220j
DSM 7.0
TechRise USB3.0 Ethernet Adapter RTL8153 [https://smile.amazon.de/gp/product/B08JC818CV/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1](url)

### Description of your environment

Rear Port of Synology DS220J
Ubiquit UniFi USW Pro Switch
PC: Realtek® 8111G Gigabit LAN controller
Cat.6a

### Output of ``sudo cat /var/log/packages/r8152.log``

````
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin preinst
2021/07/16 12:21:08     install r8152 2.15.0-2 End preinst ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/mkdir -p /volume1/@appstore/r8152
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/mkdir -p /volume1/@appstore/r8152 ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/rm -rf /volume1/@appstore/r8152
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/rm -rf /volume1/@appstore/r8152 ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/package /volu                                                                                                                        me1/@appstore/r8152
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/package /volume                                                                                                                        1/@appstore/r8152 ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/rm -rf /var/packages/r8152
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/rm -rf /var/packages/r8152 ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/mkdir -p /var/packages/r8152
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/mkdir -p /var/packages/r8152 ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/touch /var/packages/r8152/installing
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/touch /var/packages/r8152/installing ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/INFO /var/pac                                                                                                                        kages/r8152/INFO
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/INFO /var/packa                                                                                                                        ges/r8152/INFO ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/rm -rf /var/packages/r8152/scripts
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/rm -rf /var/packages/r8152/scripts ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/scripts /var/                                                                                                                        packages/r8152/scripts
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/scripts /var/pa                                                                                                                        ckages/r8152/scripts ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/rm -rf /var/packages/r8152/conf
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/rm -rf /var/packages/r8152/conf ret=[0]
2021/07/16 12:21:08     install r8152 2.15.0-2 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/conf /var/pac                                                                                                                        kages/r8152/conf
2021/07/16 12:21:08     install r8152 2.15.0-2 End /bin/mv -f /volume1/@tmp/pkginstall/extract.WrlhmX/conf /var/packa                                                                                                                        ges/r8152/conf ret=[0]
2021/07/16 12:21:09     install r8152 2.15.0-2 Begin postinst
/var/packages/r8152/scripts/postinst: line 11: /opt/sbin/spk_su: cannot execute binary file: Exec format error
2021/07/16 12:21:09     install r8152 2.15.0-2 End postinst ret=[0]
2021/07/16 12:21:48     start r8152 2.15.0-2 Begin start-stop-status start
insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
2021/07/16 12:21:58     start r8152 2.15.0-2 End start-stop-status start ret=[0]
2021/07/16 12:25:17     start r8152 2.15.0-2 Begin start-stop-status stop
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
rmmod: ERROR: Module r8152 is not currently loaded
2021/07/16 12:25:17     start r8152 2.15.0-2 End start-stop-status stop ret=[0]
2021/07/16 12:25:19     start r8152 2.15.0-2 Begin start-stop-status start
insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
2021/07/16 12:25:29     start r8152 2.15.0-2 End start-stop-status start ret=[0]
2021/07/16 12:28:18     start r8152 2.15.0-2 Begin start-stop-status stop
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
rmmod: ERROR: Module r8152 is not currently loaded
2021/07/16 12:28:19     start r8152 2.15.0-2 End start-stop-status stop ret=[0]
2021/07/16 12:28:20     start r8152 2.15.0-2 Begin start-stop-status start
insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
[... 712 lines cut ...]
          inet addr:10.10.5.8  Bcast:10.10.5.15  Mask:255.255.255.240
          inet6 addr: fe80::211:32ff:feea:33af/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:37254 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28585 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:23293644 (22.2 MiB)  TX bytes:16474439 (15.7 MiB)
          Interrupt:11

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5834 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5834 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:808219 (789.2 KiB)  TX bytes:808219 (789.2 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2021-07-18:
Thank you for the report.
[The new release r8152-rtd1296-2.15.0-3.spk](https://github.com/bb-qq/r8152/releases/download/2.15.0-3/r8152-rtd1296-2.15.0-3.spk) should fix this issue.

techsolo12, 2021-07-18:
Thank you for your reply.
But i got the same error. I hope you can investiged a second time.

````
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin preinst
2021/07/18 09:53:07     install r8152 2.15.0-3 End preinst ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/mkdir -p /volume1/@appstore/r8152
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/mkdir -p /volume1/@appstore/r8152 ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/rm -rf /volume1/@appstore/r8152
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/rm -rf /volume1/@appstore/r8152 ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/package /volume1/@appstore/r8152
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/package /volume1/@appstore/r8152 ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/rm -rf /var/packages/r8152
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/rm -rf /var/packages/r8152 ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/mkdir -p /var/packages/r8152
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/mkdir -p /var/packages/r8152 ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/touch /var/packages/r8152/installing
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/touch /var/packages/r8152/installing ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/INFO /var/packages/r8152/INFO
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/INFO /var/packages/r8152/INFO ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/rm -rf /var/packages/r8152/scripts
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/rm -rf /var/packages/r8152/scripts ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/scripts /var/packages/r8152/scripts
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/scripts /var/packages/r8152/scripts ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/rm -rf /var/packages/r8152/conf
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/rm -rf /var/packages/r8152/conf ret=[0]
2021/07/18 09:53:07     install r8152 2.15.0-3 Begin /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/conf /var/packages/r8152/conf
2021/07/18 09:53:07     install r8152 2.15.0-3 End /bin/mv -f /volume1/@tmp/pkginstall/extract.oEpwQS/conf /var/packages/r8152/conf ret=[0]
2021/07/18 09:53:08     install r8152 2.15.0-3 Begin postinst
/var/packages/r8152/scripts/postinst: line 11: /opt/sbin/spk_su: cannot execute binary file: Exec format error
2021/07/18 09:53:08     install r8152 2.15.0-3 End postinst ret=[0]
2021/07/18 09:53:08     install r8152 2.15.0-3 Begin start-stop-status start
insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
2021/07/18 09:53:23     install r8152 2.15.0-3 End start-stop-status start ret=[0]
2021/07/18 09:57:14     (system) trigger r8152 2.15.0-3 Begin start-stop-status stop
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
rmmod: ERROR: Module r8152 is not currently loaded
2021/07/18 09:57:14     (system) trigger r8152 2.15.0-3 End start-stop-status stop ret=[0]
2021/07/18 09:59:06     (system) trigger r8152 2.15.0-3 Begin start-stop-status start
insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
2021/07/18 09:59:21     (system) trigger r8152 2.15.0-3 End start-stop-status start ret=[0]
2021/07/18 10:00:00     start r8152 2.15.0-3 Begin start-stop-status stop
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
rmmod: ERROR: Module r8152 is not currently loaded
2021/07/18 10:00:00     start r8152 2.15.0-3 End start-stop-status stop ret=[0]
2021/07/18 10:00:02     start r8152 2.15.0-3 Begin start-stop-status start
insmod: ERROR: could not insert module /volume1/@appstore/r8152/r8152/r8152.ko: Operation not permitted
ls: cannot access '/sys/class/net/sit0/device/driver': No such file or directory
2021/07/18 10:00:17     start r8152 2.15.0-3 End start-stop-status start ret=[0]
````

bb-qq, 2021-07-18:
Did you run `install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su` again?

techsolo12, 2021-07-19:
At the second installation i don't used this command, because it wasn't nessesary. The installation wasn't failed or something else.
I think you must use this command only one time.
But i have tested it after reed your comment and it changed nothing. I got the same error like before. The error log is also the same.

Sjekke, 2021-07-19:
> At the second installation i don't used this command, because it wasn't nessesary. The installation wasn't failed or something else.
>
> I think you must use this command only one time.
>
> But i have [...]

I used the command when doing the upgrade. 

I would try this:

Remove any driver for the USB adapter
Reboot (don't forget to re-￼connect the existing 1Gb ethernet
Re-install the latest driver from bb-qq
Get the error, execute install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su and re-install the driver
Add the (root) start-up script 
	ifconfig eth2 up
	synonet –dhcp eth2
Reboot
Does the adapter comes on? Blinking LED.
	No? Describe your issue
	Yes. Nice. Configure your adapter for your needs.
Reboot and during reboot, disconnect your 1Gb adapter
🤞

bb-qq, 2021-07-20:
```
/var/packages/r8152/scripts/postinst: line 11: /opt/sbin/spk_su: cannot execute binary file: Exec format error
```

Since I fixed the different spk_su binaries was packaged, I wanted you to try the following steps.

* Install r8152-rtd1296-2.15.0-3.spk (to replace `/var/packages/r8152/target/r8152/spk_su`)
* The installation will fail, then run `install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su`
* Reinstall r8152-rtd1296-2.15.0-3.spk 

If this still doesn't work, you may want to uninstall the package once.

techsolo12, 2021-07-20:
Hello,

i'am sorry for the delay, i'am actually in vacation.
I have already uninstall the old package and remove the package at the second installation.
Today i tried it a third time and it works, but i did it on the same way like before.

1. Remove the old package
2. Clean trash from old package
3. Restart the Synology NAS
4. Manuall install the new package r8152-rtd1296-2.15.0-3.spk
5. Get NO installation failed error (i didn't know why? I did it only at the first installation)
6. Now the package are stable and don't stop from it self.
7. Configure the ethernet interface from WebUI.

I have a little improvement for the issue template. We should add the output from ````sudo cat /var/log/packages/r8152.log````

Thank you a lot guys!

closed 2021-07-20 by techsolo12

Sjekke, 2021-07-20:
> Hello,
>
>
>
> i'am sorry for the delay, i'am actually in vacation.
>
> I have already uninstall the old package and remove the package at the second installation.
>
> Today i tried it a third time and it [...]

Top 🤙