bb-qq/r8152 #135: [HELP wanted] Gathering compatibility/stability reports

bb-qq, 2021-07-20:
I've decided to start an effort to collect reports of my drivers working and create the compatibility matrix of the drivers. Since DSM7 has been released and many users are updating their drivers to the latest, I figure now is a good time to start.

If you can help me create the matrix, **reply to this issue using following the template below**. *For bug reports and troubleshooting questions, please create a separate ticket* to get support.

If you have previously reported, it would be appreciated if you could report it anew again. We would especially like to know how stable it is running in the latest environment.

## Template

> * Model(platform): `e.g. DS918+ (apollolake)`
> * DSM version:  `e.g. 7.0-41890`
> * Driver version:  `e.g. 2.15.0-3`
> * Ethernet device: `e.g. ASUSTOR AS-U2.5G`
> * Working status: `e.g. Stable`
>   * (If you have done any special testing, please write the details here.)
> * Maximum Throughput: `e.g. 2.29 Gbps @ MTU 9000 in iperf3`

### Working status

To determine the working status, please use the following example as a basis. 

* **Not work**
  * Failed to install the driver or the device is not recognized correctly
* **Work**
  * The devices seem to work correctly
  * No major issue with large file transfer(at least 5min)
* **Stable**
  * No major issue over 2 weeks
  * No major issue with large file transfer(at least 30min, speed degradation due to thermal throttling may occur)
* **Unstable**
  * Disconnection or other major issues are observed under long-term use or high load condition

It will help other users if you write detailed information such as what kind of tests you did and what results were observed, and how long the product has been in use.

devedse, 2021-07-22:
Since I've been running into quite some issues with the 5Gbe model https://github.com/bb-qq/aqc111/issues/48 I decided to do some testing with the 2.5Gbe devices. Here's the results.

* Model(platform): `DS415+ (avoton)`
* DSM version:  `e.g. 7.0-41890`
* Driver version:  `e.g. 2.15.0-3`
* Ethernet device: `CLUB 3D CAC-1420 2.5GBIT`
* Working status: `Tbd`

| Threads | iperf3 type | MTU | Reversed? | Speed | Notes |
| -- | -- | -- | -- | -- | -- |
| 1 | native | 1500 | | 1.77 Gbits | |
| 10 | native | 1500 | | 1.83 Gbits | |
| 1 | docker | 1500 | | 943 Mbits | This one sometimes dropped to 0.00 temporary |
| 10 | docker | 1500 | | 1.76 Gbits | |
| 1 | native | 1500 | -R | 1.85 Gbits | -R seems to be more stable |
| 10 | native | 1500 | -R | 1.85 Gbits | -R seems to be more stable |
| 1 | docker | 1500 | -R | 1.86 Gbits | -R seems to be more stable |
| 10 | docker | 1500 | -R | 1.86 Gbits | -R seems to be more stable |
| 10 | native | 9000 | | Broken | See note 1 |
| 10 | native | 9000 | -R | 1.85 Gbits | |

**Notes**
1. After the 9000 MTU iperf test without -R the whole connection dropped. Pings didn't work anymore either.

**Additional comments**

I did a few file transfer tests. The transfer went quite okay at around 150+ megabytes per second. However after a few second I could hear my NAS starting to have a lot of disk activity (and then also slow the transfer down) which is presumably because it only has like 100gb of free space. So I'm most likely running into disk fragmentation issues here.

[image]

hhessel, 2021-08-01:
Model(platform): DS1019+ (apollolake)
DSM version: 7.0-41890
Driver version: 2.15.0-3
Ethernet device: Club 3D CAC-1420
Working status: Stable (no issues)
(If you have done any special testing, please write the details here.)
Maximum Throughput: 
iperf3.exe -c hostname -p 5201 -M 9000
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  2.51 GBytes  2.15 Gbits/sec                  sender
[  4]   0.00-10.00  sec  2.51 GBytes  2.15 Gbits/sec                  receiver

Payn, 2021-08-05:
- Model(platform): **DS416play (braswell)**
- DSM version: **DSM 7.0-41890**
- Driver version: **2.15.0-3**
- Ethernet device: **i-tec USB-C Metal 2.5Gbps Ethernet Adapter (Realtek RTL8156B)**
- Working status: **Stable**
- Maximum Throughput: **2.41 Gbps @ MTU 9000 in iperf3**

PC -> NAS
`[ ID] Interval           Transfer     Bandwidth       Retr`
`[  4]   0.00-10.00  sec  2.81 GBytes  2414 Mbits/sec    0             sender`
`[  4]   0.00-10.00  sec  2.81 GBytes  2414 Mbits/sec                  receiver`
`CPU Utilization: local/sender 17.4% (0.3%u/17.1%s), remote/receiver 68.7% (4.0%u/64.7%s)`

NAS -> PC
`[ ID] Interval           Transfer     Bandwidth       Retr`
`[  4]   0.00-10.00  sec  2.84 GBytes  2439 Mbits/sec    0             sender`
`[  4]   0.00-10.00  sec  2.84 GBytes  2439 Mbits/sec                  receiver`
`CPU Utilization: local/receiver 36.6% (0.3%u/36.3%s), remote/sender 48.1% (0.4%u/47.7%s)`

**Notes**
- USB adapter needs to be plugged in the front port, both back ports are dropping connection all the time.
- For maximum file transfer speeds both PC and NAS need to have the same MTU value set to 9000 and SMB version set to **SMB2 and Large MTU** or **SMB3**.

Magstick365, 2021-08-09:
Model(platform): DS718+
DSM version: DSM 7.0-41890
Driver version: 2.15.0-3
Ethernet device: UGREEN USB3.0 to RJ45 2.5G Ethernet Card (NAS);  UGREEN USB-C to RJ45 2.5G Ethernet Card (iMac 27" 2017)
Working status: Stable
Maximum Throughput: 1.31 Gbps @ MTU 1500 (iMAc) in iperf3; 2.35 Gbps @ MTU 9000 (DS718+) in iperf3 reverse

DS718+ can be set to MTU 9000; iMac has no option for MTU 9000 under USB ethernet 'Hardware' Tap.

[image]

davidgirard, 2021-08-31:
Model(platform): DS220+ (geminilake)
DSM version: 7.0-41890
Driver version: 2.15.0-3
Ethernet device: CableCreation USB 3.0 to 2.5Gbps Ethernet Adapter
Working status: Stable

davidgirard, 2021-08-31:
Model(platform): DS220+ (geminilake)
DSM version: 7.0-41890
Driver version: 2.15.0-3
Ethernet device: ‎ANKMAX UC312G2 (RTL8156B)
Working status: Highly unstable (LAN interface appears/disappears constantly)

BlackBird11, 2021-09-02:
Model(platform): DS918+ (apollolake)
DSM version: 7.0-41890
Driver version: 2.15.0-3
Ethernet device: ‎CableCreation USB 3.0 2,5Gigabit LAN
Working status: stable for a day atm

[image]

jensfischer1515, 2021-09-13:
* Model(platform): DS920+ (geminilake)
* DSM version: 7.0-41890
* Driver version: 2.15.0-3 DSM7
* Ethernet device: CSL - USB 3.0 RJ45 Fast Ethernet Adapter
* Working status: Stable

Used to connect a mirrored port of my managed switch for usage in ntopng

RoniRahman, 2021-09-19:
Model(platform): DS920+ (geminilake)
DSM version: 7.0-41890
Driver version: 2.15.0-3 DSM7
Ethernet device: UGREEN USB C to 2.5G Ethernet Adapter from the suggested adapter list
Working status: Does Not Work.

From lsusb
```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS RS 1500MS2 FW:969.e2 .D USB FW:e2      4B2107P22316  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00122E9613F261)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 000000001)
```
From dmesg I saw this snippet which although im not super familiar with this stuff, it seems like maybe Synology is rejecting the driver kernel module?
```
[   80.300073] 8021q: 802.1Q VLAN Support v1.8
[   80.325180] vfat: synoboot1 mounted, process=syno_bios_log_c
[   80.412855] synoboot1 unmounted, process=syno_bios_log_c
[   80.801014] mii: module verification failed: signature and/or required key missing - tainting kernel
[   80.984269] usbcore: registered new interface driver aqc111
[   82.600285] tun: Universal TUN/TAP device driver, 1.6
[   82.605937] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[   84.241437] fuse init (API version 7.23)
[   84.682858] audit_printk_skb: 162 callbacks suppressed
```

svkreml, 2021-11-21:
Model(platform): _DS1618+ (Denverton)_
DSM version: _DSM 7.0.1-42218_
Driver version: _2.15.0-3_
Ethernet device: _UGREEN USB3.0 to RJ45 2.5G Ethernet Card (NAS) <------> TP-LINK <------> RTL8125 2.5GbE Controller (0x8125)_
Working status: _Stable_ ( ISCSI LUNS for VMs for weeks)
Maximum Throughput:  _2.49 Gbps @ MTU 9000 in iperf3_

```
[root@ovirt-node-2 ~]# iperf3 -c 192.168.51.1 -t 1800
Connecting to host 192.168.51.1, port 5201
[  5] local 192.168.50.201 port 46678 connected to 192.168.51.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   296 MBytes  2.49 Gbits/sec    0    830 KBytes       
[  5]   1.00-2.00   sec   296 MBytes  2.48 Gbits/sec    0   1014 KBytes       
[  5]   2.00-3.00   sec   295 MBytes  2.47 Gbits/sec    0   1014 KBytes       
[  5]   3.00-4.00   sec   290 MBytes  2.43 Gbits/sec   84    717 KBytes       
[  5]   4.00-5.00   sec   296 MBytes  2.49 Gbits/sec    0    856 KBytes       
[  5]   5.00-6.00   sec   295 MBytes  2.47 Gbits/sec    0    856 KBytes       
[  5]   6.00-7.00   sec   290 MBytes  2.43 Gbits/sec   87    708 KBytes       
[  5]   7.00-8.00   sec   295 MBytes  2.47 Gbits/sec    0    769 KBytes       
[  5]   8.00-9.00   sec   295 MBytes  2.47 Gbits/sec    0    821 KBytes 
....
[  5] 502.00-503.00 sec   291 MBytes  2.44 Gbits/sec   97    874 KBytes       
[  5] 503.00-504.00 sec   295 MBytes  2.47 Gbits/sec    0    874 KBytes       
[  5] 504.00-505.00 sec   291 MBytes  2.44 Gbits/sec   68    708 KBytes       
[  5] 505.00-506.00 sec   295 MBytes  2.47 Gbits/sec    0    708 KBytes       
[  5] 506.00-507.00 sec   295 MBytes  2.47 Gbits/sec    0    708 KBytes       
[  5] 507.00-508.00 sec   295 MBytes  2.47 Gbits/sec    0    865 KBytes       
...
[  5] 1795.00-1796.00 sec   291 MBytes  2.44 Gbits/sec   36    594 KBytes       
[  5] 1796.00-1797.00 sec   295 MBytes  2.47 Gbits/sec    0    629 KBytes       
[  5] 1797.00-1798.00 sec   295 MBytes  2.47 Gbits/sec    0    629 KBytes       
[  5] 1798.00-1799.00 sec   295 MBytes  2.47 Gbits/sec    0    629 KBytes       
[  5] 1799.00-1800.00 sec   295 MBytes  2.47 Gbits/sec    0    629 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-1800.00 sec   516 GBytes  2.46 Gbits/sec  30024             sender
[  5]   0.00-1799.96 sec   516 GBytes  2.46 Gbits/sec                  receiver
```

```
[root@ovirt-node-2 ~]# iperf3 -c 192.168.51.1  -R
Connecting to host 192.168.51.1, port 5201
Reverse mode, remote host 192.168.51.1 is sending
[  5] local 192.168.50.201 port 47070 connected to 192.168.51.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   294 MBytes  2.47 Gbits/sec                  
[  5]   1.00-2.00   sec   295 MBytes  2.47 Gbits/sec                  
[  5]   2.00-3.00   sec   293 MBytes  2.46 Gbits/sec                  
[  5]   3.00-4.00   sec   285 MBytes  2.39 Gbits/sec                  
[  5]   4.00-5.00   sec   294 MBytes  2.46 Gbits/sec                  
[  5]   5.00-6.00   sec   295 MBytes  2.47 Gbits/sec                  
[  5]   6.00-7.00   sec   294 MBytes  2.47 Gbits/sec                  
[  5]   7.00-8.00   sec   294 MBytes  2.47 Gbits/sec                  
[  5]   8.00-9.00   sec   294 MBytes  2.47 Gbits/sec                  
[  5]   9.00-10.00  sec   292 MBytes  2.45 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.86 GBytes  2.46 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  2.86 GBytes  2.46 Gbits/sec                  receiver

```

mattjanssen, 2022-01-06:
- Model(platform): DS418play (apollolake)
- DSM version: DSM 7.0.1-42218
- Driver version: 2.15.0-4
- Ethernet device: TL-UE300 TP-LINK USB 3.0 to Gigabit Ethernet Network Adapter
- Working status:  Work

terraphantm, 2022-01-07:
Model(platform): ```DS720+ (geminilake)```
DSM version: ```7.0.1-42218```
Driver version: ```2.15.0-4```
Ethernet device: ```Alxum 2.5 Gigabit Ethernet Adapter, 2.5 Gigabit Network Adapter with RTL8156B```
Working status: ```Working```

```
D:\Downloads\iperf3.10.1_64bit>iperf3 -c 192.168.1.184 --bidir
Connecting to host 192.168.1.184, port 5201
[  5] local 192.168.1.175 port 49627 connected to 192.168.1.184 port 5201
[  7] local 192.168.1.175 port 49628 connected to 192.168.1.184 port 5201
[ ID][Role] Interval           Transfer     Bitrate
[  5][TX-C]   0.00-1.00   sec   282 MBytes  2.37 Gbits/sec
[  7][RX-C]   0.00-1.00   sec   229 MBytes  1.92 Gbits/sec
[  5][TX-C]   1.00-2.00   sec   275 MBytes  2.30 Gbits/sec
[  7][RX-C]   1.00-2.00   sec   259 MBytes  2.17 Gbits/sec
[  5][TX-C]   2.00-3.00   sec   281 MBytes  2.36 Gbits/sec
[  7][RX-C]   2.00-3.00   sec   217 MBytes  1.82 Gbits/sec
[  5][TX-C]   3.00-4.00   sec   279 MBytes  2.34 Gbits/sec
[  7][RX-C]   3.00-4.00   sec   265 MBytes  2.22 Gbits/sec
[  5][TX-C]   4.00-5.00   sec   277 MBytes  2.32 Gbits/sec
[  7][RX-C]   4.00-5.00   sec   279 MBytes  2.34 Gbits/sec
[  5][TX-C]   5.00-6.00   sec   280 MBytes  2.35 Gbits/sec
[  7][RX-C]   5.00-6.00   sec   278 MBytes  2.33 Gbits/sec
[  5][TX-C]   6.00-7.00   sec   278 MBytes  2.33 Gbits/sec
[  7][RX-C]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec
[  5][TX-C]   7.00-8.00   sec   279 MBytes  2.34 Gbits/sec
[  7][RX-C]   7.00-8.00   sec   282 MBytes  2.36 Gbits/sec
[  5][TX-C]   8.00-9.00   sec   281 MBytes  2.36 Gbits/sec
[  7][RX-C]   8.00-9.00   sec   228 MBytes  1.92 Gbits/sec
[  5][TX-C]   9.00-10.00  sec   281 MBytes  2.36 Gbits/sec
[  7][RX-C]   9.00-10.00  sec   262 MBytes  2.20 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  2.73 GBytes  2.34 Gbits/sec                  sender
[  5][TX-C]   0.00-10.00  sec  2.73 GBytes  2.34 Gbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec  2.53 GBytes  2.17 Gbits/sec  1320             sender
[  7][RX-C]   0.00-10.00  sec  2.52 GBytes  2.16 Gbits/sec                  receiver

iperf Done.
```

chmooreck, 2022-01-26:
Model(platform): DS1019+ (apollolake)
DSM version: DSM 7.0.1-42218 Update 2
Driver version: 2.15.0-6_rxbuf-test
Ethernet device: UNITEK USB 3.0 to 2.5G Gigabit Ethernet Adapter
Working status: Works, just installed and bb-qq sorted out memory allocation issue.

Maximum Throughput: e.g. 2.48 Gbps @ MTU 9000 in iperf3

Connection to PC (Intel i225) thru EDIMAX GS-1005BE switch

```
$ iperf3 -c 192.168.2.2
Connecting to host 192.168.2.2, port 5201
[  5] local 192.168.2.6 port 39796 connected to 192.168.2.2 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   296 MBytes  2.48 Gbits/sec    0    551 KBytes       
[  5]   1.00-2.00   sec   295 MBytes  2.47 Gbits/sec    0    612 KBytes       
[  5]   2.00-3.00   sec   295 MBytes  2.47 Gbits/sec    0    612 KBytes       
[  5]   3.00-4.00   sec   295 MBytes  2.48 Gbits/sec    0    612 KBytes       
[  5]   4.00-5.00   sec   295 MBytes  2.48 Gbits/sec    0    612 KBytes       
[  5]   5.00-6.00   sec   294 MBytes  2.47 Gbits/sec    0    612 KBytes       
[  5]   6.00-7.00   sec   296 MBytes  2.48 Gbits/sec    0    612 KBytes       
[  5]   7.00-8.00   sec   292 MBytes  2.45 Gbits/sec    0    612 KBytes       
[  5]   8.00-9.00   sec   295 MBytes  2.48 Gbits/sec    0    612 KBytes       
[  5]   9.00-10.00  sec   295 MBytes  2.47 Gbits/sec    0    612 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.88 GBytes  2.47 Gbits/sec    0             sender
[  5]   0.00-10.08  sec  2.88 GBytes  2.45 Gbits/sec                  receiver

iperf Done.
```

rendalltrinanes, 2022-01-31:
- Model(platform): DS1019+ (apollolake)
- DSM version: Version: 7.0.1-42218 Update 2
- Driver version: 2.15.0-5
- Ethernet device: [ASUS 2.5G Ethernet USB Adapter (USB-C2500)](https://www.asus.com/Networking-IoT-Servers/Wired-Networking/Wired-Adapters/USB-C2500/) | [(US AMZN Link)](https://www.amazon.com/gp/product/B093FB9QWB)
- Working status: Work
### Special Testing
- Tested iperf3 in SynoCli Monitor Tools from SynoCommunity
- Maximum Throughput: 2.15 Gbps @ MTU 1500 in iperf3 (average out of 5 tests)
- Connection layout: Synology Front USB port <-> Asus USB-C2500 <4 inch Cat 6 patch cable> Asus USB-C2500 <-> M1 Mac mini (2020) USB 3 port (base model).

No switch used, just direct connect. Bought 2 of the same usb NIC for each end. Also used a 90º angle USB 3.0 adapter for the NIC plugged into the front USB port of the Synology.

- Blackmagic disk speed test on the USB NICs are showing double that of the 1 gig connection through my basic 1gig TP Link switch connected to the rest of my network. Previously got around 68-76 MB/s so this is a good improvement.

[image]

### iperf3 test results

```
test #1
Connecting to host 192.168.11.10, port 5201
[  5] local 192.168.11.11 port 50398 connected to 192.168.11.10 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   274 MBytes  2.30 Gbits/sec                  
[  5]   1.00-2.00   sec   274 MBytes  2.30 Gbits/sec                  
[  5]   2.00-3.00   sec   257 MBytes  2.15 Gbits/sec                  
[  5]   3.00-4.00   sec   280 MBytes  2.35 Gbits/sec                  
[  5]   4.00-5.00   sec   282 MBytes  2.35 Gbits/sec                  
[  5]   5.00-6.00   sec   262 MBytes  2.20 Gbits/sec                  
[  5]   6.00-7.00   sec   260 MBytes  2.18 Gbits/sec                  
[  5]   7.00-8.01   sec   116 MBytes   968 Mbits/sec                  
[  5]   8.01-9.00   sec   273 MBytes  2.30 Gbits/sec                  
[  5]   9.00-10.00  sec   281 MBytes  2.36 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.50 GBytes  2.15 Gbits/sec                  sender
[  5]   0.00-10.00  sec  2.49 GBytes  2.14 Gbits/sec                  receiver

iperf Done.

test #2
Connecting to host 192.168.11.10, port 5201
[  5] local 192.168.11.11 port 50458 connected to 192.168.11.10 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   249 MBytes  2.08 Gbits/sec                  
[  5]   1.00-2.00   sec   267 MBytes  2.24 Gbits/sec                  
[  5]   2.00-3.00   sec   266 MBytes  2.23 Gbits/sec                  
[  5]   3.00-4.00   sec   264 MBytes  2.22 Gbits/sec                  
[  5]   4.00-5.00   sec   261 MBytes  2.19 Gbits/sec                  
[  5]   5.00-6.00   sec   262 MBytes  2.19 Gbits/sec                  
[  5]   6.00-7.00   sec   264 MBytes  2.22 Gbits/sec                  
[  5]   7.00-8.00   sec   258 MBytes  2.15 Gbits/sec                  
[  5]   8.00-9.00   sec   267 MBytes  2.24 Gbits/sec                  
[  5]   9.00-10.00  sec   247 MBytes  2.07 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.54 GBytes  2.18 Gbits/sec                  sender
[  5]   0.00-10.00  sec  2.54 GBytes  2.18 Gbits/sec                  receiver

iperf Done.

test #3
Connecting to host 192.168.11.10, port 5201
[  5] local 192.168.11.11 port 50467 connected to 192.168.11.10 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.01   sec   141 MBytes  1.18 Gbits/sec                  
[  5]   1.01-2.00   sec   248 MBytes  2.09 Gbits/sec                  
[  5]   2.00-3.00   sec   272 MBytes  2.28 Gbits/sec                  
[  5]   3.00-4.00   sec   241 MBytes  2.02 Gbits/sec                  
[  5]   4.00-5.00   sec   274 MBytes  2.29 Gbits/sec                  
[... 32 lines cut ...]

test #5
Connecting to host 192.168.11.10, port 5201
[  5] local 192.168.11.11 port 50687 connected to 192.168.11.10 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   268 MBytes  2.24 Gbits/sec                  
[  5]   1.00-2.00   sec   271 MBytes  2.28 Gbits/sec                  
[  5]   2.00-3.00   sec   265 MBytes  2.22 Gbits/sec                  
[  5]   3.00-4.00   sec   275 MBytes  2.31 Gbits/sec                  
[  5]   4.00-5.00   sec   274 MBytes  2.30 Gbits/sec                  
[  5]   5.00-6.00   sec   257 MBytes  2.16 Gbits/sec                  
[  5]   6.00-7.00   sec   260 MBytes  2.18 Gbits/sec                  
[  5]   7.00-8.00   sec   255 MBytes  2.14 Gbits/sec                  
[  5]   8.00-9.00   sec   244 MBytes  2.05 Gbits/sec                  
[  5]   9.00-10.00  sec   246 MBytes  2.07 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.55 GBytes  2.19 Gbits/sec                  sender
[  5]   0.00-10.00  sec  2.55 GBytes  2.19 Gbits/sec                  receiver

iperf Done.
```

- I'm wondering if setting MTU to 9000 on both ends would increase the blackmagic disk speed results but the Asus drivers for the mac aren't letting me set 9000 in the network settings for that NIC. So if anyone knows anything about that or a different NIC or driver I should use, please let me know.
[image]

azbgh, 2022-03-18:
Model(platform): DS720+ (geminilake)
DSM version: 7.0.1-42218 Update 3
Driver version: 2.15.0-5
Ethernet device: [Alxum 2.5 Gigabit Ethernet Adapter](https://www.amazon.com/gp/product/B08YJ8CLMS/)
Working status: Working
Maximum Throughput:  2.36 Gbps @ MTU 1500 in iperf3,  client on PC connected through Ubiquiti Networks UniFi Switch USW-ENTERPRISE-8-POE
 
```
Accepted connection from 192.168.0.13, port 54235
[  5] local 192.168.0.152 port 5201 connected to 192.168.0.13 port 54236
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   277 MBytes  2.32 Gbits/sec
[  5]   1.00-2.00   sec   282 MBytes  2.36 Gbits/sec
[  5]   2.00-3.00   sec   282 MBytes  2.37 Gbits/sec
[  5]   3.00-4.00   sec   282 MBytes  2.37 Gbits/sec
[  5]   4.00-5.00   sec   282 MBytes  2.37 Gbits/sec
[  5]   5.00-6.00   sec   282 MBytes  2.37 Gbits/sec
[  5]   6.00-7.00   sec   282 MBytes  2.37 Gbits/sec
[  5]   7.00-8.00   sec   282 MBytes  2.36 Gbits/sec
[  5]   8.00-9.00   sec   282 MBytes  2.36 Gbits/sec
[  5]   9.00-10.00  sec   282 MBytes  2.37 Gbits/sec
[  5]  10.00-10.00  sec   239 KBytes  2.29 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.75 GBytes  2.36 Gbits/sec                  receiver
```

Robo210, 2022-03-19:
Model(platform): DS1621+ (v1000)
DSM version: 7.0.1-42218 Update 3
Driver version: 2.15.0-5
Ethernet device: UGREEN USB C to 2.5G Ethernet Adapter
Working status: Work
Maximum Throughput: 2.36 Gbits/sec

Connected to the front port via USB-C to USB-A adaptor.
Transferred 3 TB of data without issue.
CrystalDiskMark against an iSCSI device shows double the sequential read speeds (211 MB/s vs. 118 MB/s) and write speeds (172 MB/s vs. 99 MB/s).

### iperf3:
```
-----------------------------------------------------------
Server listening on 12345 (test #1)
-----------------------------------------------------------
Accepted connection from 10.0.1.26, port 57529
[  5] local 10.0.1.97 port 12345 connected to 10.0.1.26 port 57530
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   1.00-2.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   2.00-3.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   3.00-4.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   4.00-5.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   5.00-6.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   7.00-8.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   8.00-9.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   9.00-10.00  sec   280 MBytes  2.35 Gbits/sec
[  5]  10.00-10.00  sec   256 KBytes  2.09 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.73 GBytes  2.35 Gbits/sec                  receiver
-----------------------------------------------------------
Server listening on 12345 (test #2)
-----------------------------------------------------------
Accepted connection from 10.0.1.26, port 57531
[  5] local 10.0.1.97 port 12345 connected to 10.0.1.26 port 57532
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   282 MBytes  2.36 Gbits/sec    0    217 KBytes
[  5]   1.00-2.00   sec   281 MBytes  2.36 Gbits/sec    0    217 KBytes
[  5]   2.00-3.00   sec   281 MBytes  2.35 Gbits/sec    0    217 KBytes
[  5]   3.00-4.00   sec   281 MBytes  2.35 Gbits/sec    0    217 KBytes
[  5]   4.00-5.00   sec   281 MBytes  2.35 Gbits/sec    0    217 KBytes
[  5]   5.00-6.00   sec   281 MBytes  2.36 Gbits/sec    0    217 KBytes
[  5]   6.00-7.00   sec   281 MBytes  2.35 Gbits/sec    0    217 KBytes
[  5]   7.00-8.00   sec   281 MBytes  2.36 Gbits/sec    0    217 KBytes
[  5]   8.00-9.00   sec   281 MBytes  2.36 Gbits/sec    0    217 KBytes
[  5]   9.00-10.00  sec   281 MBytes  2.35 Gbits/sec    0    217 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.74 GBytes  2.36 Gbits/sec    0             sender
```

ipcintron, 2022-04-10:
Model(platform): DS2419+ (denvorton)
DSM version: DSM 7.0.1-42218 Update 3
Driver version: 2.15.0-6
Ethernet device: [Pluggable 2.5G USB A/C Adapter](https://www.amazon.com/gp/product/B084L4JL9K/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1#customerReviews)
Working status: Stable
Maximum Throughput: 2.35 Gbps @ MTU 1500 in iperf3

Tested from another device using the same adapter through a [2.5G TRENDnet switch](https://www.amazon.com/gp/product/B08XWK4HNT/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

```
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.86.35, port 45910
[  5] local 192.168.86.31 port 5201 connected to 192.168.86.35 port 45912
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   280 MBytes  2.35 Gbits/sec                  
[  5]   1.00-2.00   sec   281 MBytes  2.35 Gbits/sec                  
[  5]   2.00-3.00   sec   281 MBytes  2.35 Gbits/sec                  
[  5]   3.00-4.00   sec   281 MBytes  2.35 Gbits/sec                  
[  5]   4.00-5.00   sec   281 MBytes  2.35 Gbits/sec                  
[  5]   5.00-6.00   sec   280 MBytes  2.35 Gbits/sec                  
[  5]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec                  
[  5]   7.00-8.00   sec   280 MBytes  2.35 Gbits/sec                  
[  5]   8.00-9.00   sec   281 MBytes  2.35 Gbits/sec                  
[  5]   9.00-10.00  sec   281 MBytes  2.35 Gbits/sec                  
[  5]  10.00-10.00  sec   482 KBytes  2.17 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  2.74 GBytes  2.35 Gbits/sec                  receiver
```

Inndy, 2022-05-04:
Model(platform): DS1821+ (v1000)
DSM version: DSM 7.1-42661 Update 1
Driver version: 2.15.0-5
Ethernet device: [Kamera KA-UA2.5G](https://www.kamera.com.tw/products/cbpkammidpl001) (It's in Chinese, sorry that I can't find English version)
Working status: Work (Have not test for long enough time)
Maximum Throughput: 2.28 Gbps @ MTU 1500 in iperf3

```
Connecting to host 10.0.0.6, port 2323
[  4] local 10.0.0.10 port 53337 connected to 10.0.0.6 port 2323
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   255 MBytes  2.14 Gbits/sec
[  4]   1.00-2.00   sec   248 MBytes  2.08 Gbits/sec
[  4]   2.00-3.00   sec   268 MBytes  2.25 Gbits/sec
[  4]   3.00-4.00   sec   270 MBytes  2.27 Gbits/sec
[  4]   4.00-5.00   sec   260 MBytes  2.18 Gbits/sec
[  4]   5.00-6.00   sec   259 MBytes  2.17 Gbits/sec
[  4]   6.00-7.00   sec   270 MBytes  2.26 Gbits/sec
[  4]   7.00-8.00   sec   271 MBytes  2.27 Gbits/sec
[  4]   8.00-9.00   sec   261 MBytes  2.19 Gbits/sec
[  4]   9.00-10.00  sec   272 MBytes  2.28 Gbits/sec
[  4]  10.00-11.00  sec   265 MBytes  2.22 Gbits/sec
[  4]  11.00-12.00  sec   234 MBytes  1.96 Gbits/sec
[  4]  12.00-13.00  sec   269 MBytes  2.26 Gbits/sec
[  4]  13.00-14.00  sec   273 MBytes  2.29 Gbits/sec
[  4]  14.00-15.00  sec   252 MBytes  2.12 Gbits/sec
[  4]  15.00-16.00  sec   263 MBytes  2.20 Gbits/sec
[  4]  16.00-17.00  sec   270 MBytes  2.26 Gbits/sec
[  4]  17.00-18.00  sec   264 MBytes  2.21 Gbits/sec
[  4]  18.00-19.00  sec   263 MBytes  2.21 Gbits/sec
[  4]  19.00-20.00  sec   274 MBytes  2.29 Gbits/sec
[  4]  20.00-21.00  sec   270 MBytes  2.27 Gbits/sec
[  4]  21.00-22.00  sec   231 MBytes  1.94 Gbits/sec
[  4]  22.00-23.00  sec   273 MBytes  2.29 Gbits/sec
[  4]  23.00-24.00  sec   269 MBytes  2.26 Gbits/sec
[  4]  24.00-25.00  sec   257 MBytes  2.16 Gbits/sec
[  4]  25.00-26.00  sec   269 MBytes  2.26 Gbits/sec
[  4]  26.00-27.00  sec   266 MBytes  2.23 Gbits/sec
[  4]  27.00-28.00  sec   262 MBytes  2.20 Gbits/sec
[  4]  28.00-29.00  sec   273 MBytes  2.29 Gbits/sec
[  4]  29.00-30.00  sec   266 MBytes  2.23 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec                  sender
[  4]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec                  receiver
```

TheS1R, 2022-05-10:
Model(platform): **DS718+** (apollolake)
DSM version: **7.1-42661 Update 1**
Driver version: **2.15.0-7**
Ethernet device: **Cable Matters USB to 2.5G Ethernet Adapter (P/N 202095)**
Working status: **Work** (_needs additional testing, but appears to be stable_)
iperf3 installed using "**sudo synogear install**" on each of two Synology DS718+ NASs via **ssh**
Maximum Throughput: **2.46 Gbps** @ MTU 9000 in iperf3 from NAS-to-NAS (_DS718+ (2) client to DS718+ (1) server_); both DS718+ NASs directly connected to 2.5 Gbps switch on 2.5 Gbps network

```
# iperf3 -c 192.168.222.5
Connecting to host 192.168.222.5, port 5201
[  4] local 192.168.222.251 port 38698 connected to 192.168.222.5 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec   291 MBytes  2.44 Gbits/sec    0    830 KBytes       
[  4]   1.00-2.00   sec   290 MBytes  2.44 Gbits/sec    0   1.24 MBytes       
[  4]   2.00-3.00   sec   291 MBytes  2.44 Gbits/sec    0   1.30 MBytes       
[  4]   3.00-4.00   sec   294 MBytes  2.47 Gbits/sec    0   1.30 MBytes       
[  4]   4.00-5.00   sec   295 MBytes  2.47 Gbits/sec    0   1.37 MBytes       
[  4]   5.00-6.00   sec   295 MBytes  2.47 Gbits/sec    0   1.37 MBytes       
[  4]   6.00-7.00   sec   295 MBytes  2.47 Gbits/sec    0   1.37 MBytes       
[  4]   7.00-8.00   sec   292 MBytes  2.45 Gbits/sec    0   1.98 MBytes       
[  4]   8.00-9.00   sec   295 MBytes  2.47 Gbits/sec    0   1.98 MBytes       
[  4]   9.00-10.00  sec   295 MBytes  2.47 Gbits/sec    0   1.98 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  2.87 GBytes  2.46 Gbits/sec    0             sender
[  4]   0.00-10.00  sec  2.86 GBytes  2.46 Gbits/sec                  receiver
```

MatthieuMaindrou, 2022-05-14:
Model(platform): e.g. DS920+ (geminilake)
        DSM version: DSM 7.0.1-42218 Update 3
        Driver version: 2.15.0-7 DSM7.x 
        Ethernet device: DUBE250 2.5GbE Adapter - D-Link Corp.
        Working status: work
          
        Maximum Throughput: 2.29 Gbps @ MTU 9000 (Transmitter in iperf3)
        Maximum Throughput: 2.17 Gbps @ MTU 9000 (Receiver in iperf3)

inameemiri, 2022-06-14:
Model(platform): DS718+ (apollolake)
DSM version: DSM 7.1-42661 Update 1
Driver version: 2.15.0-7
Ethernet device: UGREEN USB A to 2.5G Ethernet Adapter & UNITEK USB 3.0 to 2.5G Gigabit Ethernet Adapter
Working status: Stable (Both Adapter above)
[image]

romanpeters, 2022-06-26:
Model(platform): RS1221+ (V1000)
DSM version: DSM 7.1-42661 Update 1 
Driver version: 2.15.0-6
Ethernet device: [USB-A 2.5G Realtek RTL8156 chipset](https://nl.aliexpress.com/item/1005003392518774.html)
Working status: Work
Maximum Throughput: 2.36 Gbps @ MTU 9000 in iperf3

Would've used the driver v 2.15.0-7, but it didn't have a package for V1000 at the time of writing.
I don't have a multi-gig switch, so I used two ethernet adapters to create a direct link between my server and my NAS, works great.
I'll update in 2 weeks whether it deserves the stable status, but it's looking really good so far.

brentb636, 2022-07-15:
> 

Template

        Model(platform):  DS1520+ (GeminiLake)
        DSM version:  7.1.42661 update 3
        Driver version:  2.15.0-7
        Ethernet device: Cable Creations CD0673
        Working status: e.g. Stable

brentb636, 2022-07-15:
> I've decided to start an effort to collect reports of my drivers working and create the compatibility matrix of the drivers. Since DSM7 has been released and many users are updating their drivers to [...]

Template

        Model(platform): e.g. DS1019+ (apollolake)
        DSM version: e.g. 7.1-42661 update 3
        Driver version: e.g. 2.15.0-7
        Ethernet device: Cable Creation CD0673
        Working status: Stable
            (If you have done any special testing, please write the details here.)
        Maximum Throughput:  2.15 Gbps @ MTU 9000 in large file transfers -Resource Monitor

zack709, 2022-07-19:
Model(platform): DS220+ (GeminiLake)
DSM version: 7.1-42661 Update 3
Driver version: 2.15.0-7
Ethernet device: JUPLINK 2.5G USB Ethernet
Working status: Stable
Maximum Throughput: e.g. 2.48 Gbps @ MTU 9000 in iperf3

Sjekke, 2022-07-19:
> Model(platform): DS220+ (GeminiLake)
> 
> DSM version: 7.1-42661 Update 3
> 
> Driver version: 2.15.0-7
> 
> Ethernet device: JUPLINK 2.5G USB Ethernet
> 
> Working status: Stable
> 
> Maximum Throughput: e.g. 2.48 Gbps @ MTU 9000 in iperf3

Where did you buy that Juplink? On there website it's saying "coming soon"?

zack709, 2022-07-19:
> > Model(platform): DS220+ (GeminiLake)
> > DSM version: 7.1-42661 Update 3
> > Driver version: 2.15.0-7
> > Ethernet device: JUPLINK 2.5G USB Ethernet
> > Working status: Stable
> > Maximum Throughput: e.g. 2.48 Gbps @ MTU 9000 in iperf3
> 
> Where did you buy that Juplink? On there website it's saying "coming soon"?

I come from Taiwan
I buy it from Shopee
buy you can buy it from this website:
[https://www.juplink.com/products/2-5g-usb-3-0-ethernet-network-adapter](https://www.juplink.com/products/2-5g-usb-3-0-ethernet-network-adapter)

stilltli, 2022-08-07:
Model(platform): DS1821+ (v1000)
DSM version: DSM 7.1-42661 Update 3
Driver version: 2.15.0-9
Ethernet device: XTVTX USB LAN Adapter
Working status: Work
As other reports sugested, when you plug in the adapter, while the NAS is running, it only works at the front port. But after a reboot it also works on the 3 ports on the back of the diskstation.
using it since one week and copied +10TB over the connection, works flawless. until now no problems.

Maximum Throughput: 2.5Gbps

bdd, 2022-08-20:
* Model(platform): e.g. DS918+ (apollolake)
* DSM version: 7.1-42661 Update 4
* Driver version: 2.15.0-10
* Ethernet device: [D-Link DUB-E250](https://www.amazon.com/dp/B08NK8SF5F)
* Working status: Work

Maximum Throughput: 2.35 Gbps @ MTU 1500 in iperf3.
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-3600.00 sec   986 GBytes  2.35 Gbits/sec  95637             sender
[  5]   0.00-3600.00 sec   986 GBytes  2.35 Gbits/sec                  receiver
```

Unlike the other USB adapter I tried, this one doesn't seem to get warm at all. After the 1 hour long line rate test above, measure with a thermocouple, adapters warmest spot was 28℃ in a room with 25℃ ambient.

No RX/TX errors, misses, collisions.
```
$ ethtool -S eth2
NIC statistics:
     tx_packets: 78801096
     rx_packets: 739216558
     tx_errors: 0
     rx_errors: 0
     rx_missed: 0
     align_errors: 0
     tx_single_collisions: 0
     tx_multi_collisions: 0
     rx_unicast: 739191024
     rx_broadcast: 8112
     rx_multicast: 17422
     tx_aborted: 0
     tx_underrun: 0
```

This adapter have all hallmarks of eventually proving to be stable but it was attached to the NAS for about 11 hours so far. Will try update this comment in 2 weeks.

war59312, 2022-08-25:
Model(platform): DS1815+ (avaton)
DSM version: DSM 6.2.4-25556 Update 6
Driver version: 2.15.0-1
Ethernet device: Club 3D CAC-1420 @ https://smile.amazon.com/gp/product/B07Q626XK2/
USB Powered Hub: SABRENT 4-Port USB 3.0 Hub@ https://smile.amazon.com/gp/product/B00TPMEOYM/
MTU: 1,500 (Jumbo Frame works up to 9,000 but is slower)
Working status: Stable

See: https://github.com/bb-qq/r8152/issues/43

The USB Ports on the DS1815+ do not have enough power on their own. Must use a USB Powered HUB.

mph56, 2022-08-29:
Model(platform): DS920+ (geminilake)
DSM version: 7.1.1-42951
Driver version:  2.13-3-1
Ethernet device: Club 2.5G 3D CAC-1420 Type A. Also tried CableCreation 2.5G Type A adapter
Working status: Unstable

Both adapters functioned identically. Tried adapters on both front and back ports. Assigned a fixed IP address.
Appeared to work initially. Measured transfer rates of about 2.3G using ipsec3.  Rsysnced  about 50 GB of 4GM files and transfer speed started out fast but showed down to the same speed as the built-in 1G adapters.
Using WSL2 on Windows, rsynced 991.48 GB of data 220,600 files in 2 hrs 28 minutes with the Club 3D adapter. The same data took 2 hrs 37 minutes to rsync using the built-in 1G adapter.

Adapter stopped working when I disconnected the cable and reconnected it. Would not work until I stopped and restarted the driver. Eventually stopped working completely and could not be restarted. Synology Assistant then showed the adapter with "Connection failed / Ready" and I could not ping the IP address or SSH into the DS902+.

 On one occasion, I lost completely network connectivity with the 1G  port on the DS920+ and was completely locked out. I had to reset the network settings.

CarlEdman, 2022-09-20:
Model(platform): DS1019+ (apollolake)
DSM version: DSM 7.1-42661 Update 4
Driver version: r8152-apollolake-2.16.3-1
Ethernet device: CableCreation 2.5G Ethernet to USB Adapter](https://smile.amazon.com/dp/B07VNFLTLD) Currently $30.99
Working status: Work
(To come)
Maximum Throughput: (To come) @ MTU 9000 in iperf3

andrus2049, 2022-11-20:
After the success with my DS218 I tested also my older DS218play, which differs only for the RAM that in the DS218play is only 1 GB.

Synology DS218play   ---> OK!
Realtek RTD1296 SoC
DSM 7.1.1-42962 Update 2

USB model:
Digitus USB Type-C™ Gigabit Ethernet Adapter 2.5G, USB-C™ + USB A (USB3.1/3.0)
https://www.digitec.ch/de/s1/product/digitus-usb-type-c-gigabit-ethernet-adapter-25g-usb-c-usb-a-usb3130-usb-c-usb-31-netzwerkadapter-16185124

USB 3 port used: on the back
Working reliably, no interruptions during transfer of large GB files.
Speed to the DS218 from a NUC Intel with 2.5G Ethernet Adapter: about 140-150 MBs.

Details:

With driver: 2.16.3-2 DSM7.x
[r8152-rtd1296-2.16.3-2.spk](https://github.com/bb-qq/r8152/releases/download/2.16.3-2/r8152-rtd1296-2.16.3-2.spk)
Assigned static IP in GUI is correctly working after reboot.

**Many thanks to [bb-qq](https://github.com/bb-qq)!!!**

[image]

redhook62, 2022-11-21:
**Synology DS420+** 
- Stable 
- iPerf3 -> 2.38gb
- Problems -> No

**DSM version**
- 7.1.1-42962 Update 1
- rear usb3 port

**Driver Version**
- 2.16.3-2 DSM7.x

**Network Card**
- Asus USBC-2500

Everything works as well.

Many thanks @bb-qq

DaveVoyles, 2022-11-30:
**Model(platform):** DS920+ (geminilake)
**DSM version:** DSM 7.1.1-42962 Update 2
**Driver version:** r8152-geminilake-2.16.3-3.spk
**Ethernet device:** ASUSTOR AS-U2.5G
**Working status:** Stable

Will test max throughput  to internet shortly. 
Maximum Throughput from in-network device: 45 MBs @ MTU default (1500) in iperf3

CarlEdman, 2022-11-30:
> Maximum Throughput from in-network device: 45 MBs @ MTU default (1500) in iperf3

Are you sure that number is right?  Even regular gigabit ethernet can reach about 110 MBytes/second and with a 2.5 adapter you should be able to see 250 MBytes/second+.

If 45 MBytes/second is correct, then something else (the disk perhaps?) is the bottleneck and a faster ethernet port won't help.

decay71, 2022-12-04:
Model(platform): DS923+ (R1000)
DSM version: DSM 7.1.1-42962 Update 2
Driver version: r8152-r1000-2.16.3-4test.spk)
Ethernet device: ULT-WIIQ
Working status: Works

Back Port, will do more througput tests the following week..

[image]

decay71, 2022-12-04:
No, i wanted to try the cheaper solution first..

Sjekke, 2022-12-04:
> No, i wanted to try the cheaper solution first..

I can imagine...

ChemEngr79, 2022-12-11:
Model(platform):  DS920+ (geminilake)
DSM version: 7.1.1-42962
Driver version: 2.16.3-3
Ethernet device: Plugable 2.5G USB C and USB to Ethernet Adapter
https://www.amazon.com/dp/B084L4JL9K?psc=1&ref=ppx_yo2ov_dt_b_product_details
Working status: Stable
[image]

I installed this yesterday and have since transferred about 5TB using backup4all without issue. My backup speed increased from an average of about 45MB/s to about 80MB/s. 

I can hit ~280MB/s for direct copying from my workstation SSD drives.

So far, this is a very nice upgrade for $30.

Sjekke, 2022-12-11:
> Model(platform): DS920+ (geminilake) DSM version: 7.1.1-42962 Driver version: 2.16.3-3 Ethernet device: Plugable 2.5G USB C and USB to Ethernet Adapter [...]

I agree with you. I also have the Plugable adapter co-working with my DS916+. I thought I was the only one without any problems ... The problem is that I live in the EU and this adapter is not available anymore and if it's available on Amazon it's veeeeery expensive

decay71, 2022-12-13:
Model(platform): DS923+ (R1000)
DSM version: DSM 7.1.1-42962 Update 2
Driver version: r8152-r1000-2.16.3-4test.spk)
Ethernet device: ULT-WIIQ
Working status: Stable 

For 10 Days no problem... 

[image]

[image]

eggmister, 2022-12-23:
Model(platform): DS1515+ (Avoton)
DSM version: DSM 7.1.1-42962 Update 1
Driver version: 2.16.3-3
Ethernet device: Pro Signal PSG91497
Working status: Stable (only when Virtual Machine Manager is installed and the NAS is configured using Open vSwitch, otherwise it is unstable and crashes every day)

kainino0x, 2022-12-30:
Model(platform): `D920+ (geminilake)`
DSM version: `DSM 7.1.1-42962 Update 3`
Driver version: `2.16.3-3`
Ethernet device: `Unitek U1313B` `0bda:8156 REV31.00`
Working status: Stable (so far, have barely used it)
Maximum Throughput: `networkstatic/iperf3` container, direct wire to another machine (Ubuntu w/ Intel I225-V (rev 02))
- 2.47 Gbps w/ MTU 9000 on both ends
- 2.35 Gbps w/ MTU 1500 on either end

ngod88, 2023-01-17:
Model(platform): `DS718+ (apollolake)`
DSM version: `DSM 7.0.1-42216 - Update 6`
Driver version: `2.16.3-4 test`
Ethernet device: `Alxum 2.5 Gigabit RTL8156GB - only rear, front not working`
Working status: `Work` 
Maximum Throughput: cannot get iPerf3, and can't get past 1GBE

Using a QSW-2104-2S

altokarev, 2023-01-25:
* Model(platform): `DS920+ (geminilake)`
* DSM version:  `7.1.1-42962`
* Driver version:  `2.16.3-3`
* Ethernet device: `D-Link DUB-2315`
* Working status: `Stable`
* Maximum Throughput: `2.45 Gbps @ MTU 9000 in iperf3`

nagyrobi, 2023-02-12:
Tested against a PC with Intel Core i5-6600T CPU @ 2.70GHz, 8GB RAM, running Debian kernel 5.15.85, equipped with the same USB3 Realtek RTL8156B, connected with a 50cm patch cord (no switch in between): 

 - Model(platform): DS416play (braswell)
 - DSM version: DSM 7.1.1-42962 Update 3
 - Driver version: 2.16.3-3
 - Ethernet device: USB3 Realtek RTL8156B
 - Working status: Stable
 - Maximum Throughput: 
   - 1.6Gbps as iperf3 client, with CPU usage 15-25%
   - 1.9Gbps as iperf3 server, with CPU usage 50% !!!

Changing to MTU 9000 made slight differences, but CPU usage was still high.

sounds2k, 2023-02-28:
Model(platform): DS218 (RTD1296)
DSM version: 7.1.1-42962 Update 4
Driver version: 2.16.3-3
Ethernet device: Cable Matters USB to 2.5G Ethernet Adapter (https://www.amazon.co.uk/dp/B0856V9FDH)
Working status: Unstable

connection comes up at 2.5G and appears to be working, you can access the GUI and browse network shares fine - however, once you try to copy a large file from the NAS the connection drops out and the NAS becomes inaccessible over that interface.

dlyubimov, 2023-03-07:
Update:  When i configure manual IP config (no dhcp) it sort of works.  The full network initialization does not go through and i still get messages like " synonetdtool.cpp:640 Failed to enable route table for [eth1] in IPV-4" but the IP4 gets configured and works even that LAN LED stays off. However, in synology network status UI the link reports up and at 2500Mbps.

Link seems to be stable via front USB but being USB2, stays at USB2 speeds (~350Mbps). When plugged in the back, the link seems to be running full speed, but fails soon after the bulk transfer.

nagyrobi, 2023-03-08:
Reboot?

AtOnOUs, 2023-03-10:
Model(platform): DS620slim (apollolake)
DSM version: DSM 7.1.1-42962 Update 4
Driver version: 2.16.3-4test
Ethernet device: Wavlink USB 2.5G NWU330GCA (Realtek RTL8156B)
Working status: Stable
Maximum Throughput: 2.37 Gbps in iperf3

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   282 MBytes  2.37 Gbits/sec
[  4]   1.00-2.00   sec   282 MBytes  2.36 Gbits/sec
[  4]   2.00-3.00   sec   282 MBytes  2.37 Gbits/sec
[  4]   3.00-4.00   sec   282 MBytes  2.36 Gbits/sec
[  4]   4.00-5.00   sec   276 MBytes  2.31 Gbits/sec
[  4]   5.00-6.00   sec   283 MBytes  2.37 Gbits/sec
[  4]   6.00-7.00   sec   282 MBytes  2.37 Gbits/sec
[  4]   7.00-8.00   sec   283 MBytes  2.37 Gbits/sec
[  4]   8.00-9.00   sec   279 MBytes  2.34 Gbits/sec
[  4]   9.00-10.00  sec   277 MBytes  2.32 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  2.74 GBytes  2.35 Gbits/sec                  sender
[  4]   0.00-10.00  sec  2.74 GBytes  2.35 Gbits/sec                  receiver

SMB:
282MB/c from NAS to PC
265MB/c from PC to NAS

evanrich, 2023-03-25:
> 18 I tested also my older DS218play, which differs only for the RAM that in the DS218play is only 1 GB.
> 
> Synology DS218play --->

You posted this 2 weeks ago, how has it performed since?  still stable?  Just bought 2 of these nics for my laptop and the DS620Slim, the 5gbps adapters seem to be more problematic, so figure the 2.5 will be a bit easier.

AtOnOUs, 2023-03-25:
Its stable, but I don't have a heavy load on NAS.

midifail, 2023-03-30:
Model(platform): e.g. DS1817+ (apollolake)
DSM version:  DSM 7.1.1-42962
Driver version:  2.1.6.3-3
Ethernet device: CLUB 3D CAC-1420 2.5GBIT
Working status: Stable

USB Adapter installed on the backside of the Nas (middle Port).

Thank you for the awesome work! :-)

JimBol22, 2023-04-22:
Model(platform): DS920+
DSM version: DSM 7.1.1-42962 update 4
Driver version: Apple Mac Ventura (default driver, adapter is recognized and supported)
Ethernet device:  XRR WZ_UWK01 RJ45 LAN Ethernet (2.5 G/s / 2500 M/s) (via Amazon)
Working status: Stable
Maximum Throughput: e.g. 2.4 Gbps @ MTU 9000 in OpenSpeedTest

Gajotres, 2023-05-16:
**Model(platform)**: DS1621+
**DSM version**: DSM 7.1.1
**Driver version**: 2.16.3-4_7.1 (r8152-v1000-2.16.3-4_7.1.spk)
**Ethernet device**: CLUB 3D CAC-1420
**Working status**: Work

blademan007, 2023-05-20:
Model(platform): DS918+ (apollolake)
DSM version: 6.2.4-25556 Update 6
Driver version: 2.16.3-4
Ethernet device: Plugable 2.5G USB C and USB to Ethernet Adapter, SKU: USBC-E2500, Realtek RTL8156B
Working status: Stable. Front USB port
MTU 1500
`$ iperf3 -c 192.168.86.202 -i 1 -t 20 -P 4      
...
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-20.00  sec  1.37 GBytes   589 Mbits/sec    0             sender
[  5]   0.00-20.00  sec  1.37 GBytes   588 Mbits/sec                  receiver
[  7]   0.00-20.00  sec  1.37 GBytes   590 Mbits/sec    0             sender
[  7]   0.00-20.00  sec  1.37 GBytes   590 Mbits/sec                  receiver
[  9]   0.00-20.00  sec  1.41 GBytes   607 Mbits/sec    0             sender
[  9]   0.00-20.00  sec  1.41 GBytes   607 Mbits/sec                  receiver
[ 11]   0.00-20.00  sec  1.37 GBytes   588 Mbits/sec    0             sender
[ 11]   0.00-20.00  sec  1.37 GBytes   587 Mbits/sec                  receiver
[SUM]   0.00-20.00  sec  5.53 GBytes  2.37 Gbits/sec    0             sender
[SUM]   0.00-20.00  sec  5.52 GBytes  2.37 Gbits/sec                  receiver

iperf Done.
`

hy-ts, 2023-05-25:
Model(platform): DS918+ (apollolake)
DSM version: 7.2-64561 (7.2 final)
NAS: 16GB RAM; 4 pieces of Seagate Exos X18 18TB HDD; SHR-1; Btrfs; Unencrypted volume; MTU=9000
Driver version: 2.16.3-4 (r8152-apollolake-2.16.3-4_7.2.spk)
Ethernet device: Club 3D CAC-1420 - Two pieces of them in one NAS
Working status: Stable (since 22-March-2023)
Maximum Throughput: see below

**File copy speed tests with file size of 148368.4 MByte (148.4 GB):**
PC: Windows 11 Pro 22H2 Build 22621.1778 (SMB multichannel working well in both directions)
Ethernet device in PC: IOCREST IO-PCE8125B-4GLAN RTL8125B 4 Port 2.5 GbE Ethernet Card (Using 4 ports with 4 LAN CAT.7 cables)
Important Realtek driver settings for the 2.5 GbE LAN Card ("Realtek Gaming 2.5GbE Family Controller") in PC to get the best speeds:
- Advanced EEE: Disabled
- EEE Max Support Speed: 10 Mbps Full duplex
- Energy-efficient Ethernet: Disabled
- Green Ethernet: Disabled
- Gigabit Lite: Disabled
- Jumbo Frame: 9014 bytes
- Power Saving Mode: Disabled
- Speed and duplex: 2.5 Gbps Full duplex (Not auto)

(If I enable the different power saving modes, I get about 10% lower speeds.)

**NAS-to-PC: 148368.4 MB / 308 s = 481 MB/s
PC-to-NAS: 148368.4 MB / 342 s = 433 MB/s**

war59312, 2023-05-25:
That can't be with 2.5 GbE adapters. They max out at 312.5 MB/s (at that is if perfect 2.5Gbps).

To hit 481 MB/s you would have to have at least 5 GbE adapters. As 481MB/s is 3.85 Gbps.

hy-ts, 2023-05-25:
@war59312 
You're absolutely right. However, I use two 2.5 GbE USB adapters connected to the Synology NAS (at the front and the rear USB ports), which together (roughly) equals one 5 GbE adapter. And my computer has four 2.5 GbE ports (roughly = 10 GbE).

[128 more replies]