bb-qq/r8152 #136: DS218 Speed Drop

Joe-LAI24, 2021-07-23:
NAS Model: DS218
OS: DSM 7.0
Network adapter chipset: RTL8156

When I using SMB to access the NAS with a 2.5Gbps Network adapter, data transfer will be normal for a few seconds. and after that the speed will go down to 0 for few seconds, it will be normal again. and it keeps happening, every few seconds.

I am thinking is it possible the smb cache problem?

But i try connect a slower wifi with my PC, and test,the speed around 30M/s ,but the speed dropping keep happed.

xmiqz, 2021-07-24:
I also met for this problem

bb-qq, 2021-08-22:
To summarize the reports in this forum, it seems that the USB controller of the RTD1296 platform has a flaw that prevents this driver from working stably.

It's odd that it doesn't work with the RealTek hardware, but there doesn't seem to be anything I can do about it at the moment. Sorry.

daveyeger, 2021-08-25:
@bb-qq it would appear based on the stack traces there is a malloc error not being handled in the code. Able to get 1.8gb sustained and then the package crashes. Restarting the package brings it back up but then it's the same behavior.

Joe-LAI24, 2021-08-25:
@bb-qq 
It’s ok, I can see how difficult to make the driver  be compatible on an Embedded system.
Thanks for your amazing jobs and code.

BTW, I test the same adapter on DS220+ and DS920+ ,so far they can run perfect(test restart and plug/unplug). So I think your driver is running well on Gemini base product.

Will also try on my other ARM base machine. DS214 and DS218J

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq