bb-qq/r8152 #137: Why need spk_su.c?

riverscn, 2021-07-24:
I read https://help.synology.com/developer-guide/privilege/privilege_config.html and have a question.

Why not write `conf/privilege` file in the repo, instead of spk_su.c during installation process?

bb-qq, 2021-07-24:
Since DSM 7.0, root privilege is only allowed for packaged created by Synology.
https://help.synology.com/developer-guide/privilege/preface.html

To avoid this limitation, we need to modify `conf/privilege` after the timing when the Package Manager checks its value. And privilege escalation is required to modify `conf/privilege` because post-process of installation is executed as logon user of DSM.

riverscn, 2021-07-25:
TIL!

So it seems there is no official configuration for user package which contains a hardware driver.

bb-qq, 2021-08-22:
Yes, you are right.

closed 2021-08-22 by bb-qq