bb-qq/r8152 #138: DS418J CPU Spike and dropped network connection

rswiz, 2021-07-25:
### Description of the problem

Anytime that I download a file from my NAS, to my local computer the process becomes hung, and eventually stops or I have to stop it manually. Small files are ok (anything under 500MB) , the driver installed on the NAS is running and I am connecting to the IP address of the USB Ethernet adapter. The adapter ID is eth1. ** Now copying from PC to NAS works, but at 100 MB/s.

### Description of your products

> - DS418j
>     Linux family_nas 4.4.180+ #41890 SMP Thu Jul 15 03:42:46 CST 2021 aarch64 GNU/Linux synology_rtd1296_ds418j
>     - DSM version 7.0-41890
> - Club CAC-1420 USB 2.5 GBPE dongle

### Description of your environment

> - connect to a unmanaged switch
>     - PC: Realtek USB 2.5GbE Family Controller
>     - Hub: Qnap QSW-1105-5T-US
> - cable Cat.7 from office, and NAS CAT.6 to QNAP hub

### Output of ``dmesg`` command

````
[ 2848.793672] DMA free:156512kB min:65536kB low:81920kB high:98304kB active_ano                                                               n:122256kB inactive_anon:122780kB active_file:121984kB inactive_file:76252kB une                                                               victable:1120kB isolated(anon):0kB isolated(file):0kB present:1048576kB managed:                                                               1011788kB mlocked:1120kB dirty:8kB writeback:0kB mapped:140340kB shmem:65728kB s                                                               lab_reclaimable:193028kB slab_unreclaimable:111976kB kernel_stack:10912kB pageta                                                               bles:22392kB unstable:0kB bounce:0kB free_pcp:1368kB local_pcp:692kB free_cma:30                                                               964kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[ 2848.841974] lowmem_reserve[]: 0 0 0
[ 2848.845604] DMA: 11214*4kB (UMEHC) 6029*8kB (UMEH) 1445*16kB (UMEHC) 291*32kB                                                                (MHC) 1*64kB (C) 1*128kB (C) 0*256kB 0*512kB 0*1024kB 1*2048kB (C) 7*4096kB (C)                                                                = 156432kB
[ 2848.861258] 70799 total pagecache pages
[ 2848.865187] 4528 pages in swap cache
[ 2848.868864] Swap cache stats: add 176330, delete 171802, find 3499241/3513458
[ 2848.876179] Free swap  = 1669656kB
[ 2848.879675] Total swap = 2097084kB
[ 2848.883159] 262144 pages RAM
[ 2848.886106] 0 pages HighMem/MovableOnly
[ 2848.890057] 9197 pages reserved
[ 2848.893277] 8192 pages cma reserved
[ 2918.985222] r8152 5-1:1.0 eth1: Tx timeout
[ 2918.991729] r8152 5-1:1.0 eth1: Tx status -2
[ 2918.996124] r8152 5-1:1.0 eth1: Tx status -2
[ 2919.000536] r8152 5-1:1.0 eth1: Tx status -2
[ 2919.004936] r8152 5-1:1.0 eth1: Tx status -2
[ 2920.970629] r8152 5-1:1.0 eth1: get_registers -108
[ 2920.975620] r8152 5-1:1.0 eth1: get_registers -71
[ 2920.980497] r8152 5-1:1.0 eth1: get_registers -71
[ 2920.986223] r8152 5-1:1.0 eth1: get_registers -71
[ 2920.991096] r8152 5-1:1.0 eth1: get_registers -71
[ 2920.996395] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.001286] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.006597] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.012371] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.017312] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.022185] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.027067] xhci-hcd xhci-hcd.5.auto: URB transfer length is wrong, xHC issue                                                               ? req. len = 4, act. len = 4294967292
[ 2921.037772] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.042672] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.047826] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.052702] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.059557] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.064474] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.069399] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.074270] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.079199] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.084076] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.089013] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.093919] r8152 5-1:1.0 eth1: get_registers -71
[ 2921.137868] usb 5-1: reset SuperSpeed USB device number 4 using xhci-hcd
rob@family_nas:~$ clear
rob@family_nas:~$ dmesg
[    0.009795] pctrl-rtk: pctrl_l4_icg_jpeg::ENABLE_HW_PM, ret = -1
[    0.009855] pctrl-rtk: pctrl_gpu::POWER_OFF, ret = 0
[    0.009940] pctrl-rtk: pctrl_gpu_core_1::POWER_OFF, ret = 0
[    0.010023] pctrl-rtk: pctrl_gpu_core_2::POWER_OFF, ret = 0
[    0.010107] pctrl-rtk: pctrl_gpu_core_3::POWER_OFF, ret = 0
[... 1798 lines cut ...]
          inet addr:192.168.1.50  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fee1:e346/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1258641 errors:0 dropped:0 overruns:0 frame:0
          TX packets:329376 errors:18 dropped:44 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1824461569 (1.6 GiB)  TX bytes:375386084 (357.9 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:9117 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9117 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2092873 (1.9 MiB)  TX bytes:2092873 (1.9 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

rswiz, 2021-07-25:
[image]

bb-qq, 2021-11-13:
Just comment: This log indicates xHCI host controller hardware has some issue and it seems to be the cause of this problem.

The recent kernel has some mechanisms to avoid this kind of issue called "quirks", but the version Synology uses does not support this, unfortunately.

rswiz, 2021-11-13:
Thanks for checking, I'll wait for an upgrade opportunity.  😉