bb-qq/r8152 #139: ASUS USB-C2500 not working

PMu1, 2021-07-29:
### Description of the problem

Connecting the ASUS USB-C2500 2.5 Gbe NIC using USB-A to the Synology DS720+ and using the 2.15.0-1 release for gemini lake. In Info Center I can see the 2.5 Gbe NIC, but no LAN3 appears in Network Interface tab. . 
I've tried turning off the RTL package and back on, and I've tried rebooting the NAS, but still the NIC does not show up in the Network Interface tab. 

### Description of your products

> - Synology DS720+
>     - Linux BitStreamDepot 4.4.59+ #25556 SMP PREEMPT Thu Mar 18 13:00:34 CST 2021 x86_64 GNU/Linux synology_geminilake_720+
>     - DSM 6.2.4-25556
> - ASUS USB-C2500

### Description of your environment

> - connect directly to a NAS
>     - ASUS SUC-C2500
> - Cat.6

### Output of ``dmesg`` command

````
[17618.131613] usbcore: registered new interface driver r8152
[17641.861715] xhci_hcd 0000:00:15.0: device is plugged out, empty URBs
[17641.885118] usb 2-1: USB disconnect, device number 3
[17641.938711] init: btacd main process (16123) killed by TERM signal
[17642.003348] usbcore: deregistering interface driver btusb
[17647.439061] init: bluetoothd main process (16115) killed by KILL signal
[17647.538395] NET: Unregistered protocol family 31
[17648.751436] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[17649.809176] Bluetooth: Core ver 2.21
[17649.813201] NET: Registered protocol family 31
[17649.818171] Bluetooth: HCI device and connection manager initialized
[17649.825282] Bluetooth: HCI socket layer initialized
[17649.830737] Bluetooth: L2CAP socket layer initialized
[17649.836384] Bluetooth: SCO socket layer initialized
[17649.845245] Bluetooth: RFCOMM TTY layer initialized
[17649.850698] Bluetooth: RFCOMM socket layer initialized
[17649.856445] Bluetooth: RFCOMM ver 1.11
[17649.862501] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[17649.869126] Bluetooth: HIDP socket layer initialized
[17649.883148] usbcore: registered new interface driver btusb
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS NS 650M1 FW:929.a8 .D USB FW:a8  4B2029P35079  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00911E16345A82)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0b05:1976:3105 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 501000001)
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr X
          inet addr:XXX.XXX.X.X  Bcast:XXX.XX.X.X  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr X
          inet addr:XXX.XXX.X.X  Bcast:XXX.XXX.X.X  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:9000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:97 base 0x2000  

eth1      Link encap:Ethernet  HWaddr X
          inet addr:XXX.XXX.X.X  Bcast:XXX.XXX.X.X  Mask:255.255.255.0
          inet6 addr: X
          inet6 addr:X
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:21211 errors:0 dropped:0 overruns:0 frame:0
          TX packets:50777 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5026289 (4.7 MiB)  TX bytes:33959003 (32.3 MiB)
          Interrupt:96 base 0xe000  

lo        Link encap:Local Loopback
          inet addr:XXX.XXX.X.X Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:39792 errors:0 dropped:0 overruns:0 frame:0
          TX packets:39792 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:4889831 (4.6 MiB)  TX bytes:4889831 (4.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

Payn, 2021-08-05:
Can confirm, not working on Synology DS416play with DSM 7.0-41890 either.

rclaycock, 2021-09-10:
Also not working for me. Has anyone found a fix for this?

Thanks, R

Akoi89, 2021-09-24:
Same here, not working, would love to see this device supported since it seems to be a well built unit.

bb-qq, 2021-09-24:
I uploaded [2.15.0-4 release](https://github.com/bb-qq/r8152/releases/tag/2.15.0-4) to support ASUS USB-C2500.
I would appreciate if you report whether it works and its stability.

Akoi89, 2021-09-24:
> I uploaded [2.15.0-4 release](https://github.com/bb-qq/r8152/releases/tag/2.15.0-4) to support ASUS USB-C2500.
> I would appreciate it if you report whether it works and its stability.

I was literally just about to return it to amazon and you got me to give it one last chance. :D 

I just installed it a few mins ago, after plugging it in and rebooting it the device showed up when plugged into the front port of my 920+, testing it out right now by copying over "Halo The Master Chief Collection" (106GB folder) from my desktop which also has a PCI-E Realtek 2.5GB NIC in it.  Doesn't seem to be running hot and its been running for a few minutes (about 7) no crashes, only using MTU 1500 right now but it seems to work well so far.

Will update if I see any odd behavior and I will be testing it by throwing more files on it.

EDIT: the 106GB folder copied to it successfully without any crashes and went way faster than it would have with the built in NIC.

PMu1, 2021-09-24:
> I uploaded [2.15.0-4 release](https://github.com/bb-qq/r8152/releases/tag/2.15.0-4) to support ASUS USB-C2500.
> I would appreciate it if you report whether it works and its stability.

Thanks for the update! 
I've been using another NIC, but will give the Asus another try with your new release.

bb-qq, 2021-11-13:
The work to support the device itself has been completed, so I close this issue.

Sorry, this change is not included past drivers for DSM6.2 and I do not plan to update these drivers due to the difficulty in maintaining the development environment.

closed 2021-11-13 by bb-qq