bb-qq/r8152 #140: Multiple Realtek Adapters?

tevans62, 2021-07-30:
I had so much success with a single 2.5Gbe Cable Creations adapter (thank you!) with DSM 7.0 that I wanted to add a second one for my 9000byte MTU storage VLAN.

However the the 2nd (exact same adapter) results in some weird log messages, and settles on being a " Realtek CDROM".
I did a quick search and didn't see this yer reported, apologies if a dupe:

[601197.231135] usb 3-2: new SuperSpeed USB device number 3 using etxhci_hcd-170202
[601197.257399] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[601197.270420] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[601197.280204] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 3 ep 3 with no TDs queued?
[601197.290019] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 3 ep 6 with no TDs queued?
[601198.403114] usb 3-2: reset SuperSpeed USB device number 3 using etxhci_hcd-170202
[601198.426755] usb 3-2: device firmware changed
[601198.431685] usb 3-2: USB disconnect, device number 3
[601198.456866] r8152 3-2:1.0 (unregistered net_device): get_registers -19
[601198.464287] r8152 3-2:1.0 (unregistered net_device): Get ether addr fail
[601198.472172] r8152 3-2:1.0 eth5: v2.15.0 (2021/04/15)
[601198.477839] r8152 3-2:1.0 eth5: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.

[601198.495584] r8152 3-2:1.0 eth5: chip rev 14
[601210.133423] usb 3-2: new SuperSpeed USB device number 4 using etxhci_hcd-170202
[601210.157559] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[601210.167486] usb-storage 3-2:1.0: USB Mass Storage device detected
[601210.174651] scsi11 : usb-storage 3-2:1.0
[601211.182865] scsi 11:0:0:0: CD-ROM            Realtek  USB CD-ROM               2.00 PQ: 0 ANSI: 0
[601211.193236] scsi 11:0:0:0: Attached scsi generic sg9 type 5
[601266.347510] usb 3-2: USB disconnect, device number 4
[601273.256744] usb 2-3: USB disconnect, device number 2
[601287.521870] usb 3-3: new SuperSpeed USB device number 5 using etxhci_hcd-170202
[601287.548009] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[601287.560908] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[601287.570813] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[601287.580586] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[601288.693726] usb 3-3: reset SuperSpeed USB device number 5 using etxhci_hcd-170202
[601288.716488] usb 3-3: device firmware changed
[601288.721422] usb 3-3: USB disconnect, device number 5
[601288.746550] r8152 3-3:1.0 (unregistered net_device): get_registers -19
[601288.753967] r8152 3-3:1.0 (unregistered net_device): Get ether addr fail
[601288.761901] r8152 3-3:1.0 eth5: v2.15.0 (2021/04/15)
[601288.767573] r8152 3-3:1.0 eth5: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.

[601288.785286] r8152 3-3:1.0 eth5: chip rev 14
[601300.423029] usb 3-3: new SuperSpeed USB device number 6 using etxhci_hcd-170202
[601300.447292] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[601300.458026] usb-storage 3-3:1.0: USB Mass Storage device detected
[601300.465143] scsi12 : usb-storage 3-3:1.0
[601301.472258] scsi 12:0:0:0: CD-ROM            Realtek  USB CD-ROM               2.00 PQ: 0 ANSI: 0
[601301.482544] scsi 12:0:0:0: Attached scsi generic sg9 type 5
[602212.723525] usb 3-3: USB disconnect, device number 6
[602224.417661] usb 3-2: new SuperSpeed USB device number 7 using etxhci_hcd-170202
[602224.443925] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[602224.456825] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[602224.466588] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[602224.476457] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[602225.589637] usb 3-2: reset SuperSpeed USB device number 7 using etxhci_hcd-170202
[602225.612402] usb 3-2: device firmware changed
[602225.617319] usb 3-2: USB disconnect, device number 7
[602225.642453] r8152 3-2:1.0 (unregistered net_device): get_registers -19
[602225.649874] r8152 3-2:1.0 (unregistered net_device): Get ether addr fail
[602225.657887] r8152 3-2:1.0 eth5: v2.15.0 (2021/04/15)
[602225.663562] r8152 3-2:1.0 eth5: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.

[602225.681273] r8152 3-2:1.0 eth5: chip rev 14
[602237.305937] usb 3-2: new SuperSpeed USB device number 8 using etxhci_hcd-170202
[602237.330203] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[602237.340141] usb-storage 3-2:1.0: USB Mass Storage device detected
[602237.347202] scsi13 : usb-storage 3-2:1.0
[602238.354362] scsi 13:0:0:0: CD-ROM            Realtek  USB CD-ROM               2.00 PQ: 0 ANSI: 0
[602238.364723] scsi 13:0:0:0: Attached scsi generic sg9 type 5
[602323.849193] usb 3-2: USB disconnect, device number 8
[602430.654218] usb 2-4: new high-speed USB device number 3 using etxhci_hcd-170202
[602430.674248] usb 2-4: not running at top speed
[602434.146284] usb 3-4: new SuperSpeed USB device number 9 using etxhci_hcd-170202
[602434.172547] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[602434.185083] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[602434.194861] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[602434.204741] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[602435.319155] usb 3-4: reset SuperSpeed USB device number 9 using etxhci_hcd-170202
[602435.341901] usb 3-4: device firmware changed
[602435.346826] usb 3-4: USB disconnect, device number 9
[602435.371971] r8152 3-4:1.0 (unregistered net_device): get_registers -19
[602435.379380] r8152 3-4:1.0 (unregistered net_device): Get ether addr fail
[602435.387294] r8152 3-4:1.0 eth5: v2.15.0 (2021/04/15)
[602435.392962] r8152 3-4:1.0 eth5: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.

[602435.410673] r8152 3-4:1.0 eth5: chip rev 14
[602447.048443] usb 3-4: new SuperSpeed USB device number 10 using etxhci_hcd-170202
[602447.072704] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[602447.082503] usb-storage 3-4:1.0: USB Mass Storage device detected
[602447.089623] scsi14 : usb-storage 3-4:1.0
[602448.096811] scsi 14:0:0:0: CD-ROM            Realtek  USB CD-ROM               2.00 PQ: 0 ANSI: 0
[602448.107101] scsi 14:0:0:0: Attached scsi generic sg9 type 5

bb-qq, 2021-08-22:
I suspect that the USB port is not supplying enough power to use two devices.
Could you try connecting the device via a self-powered USB hub with an AC adapter?

Also, I am curious about why you want to use 2 devices at the same time.

tevans62, 2021-08-22:
I have multiple vlans and while I know I can configure them in the single interface via cli and configure files (though mixing nuts has me perplexed) was hoping to have each on different physical interfaces and  benchmark in that configuration.

rswiz, 2021-08-28:
> I have multiple vlans and while I know I can configure them in the single interface via cli and configure files (though mixing nuts has me perplexed) was hoping to have each on different physical interfaces and benchmark in that configuration.

Hello OP, you mean that you had a 2.5 GBe USB 3.0 adapter working on your Synology NAS, can I ask you what the model of your device is please?

bb-qq, 2021-09-24:
After some investigation, it seems that I need to switch the operation mode to make the device work as an Ethernet device instead of a CD-ROM. However, Synology's system omits the mechanism to do so.

It would be nice if the device could be recognized as an Ethernet device from the start, but I don't know what is causing it to be recognized as a CD-ROM.

If you plug and unplug the device manually, will the result be the same?

Translated with www.DeepL.com/Translator (free version)

tevans62, 2021-09-24:
> Hello OP, you mean that you had a 2.5 GBe USB 3.0 adapter working on your Synology NAS, can I ask you what the model of your device is please?

DS1815+

tevans62, 2021-09-24:
> After some investigation, it seems that I need to switch the operation mode to make the device work as an Ethernet device instead of a CD-ROM. However, Synology's system omits the mechanism to do [...]

I have tried that without success. Is there a cli-based mechanism to switch the operation mode?

Any reason it only happens with the 2nd device ?

bb-qq, 2021-09-24:
> Is there a cli-based mechanism to switch the operation mode?

[usb_modeswitch](https://www.draisberghof.de/usb_modeswitch/) seems to be that([ref](https://bbs.archlinux.org/viewtopic.php?id=228195)), but I don't know if this can be made to work on Synology Linux.

> Any reason it only happens with the 2nd device ?

I have no idea...

ossource, 2021-10-20:
I had an issue with my DS415+, too.
I fixed it with a USB hub in between NAS and the USB network card. 

1. you can use a powered hub which delivers more power
2. you can avoid the USB controller chip of the NAS which is not the best USB controller. There are compatibility issues (you can see the effect on the shorter compatibility list of the c2000 chipset... )

rswiz, 2021-10-21:
> I had an issue with my DS415+, too. I fixed it with a USB hub in between NAS and the USB network card.
>
> 1. you can use a powered hub which delivers more power
> 2. you can avoid the USB controller [...]

I wonder if my 418play will work this way? Can I ask what hub you bought as the ones I'm finding are priced ridiculously.

bb-qq, 2022-08-25:
The latest release allows the use of udev rules, which may solve this problem.

After installing the package, try the following command:

```
sudo bash /var/packages/r8152/scripts/install-udev-rules

ashway83, 2023-01-09:
I have the same issue on DS918+: each network adapter works separately, and when two are inserted simultaneously, the driver is not loaded for the second inserted.
It looks like the USB driver changes the serial number when r8152 is already in the middle of loading, which interrupts driver initialization. 

```
XXXX@XXXXXX:/$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650078DD9AE41654)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000000)
```

dmesg shows a continuous loop of the messages like below:
```
[ 3069.787570] r8152 2-2:1.0 eth3: chip rev 14
[ 3069.792251] r8152 2-2:1.0 eth3: rx_buf_sz will be overridden to 16384
[ 3069.799565] usb 2-2: USB disconnect, device number 4
[ 3069.985238] usb 2-2: new SuperSpeed USB device number 5 using xhci_hcd
[ 3070.003727] drivers/usb/core/hub.c (2940) Same device found. Change serial to 4013000000
[ 3070.140398] usb 2-2: reset SuperSpeed USB device number 5 using xhci_hcd
[ 3070.158932] usb 2-2: device firmware changed
[ 3070.163732] usb usb2-port2: logical disconnect on port 2 [kworker/2:2]
[ 3070.191266] r8152 2-2:1.0 (unnamed net_device) (uninitialized): get_registers -19
[ 3070.199683] r8152 2-2:1.0 (unnamed net_device) (uninitialized): Get ether addr fail
[ 3070.208653] r8152 2-2:1.0 eth3: v2.16.3 (2022/07/06)
[ 3070.214221] r8152 2-2:1.0 eth3: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 3070.231618] r8152 2-2:1.0 eth3: chip rev 14
[ 3070.236293] r8152 2-2:1.0 eth3: rx_buf_sz will be overridden to 16384
[ 3070.243610] usb 2-2: USB disconnect, device number 5
[ 3070.429255] usb 2-2: new SuperSpeed USB device number 6 using xhci_hcd
[ 3070.447745] drivers/usb/core/hub.c (2940) Same device found. Change serial to 4013000000
[ 3070.581420] usb 2-2: reset SuperSpeed USB device number 6 using xhci_hcd
[ 3070.599530] usb 2-2: device firmware changed
[ 3070.604335] usb usb2-port2: logical disconnect on port 2 [kworker/2:2]
[ 3070.633250] r8152 2-2:1.0 (unnamed net_device) (uninitialized): get_registers -19
[ 3070.641668] r8152 2-2:1.0 (unnamed net_device) (uninitialized): Get ether addr fail
[ 3070.650651] r8152 2-2:1.0 eth3: v2.16.3 (2022/07/06)
[ 3070.656219] r8152 2-2:1.0 eth3: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 3070.673643] r8152 2-2:1.0 eth3: chip rev 14
[ 3070.678318] r8152 2-2:1.0 eth3: rx_buf_sz will be overridden to 16384
[ 3070.685619] usb 2-2: USB disconnect, device number 6
[ 3070.870286] usb 2-2: new SuperSpeed USB device number 7 using xhci_hcd
[ 3070.888773] drivers/usb/core/hub.c (2940) Same device found. Change serial to 4013000000
[ 3071.023439] usb 2-2: reset SuperSpeed USB device number 7 using xhci_hcd
[ 3071.042534] usb 2-2: device firmware changed
[ 3071.047323] usb usb2-port2: logical disconnect on port 2 [kworker/2:2]
[ 3071.075251] r8152 2-2:1.0 (unnamed net_device) (uninitialized): get_registers -19
[ 3071.083654] r8152 2-2:1.0 (unnamed net_device) (uninitialized): Get ether addr fail
[ 3071.092255] usb 2-2: USB disconnect, device number 7
[ 3071.098321] r8152 2-2:1.0 eth3: v2.16.3 (2022/07/06)
[ 3071.103880] r8152 2-2:1.0 eth3: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 3071.121271] r8152 2-2:1.0 eth3: chip rev 14
[ 3071.125941] r8152 2-2:1.0 eth3: rx_buf_sz will be overridden to 16384
```

```
XXXX@XXXXXX:/$ udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[3179.162617] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3/queues/rx-0 (queues)
KERNEL[3179.162654] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3/queues/tx-0 (queues)
KERNEL[3179.162684] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3 (net)
[... 11 lines cut ...]
UDEV  [3179.532026] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:2.0 (usb)
UDEV  [3179.533161] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:2.1 (usb)
UDEV  [3179.533563] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:2.1 (usb)
UDEV  [3179.538411] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0 (usb)
KERNEL[3179.565431] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3 (net)
KERNEL[3179.565463] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3/queues/rx-0 (queues)
KERNEL[3179.565476] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3/queues/tx-0 (queues)
UDEV  [3179.583939] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3 (net)
UDEV  [3179.584968] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3/queues/rx-0 (queues)
UDEV  [3179.584994] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/net/eth3/queues/tx-0 (queues)
[... cut ...]
KERNEL[3179.612191] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0 (usb)
KERNEL[3179.612878] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2 (usb)
UDEV  [3179.770305] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2 (usb)
KERNEL[3179.813988] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2 (usb)
KERNEL[3179.814926] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:2.0 (usb)
KERNEL[3179.815196] add      /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:2.1 (usb)
KERNEL[3179.815378] remove   /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:2.0 (usb)
```

ashway83, 2023-01-10:
I can confirm my previous assumption - if I unload r8152 kernel module and then `insmod` it manually, both devices are initialized correctly.
It looks like it has something to do with hotplug events - by default, the kernel module is loaded when the first adapter is recognized. When the second adapter is recognized on the USB bus, the r8152 driver tries to initialize it. Still, it's interrupted by the USB driver that detects conflict in serial numbers and triggers reenumeration of the second adapter.

As a workaround, loading of the r8152 driver could be blocked via udev rules or in /etc/modprobe.d/blacklist.conf, so then r8152 kernel module could be loaded manually from the script when both adapters are added and serial number is corrected for the second one.

@bb-qq could you please look into this?

ashway83, 2023-01-10:
So, if the NAS is rebooted, both adapters are added successfully, as on boot, by the time the r8152 package starts, both adapters are already added by the USB driver (and the serial number of the second adapter is changed).
In most cases, adapters won't be replugged (at least, I don't see the use case), but if the second adapter is replugged - it won't be added.
I have only two noname adapters from AliExpress with the same serial number, which shouldn't be the case with the legit manufacturers and the reported issue won't occur.
It looks like the main problem is that the USB driver changes the serial number too late - after the device is already added.
```
[ 3070.003727] drivers/usb/core/hub.c (2940) Same device found. Change serial to 4013000000
```
I haven't checked this code, but in theory, serial number conflict could be detected earlier - before the USB device is added.

tevans62, 2023-01-11:
This is what is happening to me, with or without a powered USB3 hub.

I however on my 1518+ cannot find a workaround using manual rmmod/insmod. On reboot I see references to the 2nd adapter, but stuck in the same serial number loop thing:

 1003.266423] usb 3-2.4: reset SuperSpeed USB device number 22 using etxhci_hcd-170202
[ 1003.289188] usb 3-2.4: device firmware changed
[ 1003.295332] r8152 3-2.4:1.0 (unregistered net_device): get_registers -19
[ 1003.302856] r8152 3-2.4:1.0 (unregistered net_device): Get ether addr fail
[ 1003.302944] usb 3-2.4: USB disconnect, device number 22
[ 1003.316690] r8152 3-2.4:1.0 eth5: v2.16.3 (2022/07/06)
[ 1003.322454] r8152 3-2.4:1.0 eth5: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 1003.340075] r8152 3-2.4:1.0 eth5: chip rev 14
[ 1003.344975] r8152 3-2.4:1.0 eth5: rx_buf_sz will be overridden to 16384
[ 1004.589251] usb 3-2.4: new SuperSpeed USB device number 23 using etxhci_hcd-170202
[ 1004.615772] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[ 1004.628314] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 4 ep 2 with no TDs queued?
[ 1004.638199] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 4 ep 3 with no TDs queued?
[ 1004.647875] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 4 ep 6 with no TDs queued?
[ 1005.723337] usb 3-2.4: reset SuperSpeed USB device number 23 using etxhci_hcd-170202
[ 1005.746218] usb 3-2.4: device firmware changed
[ 1005.752108] r8152 3-2.4:1.0 (unregistered net_device): get_registers -19
[ 1005.759633] r8152 3-2.4:1.0 (unregistered net_device): Get ether addr fail
[ 1005.767352] usb 3-2.4: USB disconnect, device number 23
[ 1005.767625] r8152 3-2.4:1.0 eth5: v2.16.3 (2022/07/06)
[ 1005.767627] r8152 3-2.4:1.0 eth5: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 1005.767629] r8152 3-2.4:1.0 eth5: chip rev 14
[ 1005.767631] r8152 3-2.4:1.0 eth5: rx_buf_sz will be overridden to 16384
[ 1007.043422] usb 3-2.4: new SuperSpeed USB device number 24 using etxhci_hcd-170202
[ 1007.069686] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000 
[ 1007.082211] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 4 ep 2 with no TDs queued?
[ 1007.091893] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 4 ep 3 with no TDs queued?
[ 1007.101672] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 4 ep 6 with no TDs queued?
[ 1008.177133] usb 3-2.4: reset SuperSpeed USB device number 24 using etxhci_hcd-170202
[ 1008.200143] usb 3-2.4: device firmware changed
[ 1008.206281] r8152 3-2.4:1.0 (unregistered net_device): get_registers -19
[ 1008.213802] r8152 3-2.4:1.0 (unregistered net_device): Get ether addr fail
[ 1008.221865] usb 3-2.4: USB disconnect, device number 24
[ 1008.222325] r8152 3-2.4:1.0 eth5: v2.16.3 (2022/07/06)
[ 1008.222327] r8152 3-2.4:1.0 eth5: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 1008.222329] r8152 3-2.4:1.0 eth5: chip rev 14
[ 1008.222331] r8152 3-2.4:1.0 eth5: rx_buf_sz will be overridden to 16384
[ 1009.496955] usb 3-2.4: new SuperSpeed USB device number 25 using etxhci_hcd-170202
[ 1009.523599] drivers/usb/core/ethub.c (2055) Same device found. Change serial to 001000000

bb-qq, 2023-01-29:
I can't come up with a good solution to this problem as it seems to be happening at the udev layer.

But the serial numbers of @ashway83 's and @tevans62  's devices seem to be different, so perhaps a combination of devices from different vendors might work.

nmiller0113, 2023-03-27:
@ashway83 @tevans62 - did either of you find a workaround to this besides trying two different vendors? I'm using two of these from Amazon, which I also have working great on a lot of other things https://www.amazon.com/gp/product/B084L4JL9K/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

@bb-qq - Just an FYI that I'm using [2.16.3-4](https://github.com/bb-qq/r8152/releases/tag/2.16.3-4) on a DS916+ which is not a model you currently have listed. The only problem I'm having is getting multiple adapters running from the same vendor. Thanks!

tevans62, 2023-03-27:
I gave up.  Decided not to spend the money. 
This could all be fixed for me if Synology easily allowed multiple vlans and IPs on an interface.

Sjekke, 2023-03-28:
> I gave up.  Decided not to spend the money. 
> 
> This could all be fixed for me if Synology easily allowed multiple vlans and IPs on an interface. 

https://techblik.nl/2021/02/10/synology-dsm7-vlan/

bb-qq, 2023-04-22:
Can someone please try to enable the Hot Plug feature?
(Note: This command modifies your system outside of package management.)
```
sudo bash /var/packages/r8152/scripts/install-udev-rules
```

Bafti71, 2023-04-22:
I did run this command but nothing changes.

bb-qq, 2023-04-30:
In #300, the thread owner seems to have worked well with this command.

Could you reboot the system after enabling hot plugging and post the output of `dmesg` here?

TheS1R, 2023-06-28:
I am able to enable multiple same vendor adapters after enabling Hot Plug feature and rebooting NAS (DS718+, DSM 7.2 Update 1).

bb-qq, 2023-07-02:
Not all members participating in this issue have reported that it worked, but it seems to be working, so I would like to close this issue.
If this does not work in your environment, please create a new individual issue.

closed 2023-07-02 by bb-qq

JinhuW, 2024-06-08:
> Can someone please try to enable the Hot Plug feature? (Note: This command modifies your system outside of package management.)
> 
> ```
> sudo bash /var/packages/r8152/scripts/install-udev-rules
> ```

I can confirm it works for me, (DS 1019+)

Glupikosz, 2024-06-12:
I can confirm it works for me, (DS 420+)

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65002758B5346E87)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (CMI USB 10/100/1G/2.5G LAN 4013000000)

[image]

VAllens, 2024-06-30:
> The latest release allows the use of udev rules, which may solve this problem.
> 
> After installing the package, try the following command:
> 
> ```
> sudo bash /var/packages/r8152/scripts/install-udev-rules
> ```

It works for me!
Finally solved, this problem has been bothering me all day!!!

lsusb :
```bash
root@MM-Cloud:~# lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500699812CA6020)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000000)
  |__4-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001300E04C6800D7)
```

dmesg: 
```bash
root@MM-Cloud:~# dmesg
[   53.912319] ACPI: bus type USB registered
[   53.917219] usbcore: registered new interface driver usbfs
[   53.923611] usbcore: registered new interface driver hub
[   53.929831] usbcore: registered new interface driver ethub
[   53.936230] usbcore: registered new device driver usb
[   53.976537] xhci_hcd 0000:00:14.0: setting latency timer to 64
[   53.976551] xhci_hcd 0000:00:14.0: xHCI Host Controller
[   53.982427] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 1
[   54.190860] xhci_hcd 0000:00:14.0: cache line size of 64 is not supported
[   54.190900] xhci_hcd 0000:00:14.0: irq 133 for MSI/MSI-X
[   54.191363] xHCI xhci_add_endpoint called for root hub
[   54.191368] xHCI xhci_check_bandwidth called for root hub
[   54.191599] hub 1-0:1.0: USB hub found
[   54.195829] hub 1-0:1.0: 7 ports detected
[   54.201455] xhci_hcd 0000:00:14.0: xHCI Host Controller
[   54.207318] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[   54.216070] xHCI xhci_add_endpoint called for root hub
[   54.216078] xHCI xhci_check_bandwidth called for root hub
[   54.216311] hub 2-0:1.0: USB hub found
[   54.220542] hub 2-0:1.0: 6 ports detected
[   54.267438] etxhci_hcd-170202 0000:04:00.0: Etron xHCI Host Controller
[   54.274774] etxhci_hcd-170202 0000:04:00.0: new USB bus registered, assigned bus number 3
[   54.296297] etxhci_hcd-170202 0000:04:00.0: irq 134 for MSI/MSI-X
[   54.297262] xHCI etxhci_add_endpoint called for root hub
[   54.297268] xHCI etxhci_check_bandwidth called for root hub
[   54.297511] ethub 3-0:1.0: USB hub found
[   54.301936] ethub 3-0:1.0: 2 ports detected
[   54.306761] etxhci_hcd-170202 0000:04:00.0: Etron xHCI Host Controller
[   54.314080] etxhci_hcd-170202 0000:04:00.0: new USB bus registered, assigned bus number 4
[   54.324146] xHCI etxhci_add_endpoint called for root hub
[   54.324153] xHCI etxhci_check_bandwidth called for root hub
[   54.324398] ethub 4-0:1.0: USB hub found
[   54.328824] ethub 4-0:1.0: 2 ports detected
[   54.342086] braswell_synobios: module license 'Synology Inc.' taints kernel.
[   54.349982] Disabling lock debugging due to kernel taint
[   54.356811] synobios open /dev/ttyS1 success
[   54.361706] 2024-6-30 10:4:55 UTC
[   54.365417] synobios: load, major number 201
[   54.370197] Brand: Synology
[   54.373320] Model: DS-716+II
[   54.376532] This is default settings: set group disks wakeup number to 1, spinup time deno 1
[   54.385985] synobios cpu_arch proc entry initialized
[   54.391538] synobios crypto_hw proc entry initialized
[   54.397190] synobios syno_platform proc entry initialized
[... 25 lines cut ...]
...
[  107.688148] usb 4-1: reset SuperSpeed USB device number 2 using etxhci_hcd-170202
[  107.900384] netif_napi_add() called with weight 256 on device eth%d
[  107.917864] r8152 4-1:1.0 eth2: v2.17.1 (2023/06/13)
[  107.923477] r8152 4-1:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[  107.940930] r8152 4-1:1.0 eth2: chip rev 14
[  107.947430] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 2 with no TDs queued?
[  107.957635] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 3 with no TDs queued?
[  107.971317] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 6 with no TDs queued?
[  108.218831] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  109.084021] usb 4-2: reset SuperSpeed USB device number 3 using etxhci_hcd-170202
[  110.137784] IPv6: ADDRCONF(NETDEV_UP): eth3: link is not ready
[  110.391735] r8152 4-2:1.0 eth3: v2.17.1 (2023/06/13)
[  110.397307] r8152 4-2:1.0 eth3: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[  110.414754] r8152 4-2:1.0 eth3: chip rev 14
[  110.419529] usbcore: registered new interface driver r8152
[  111.169383] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[  111.177589] r8152 4-1:1.0 eth2: carrier on
[  113.351853] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
[  113.359886] r8152 4-2:1.0 eth3: carrier on
```

rcorex, 2025-08-26:
unfortunately, this problem still happens for me. i have a DS1618+ and 2 Realtek adapters with identical serial numbers and ran into the looping problem. installing the hotplug scripts did not solve the problem. if i unload the r8152 module then both adapters can be seen in lsusb -i with different serial numbers but no eth device started. if load the r8152 module again the 2nd eth device cannot be started and dmesg shows the same problem loop:

[ 1504.054632] usb 2-2.2: new SuperSpeed USB device number 9 using xhci_hcd
[ 1504.072239] drivers/usb/core/hub.c (2968) Same device found. Change serial to 4F13000000
[ 1504.166830] usb 2-2.2: reset SuperSpeed USB device number 9 using xhci_hcd
[ 1504.184976] usb 2-2.2: device firmware changed
[ 1504.189564] usb 2-2-port2: logical disconnect on port 2 [kworker/3:7]
[ 1504.196302] r8152 2-2.2:1.0: read type=0x00cc, index=0xd818 fail -19
[ 1504.203166] usb 2-2.2: USB disconnect, device number 9
[ 1504.515639] usb 2-2.2: new SuperSpeed USB device number 10 using xhci_hcd
[ 1504.534401] drivers/usb/core/hub.c (2968) Same device found. Change serial to 4F13000000
[ 1504.628827] usb 2-2.2: reset SuperSpeed USB device number 10 using xhci_hcd
[ 1504.646979] usb 2-2.2: device firmware changed
[ 1504.651563] usb 2-2-port2: logical disconnect on port 2 [kworker/3:7]
[ 1504.658303] r8152 2-2.2:1.0: read type=0x00cc, index=0xd818 fail -19
[ 1504.665166] usb 2-2.2: USB disconnect, device number 10
[ 1504.977637] usb 2-2.2: new SuperSpeed USB device number 11 using xhci_hcd
[ 1504.996225] drivers/usb/core/hub.c (2968) Same device found. Change serial to 4F13000000
[ 1505.091582] usb 2-2.2: reset SuperSpeed USB device number 11 using xhci_hcd
[ 1505.110005] usb 2-2.2: device firmware changed

luckily i had more adapters and found one which had a different serial number which worked.