bb-qq/r8152 #141: DS418 with 8156B

stcssk, 2021-07-30:
The 2.5G network card is configured successfully, but the data cannot be transmitted from the 2.5G network port, which will cause the NAS down.

stcssk, 2021-07-30:
DSM7.0

bb-qq, 2021-08-22:
To summarize the reports in this forum, it seems that the USB controller of the RTD1296 platform has a flaw that prevents this driver from working stably.

It's odd that it doesn't work with the RealTek hardware, but there doesn't seem to be anything I can do about it at the moment. Sorry.

bb-qq, 2022-02-05:
I created an experimental driver with low memory consumption. Could you try this?
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-rtd1296-2.15.0-6_rxbuf-test.spk

stcssk, 2022-02-05:
您好，我已收到您的邮件，我会尽快查看并处理，辛苦了，谢谢！~

stcssk, 2022-02-05:
Very thx.I will try in few days！Devices are not nearby now.thx！！

 

 https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-rtd1296-2.15.0-6_rxbuf-test.spk

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq