bb-qq/r8152 #142: New Device Request: LLANO Wired 2500Mbps USB C 2.5G External Network Card Type-A To RJ45

zhuanyi, 2021-07-31:
### Description of the problem

> Device is LLANO Wired 2500Mbps USB C 2.5G External Network Card Type-A To RJ45  (Aliexpress listing: https://www.aliexpress.com/item/1005002532990873.html), (not affiliated with seller) the chip inside is RTL8156B. 

VID: 0bda
PID: 8156
REV: 31.00

Product with the different brand but same VID, PID and REV seems to work as reported in here: https://github.com/bb-qq/r8152/issues/115. (unsure DSM version though).

The light on the converter is on and there is occasional blinking but DSM is showing disconnected and no valid IP was obtained. Also the hub I connect DS to is blinking yellow on the port connected with DS which indicates the speed negotiated is less than 2.5Gbps.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle
DSM 6.2.4-25556
Linux xxxxx 3.10.105 #25556 SMP Thu Mar 18 12:51:35 CST 2021 x86_64 GNU/Linux synology_cedarview_412+
Product name: 
LLANO Wired 2500Mbps USB C 2.5G External Network Card Type-A To RJ45  (Aliexpress listing: https://www.aliexpress.com/item/1005002532990873.html)

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

Connect via QNAP hub to laptop which has another 2.5G USB-C adapter with the same chipset (same VID/PID and REV number)
QNAP hub is QSW-1105-5T 
Cable is Cat. 5E

### Output of ``dmesg`` command

This is the part I think that's relevant only as the full dmesg output is quite long, feel free to let me know if you'd like me to send a full output.

````
[  177.871477] init: iscsi_pluginserverd main process (12322) terminated with st                                                                                                                                                                             atus 255
[  177.879656] init: iscsi_pluginserverd main process ended, respawning
[  177.994510] ata3.00: configured for UDMA/133
[  177.998901] ata3: EH complete
[  178.448177] ata4.00: configured for UDMA/133
[  178.452493] ata4: EH complete
[  180.301930] loop: module loaded
[  188.520515] r8152: module verification failed: signature and/or required key                                                                                                                                                                              missing - tainting kernel
[  188.597708] r8152 8-2:2.0: Unknown version 0x0000
[  188.608941] usbcore: registered new interface driver r8152
[  188.616688] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 5 ep 2 wi                                                                                                                                                                             th no TDs queued?
[  188.629904] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 5 ep 3 wi                                                                                                                                                                             th no TDs queued?
[  188.638705] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 5 ep 6 wi                                                                                                                                                                             th no TDs queued?
[  189.712415] usb 8-1.2: reset SuperSpeed USB device number 5 using etxhci_hcd-                                                                                                                                                                             170202
[  189.911444] r8152 8-1.2:1.0 eth2: v2.15.0 (2021/04/15)
[  189.916709] r8152 8-1.2:1.0 eth2: This product is covered by one or more of t                                                                                                                                                                             he following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[  189.932717] r8152 8-1.2:1.0 eth2: chip rev 6
[  193.516901] init: nmbd main process (15115) killed by TERM signal
[  197.319591] EXT4-fs (sdr1): barriers disabled
[  197.746484] EXT4-fs (sdr1): mounted filesystem with ordered data mode. Opts:                                                                                                                                                                              nodelalloc,synoacl,data=ordered,oldalloc
[  198.815051] init: scsi_plugin_server main process (12315) killed by TERM sign                                                                                                                                                                             al
[  199.135613] init: synoscheduler-vmtouch main process (4621) killed by TERM si                                                                                                                                                                             gnal
[  199.197460] init: iscsi_pluginserverd main process (15871) killed by TERM sig                                                                                                                                                                             nal
[  199.362107] init: iscsi_pluginengined main process (15787) killed by TERM sig                                                                                                                                                                             nal
[  199.600191] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  199.875820] init: syno-check-disk-compatibility main process (15714) terminat                                                                                                                                                                             ed with status 255
[  200.778809] init: scsi_plugin_server main process (16010) killed by TERM sign                                                                                                                                                                             al
[  205.471209] init: upnpd main process (11893) killed by KILL signal
[  205.599275] init: synoscheduler-vmtouch main process (16252) killed by TERM s                                                                                                                                                                             ignal
[... 70 lines cut ...]
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:399630 errors:0 dropped:0 overruns:0 frame:0
          TX packets:399630 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:44340048 (42.2 MiB)  TX bytes:44340048 (42.2 MiB)

tun1000   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:169.254.166.66  P-t-P:169.254.166.66  Mask:255.255.248.0
          inet6 addr: fec0:a2b2:6::642/112 Scope:Site
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1400  Metric:1
          RX packets:9965 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10582 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:2616272 (2.4 MiB)  TX bytes:5142067 (4.9 MiB)

````

jglathe, 2021-08-01:
Hi @zhuanyi ,

just had a look on the dubug output of yours. 
`[  188.597708] r8152 8-2:2.0: Unknown version 0x0000`
This is the (only) response regarding the Realtek USB 10/100/1G/2.5G LAN 001000001 dongle - looks like not recognized or something like this. Need to look into the source to find where this happens. 
All the other messages are related to the Realtek USB 10/100/1000 LAN 000001 dongle. This is still shown as LAN2... as I said, I would remove it from the setup if unused. It also shows "normal" behaviour without a link/cable. 

with best regards

Jens

jglathe, 2021-08-01:
Hi @zhuanyi ,

took a look into the code:  rtl8152_get_version() does this when the return value is not known. The version query is masked (0x7cf0, "111110011110000"), so if there is a version oand not really 0, we won't see it. So to get closer I would first show the unmasked value. But now we're talking driver hacking, your dongle won't work with this driver. Sorry.

with best regards

Jens

zhuanyi, 2021-08-01:
Mmm...I am afraid you are right... I have shutdown the DS, unplugged the USB hub (almost forgot about it) and then power on the DS again, and this is what I get from the output (after the DS comes back online):

```
 dmesg|grep r8152
[  172.745818] r8152: module verification failed: signature and/or required key missing - tainting kernel
[  172.756193] usbcore: registered new interface driver r8152
```

```
lsmod|grep r8152
r8152                 196503  0
usbcore               176590  9 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd
```
but this is what scared me:
[image]

so I guess it wasn't getting detected after all...

Oh well, at least we tried, and thanks a lot for your help!

zhuanyi, 2021-08-01:
Also, @jglathe Is there anything I could do to help to make this work though? Never played with driver hacking but I can code in C++ and Python if those helps!

jglathe, 2021-08-01:
Well it is quite a way to get a working build environment - I haven’t gotten one available, yet. Just looking at the dongle picture I see a further complication: this is a USB type C dongle, not type A like mine. So you will be using some kind of adapter which will have its own electronics in it? Translating between the protocols? It registers as 3.20, not 3.00. This could be enough to let this be a futile exercise. I would suggest to try getting a type A dongle, should be the better way. And with this dongle you have, I would first take a look on a normal current pc if it works with linux. If that is made sure of, it would be time to compare the version of r8152.c used.

zhuanyi, 2021-08-01:
I see, thanks for your help, looks like it is definitely beyond my skill level then (I wish I had studied for electrical engineering back in school!) I actually bought the USB-A type (they have both and I used USB-A as that's what my DS has.

jglathe, 2021-08-01:
Still worth a try on a „normal“ PC or laptop with linux on it.

zhuanyi, 2021-08-01:
Mmm...you actually gave me a hint here...I do have an old laptop running Minx Linux (a fork of Ubuntu) and when I plugged the dongle into the only USB port it has, this is what I saw in dmesg:

```
[342957.196307] usb 2-1.1: new high-speed USB device number 4 using ehci-pci
[342957.317973] usb 2-1.1: New USB device found, idVendor=0bda, idProduct=8156, bcdDevice=31.00
[342957.317977] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[342957.317979] usb 2-1.1: Product: USB 10/100/1G/2.5G LAN
[342957.317981] usb 2-1.1: Manufacturer: Realtek
[342957.317983] usb 2-1.1: SerialNumber: 001000001
[342957.464703] cdc_ncm 2-1.1:2.0: MAC-Address: 00:e0:4c:68:00:1d
[342957.464708] cdc_ncm 2-1.1:2.0: setting rx_max = 16384
[342957.464821] cdc_ncm 2-1.1:2.0: setting tx_max = 16384
[342957.465330] cdc_ncm 2-1.1:2.0 usb0: register 'cdc_ncm' at usb-0000:00:1d.0-1.1, CDC NCM, 00:e0:4c:68:00:1d
[342957.466267] usbcore: registered new interface driver cdc_ncm
[342957.483601] usbcore: registered new interface driver cdc_wdm
[342957.490379] usbcore: registered new interface driver cdc_mbim
[342957.525853] cdc_ncm 2-1.1:2.0 enx00e04c68001d: renamed from usb0
[342981.035298] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342981.291366] IPv6: ADDRCONF(NETDEV_CHANGE): enx00e04c68001d: link becomes ready
[342981.547297] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342982.059338] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342982.571332] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342983.083295] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342983.595322] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342984.107328] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342984.619323] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342985.131325] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342985.643321] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342986.155297] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342986.667289] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342987.179305] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342987.691287] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342988.203293] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342988.715337] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342989.227290] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342989.739284] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342990.251309] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342990.763286] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342991.275315] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342991.787313] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342992.299307] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342992.811302] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342993.323299] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342993.835280] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342994.347294] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342994.859279] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342995.371275] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342995.883290] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342996.395332] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342996.907270] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342997.419305] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342997.931300] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342998.443287] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342998.955297] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342999.467279] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[342999.979287] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343000.491342] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343001.003312] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343001.515266] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343002.027349] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343002.539267] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343003.051267] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343003.563274] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343004.075313] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[... cut ...]
[343052.715256] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343053.227278] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343053.739219] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
[343054.251253] cdc_ncm 2-1.1:2.0 enx00e04c68001d: 1000 mbit/s downlink 1000 mbit/s uplink
```

And the dongle worked without any installation required (I have disabled Wifi to make sure it wasn't wifi connecting). So somehow it does gets recognized in Ubuntu even with that weird S/N.

zhuanyi, 2021-08-01:
A bit more progress....so all my tests previously done were plugged into the back of the DS, but I decided to give the front USB a try and after stop and start the driver package in Package Centre, it seems to work, although only at the Gigabit speed as the front panel port for DS412+ is only USB 2.0.

```
[ 1835.406631] usb 2-4: new high-speed USB device number 3 using ehci-pci
[ 1835.674526] usb 2-4: reset high-speed USB device number 3 using ehci-pci
[ 1846.402878] netif_napi_add() called with weight 256 on device eth%d
[ 1846.420284] r8152 2-4:1.0 eth2: v2.15.0 (2021/04/15)
[ 1846.425290] r8152 2-4:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[ 1846.441206] r8152 2-4:1.0 eth2: chip rev 14
[ 1849.171795] init: scsi_plugin_server main process (12140) killed by TERM signal
[ 1849.651753] init: nmbd main process (19539) killed by TERM signal
[ 2093.125669] usbcore: deregistering interface driver r8152
[ 2143.227302] usb 2-4: reset high-speed USB device number 3 using ehci-pci
[ 2143.466465] r8152 2-4:1.0 eth2: v2.15.0 (2021/04/15)
[ 2143.471516] r8152 2-4:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[ 2143.487186] r8152 2-4:1.0 eth2: chip rev 14
[ 2143.491485] usbcore: registered new interface driver r8152
[ 2153.603289] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[ 2153.829455] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[ 2153.837235] r8152 2-4:1.0 eth2: carrier on
[ 2156.859999] init: scsi_plugin_server main process (19565) killed by TERM signal
[ 2157.015213] init: nmbd main process (27494) killed by TERM signal
[ 2157.285079] init: dhcp-client (eth2) main process (26912) killed by TERM signal
[ 2158.561787] init: nmbd main process (27785) killed by TERM signal
[ 2159.482250] init: iscsi_pluginserverd main process (27504) killed by TERM signal
[ 2159.572379] init: iscsi_pluginengined main process (27499) killed by TERM signal
[ 2160.262046] init: scsi_plugin_server main process (27513) killed by TERM signal
[ 2162.256713] init: iscsi_pluginserverd main process (28088) killed by TERM signal
[ 2162.351556] init: iscsi_pluginengined main process (28074) killed by TERM signal
[ 2162.595351] init: scsi_plugin_server main process (28218) killed by TERM signal
[ 2163.934436] init: nmbd main process (28439) killed by TERM signal
[ 2166.762762] init: nmbd main process (29561) killed by TERM signal
[ 2167.753110] init: upnpd main process (15646) killed by KILL signal
[ 2168.646460] init: scsi_plugin_server main process (28718) killed by TERM signal
[ 2169.089646] init: nmbd main process (29912) killed by TERM signal
[ 2170.585597] init: iscsi_pluginserverd main process (29863) killed by TERM signal
[ 2170.601873] init: iscsi_pluginengined main process (29852) killed by TERM signal
[ 2170.754861] init: scsi_plugin_server main process (29907) killed by TERM signal

```

[image]

So I guess there must be something different between the USB at the front vs at the back (other than 2.0 vs 3.0).

jglathe, 2021-08-01:
Hey not bad, congratulations. But now this will not get easier: what do we know about the USB3 chip? It is not part of the ATOM CPU, so would probably be some kind of extra chip... ASMedia or something. What does lspci say?

jglathe, 2021-08-01:
OTOH, do you have a USB3 hub you could plug in? Could also be one of those pesky R8153 USB dongles with 3 USB3 slots. Maybe this masks the protocol deficiency/quirk of the host chip.

zhuanyi, 2021-08-01:
mmm....my guess is the one in the red box is the USB 3.0 ports since the two attached USB external HDDs are connected to it:

[image]

this is my lspci output....not sure if the numbers would make any sense to me though :(

```
lspci
00:00.0 Class 0600: Device 8086:0bf3 (rev 04)
00:02.0 Class 0300: Device 8086:0be2 (rev 0b)
00:19.0 Class 0200: Device 8086:10e5
00:1a.0 Class 0c03: Device 8086:3a37
00:1a.7 Class 0c03: Device 8086:3a3c
00:1c.0 Class 0604: Device 8086:3a40
00:1c.1 Class 0604: Device 8086:3a42
00:1c.2 Class 0604: Device 8086:3a44
00:1c.3 Class 0604: Device 8086:3a46
00:1c.4 Class 0604: Device 8086:3a48
00:1c.5 Class 0604: Device 8086:3a4a
00:1d.0 Class 0c03: Device 8086:3a34
00:1d.1 Class 0c03: Device 8086:3a35
00:1d.2 Class 0c03: Device 8086:3a36
00:1d.7 Class 0c03: Device 8086:3a3a
00:1e.0 Class 0604: Device 8086:244e (rev 90)
00:1f.0 Class 0601: Device 8086:3a16
00:1f.2 Class 0106: Device 8086:3a22
00:1f.3 Class 0c05: Device 8086:3a30
01:00.0 Class 0200: Device 8086:10d3
02:00.0 Class 0200: Device 8086:10d3
03:00.0 Class 0180: Device 1095:3531 (rev 01)
04:00.0 Class 0c03: Device 1b6f:7023 (rev 01)
```

zhuanyi, 2021-08-01:
> OTOH, do you have a USB3 hub you could plug in? Could also be one of those pesky R8153 USB dongles with 3 USB3 slots. Maybe this masks the protocol deficiency/quirk of the host chip.

Yap I do, that was the dongle with the Gigabit Ethernet port that initially confused me! Good point, let me give that a shot!

jglathe, 2021-08-01:
Etron Technology, Inc. EJ168 USB 3.0 Host Controller Hardware ID PCI\VEN_1B6F&DEV_7023

zhuanyi, 2021-08-01:
Incredible, worked!!!

[image]

Kinda weird to have a LAN4, but no complains!!! Thanks for your help!!!!

jglathe, 2021-08-01:
Well. I did some search, and... the fixes in https://github.com/torvalds/linux/blob/master/drivers/usb/host/xhci-pci.c#L258 are dangerously close to the hardware release date of the DS-412+. Some are newer. Which could mean that they're not in the kernel that is used by it. 
[image]
I just read that you gt it to work with the hub daisy chain. That is awesome. I will remember this trick :)

zhuanyi, 2021-08-01:
> Well. I did some search, and... the fixes in https://github.com/torvalds/linux/blob/master/drivers/usb/host/xhci-pci.c#L258 are dangerously close to the hardware release date of the DS-412+. Some are newer. Which could mean that they're not in the kernel that is used by it.

Ah I see....I don't have any hope that DSM6 will still be updated, hopefully one day people would be able to port DSM 7 into my old NAS...but for now at least something works.

> I just read that you got it to work with the hub daisy chain. That is awesome. I will remember this trick :)

Wouldn't have been possible without your help! Has been spending days trying to figure it out, thanks again!

closed 2021-08-01 by zhuanyi

jglathe, 2021-08-01:
I get up to 2Gb/s writing while doing image backup from my living room PC, which is kinda „main“. The board has a RTL8125 on it, backbone is 5GB/s (two stories down to the NAS). You get used to these kinds of speeds, just about halves backup time. But: I also got a 10Gb link in a DS1517+, with mainly RAID6 arrays, and writing speed would barely creep above 100MiB/s. Reading can top the 250MiB/s of the receiving interface, but takes some time. Horses four courses, I guess.