bb-qq/r8152 #144: Disappear in DSM network adapter after reboot

rickywu, 2021-08-05:
The first time I install it when use Proxmox VE pass USB device to VM works well, but after reboot,
lsusb list it but not appear in DSM network adapter.

I verified this works

01:1b.0) hub
  |__2-1         0bda:815**6**:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

but this not work:

01:1b.0) hub
  |__2-1         0bda:815**1**:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)

### Description of your products

Linux NAS 4.4.59+ #25426 SMP PREEMPT Tue May 12 04:54:55 CST 2020 x86_64 GNU/Linux synology_apollolake_918+
DSM version 6.2.3
UGreen 2.5G

### Output of ``dmesg`` command

````
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.59+ (root@build3) (gcc version 4.9.3 20150311 (prerelease) (crosstool-NG 1.20.0) ) #25426 SMP PREEMPT Tue May 12 04:54:55 CST 2020
[    0.000000] Command line: syno_hdd_powerup_seq=1 HddHotplug=0 syno_hw_version=DS918+ vender_format_version=2 console=ttyS0,115200n8 withefi elevator=elevator quiet syno_hdd_detect=0 syno_port_thaw=1 root=/dev/md0 sn=1780PDN123456 mac1=001132123456 netif_num=1 vid=0x1908 pid=0x0226
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] CPU: vendor_id 'AuthenticAMD' unknown, using generic init.
               CPU: Your system may be unstable.
[    0.000000] x86/fpu: Legacy x87 FPU detected.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000100-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffdbfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffdc000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.14.0-0-g155821a1990b-prebuilt.qemu.org 04/01/2014
[    0.000000] Kernel/User page tables isolation: disabled
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x7ffdc max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0080000000 mask FF80000000 uncachable
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC
[    0.000000] Base memory trampoline at [ffff880000098000] 98000 size 28672
[    0.000000] BRK [0x01a06000, 0x01a06fff] PGTABLE
[    0.000000] BRK [0x01a07000, 0x01a07fff] PGTABLE
[    0.000000] BRK [0x01a08000, 0x01a08fff] PGTABLE
[    0.000000] BRK [0x01a09000, 0x01a09fff] PGTABLE
[    0.000000] BRK [0x01a0a000, 0x01a0afff] PGTABLE
[    0.000000] RAMDISK: [mem 0x364bb000-0x37ffffff]
[    0.000000] ACPI: Early table checksum verification disabled
[... 666 lines cut ...]
eth0      Link encap:Ethernet  HWaddr 00:11:32:12:34:56
          inet addr:192.168.0.3  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe12:3456/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:75534 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6076 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:82621839 (78.7 MiB)  TX bytes:2431354 (2.3 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:82 errors:0 dropped:0 overruns:0 frame:0
          TX packets:82 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:18958 (18.5 KiB)  TX bytes:18958 (18.5 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

rickywu, 2021-10-15:
It disapear again, do you know the reason?

bb-qq, 2021-11-13:
Could you try to increase the delay of loading driver written in `/var/packages/r8152/scripts/start-stop-status`? `sleep 15` is it.

fallaha56, 2021-12-05:
Hi guys, similar problem here but just a good old fashioned reboot into DSM 6.2 is enough to make the 8156 adaptor disappear

Any suggestions how to make the Syno see it on reboot without having to disconnect network cables and stop/start the driver manually?

fallaha56, 2021-12-05:
Do I have to modify the driver script and then re-upload?

> Could you try to increase the delay of loading driver written in `/var/packages/r8152/scripts/start-stop-status`? `sleep 15` is it.

Am I supposed to modify the script then re-upload the driver?

Thanks -and sorry for my n00bness here

rickywu, 2021-12-06:
I already changed to 15s and seems my USB address changed so DSM cannot recongize it
I'll try to find the reason why address changed and verify this problem again.

rickywu, 2021-12-10:
@bb-qq 
I changed to 20s still has this problem
lsusb shows that USB device with correct product id and ip a s nothing found

rickywu, 2021-12-30:
I solved by install drivers in pve host os, after that it can be recognozed correcttly

```
apt install dwarves
apt install pve-headers-5.11.22-1-pve/now build-essential dkms
cp /sys/kernel/btf/vmlinux /usr/lib/modules/`uname -r`/build/

#install by compile
make
make install
mkdir -p /lib/modules/`uname -r`/kernel/net/usb
cp /root/r8152-2.15.0/r8152.ko /lib/modules/`uname -r`/kernel/net/usb/
#must install rules
make install_rules
depmod -a
update-initramfs -u

#install by dkms
dkms add -m r8125 -v 2.15.0
dkms build -m r8125 -v 2.15.0
dkms install -m r8125 -v 2.15.0
depmod -a
modprobe r8125

#load for reboot
echo "r8152" | sudo tee /etc/modules-load.d/r8152.conf

#reboot and check
lsmod | grep 'r8152'
```

closed 2021-12-30 by rickywu