bb-qq/r8152 #145: Network interface keeps restarting after connected PC is shut down

Payn, 2021-08-06:
### Description of the problem

Like i wrote here https://github.com/bb-qq/r8152/issues/135#issuecomment-893837661 this setup is working and stable during normal use. When i shut down PC connected to NAS with that adapter, network interface keeps restarting generating spam like this in the log:

[image]

Thousands of log entries during night. Tried using both DHCP and static IPs for this interface, several NAS restarts, 
Only thing that stops this spam is disconnect and reconnect either USB adaper on NAS side or network cable on PC side while PC is still shut down. 
Motherboard doesn't power off the NIC, LEDs are still blinking while this is happening. Didn't find any related settings in BIOS.

### Description of your products

> - DS416play (braswell)
>     - Linux NAS 3.10.108 #41890 SMP Thu Jul 15 03:41:52 CST 2021 x86_64 GNU/Linux synology_braswell_416play
>     - DSM 7.0-41890
> - i-tec USB-C Metal 2.5Gbps Ethernet Adapter

### Description of your environment

> - connect directly to a PC 
>     - PC: Realtek® RTL8125-CG 2.5G LAN
> - cable Cat.6a

### Output of ``dmesg`` command (pasting only segment between consecutive link UP and DOWN)

````
[Fri Aug  6 23:27:08 2021] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[Fri Aug  6 23:27:08 2021] ppid:1(systemd), pid:6393(synoneteventd), dirtied inode 805159 (.net_link_is_processing) on tmpfs
[Fri Aug  6 23:27:08 2021] r8152 2-2:1.0 eth2: carrier on
[Fri Aug  6 23:27:08 2021] ppid:1(systemd), pid:17242(syslog-ng), dirtied inode 21267 (.SYNOSYSDB-wal) on md0
[Fri Aug  6 23:27:08 2021] ppid:1(systemd), pid:17242(syslog-ng), dirtied inode 24952 (.SYNOSYSDB-shm) on md0
[Fri Aug  6 23:27:08 2021] ppid:1(systemd), pid:17241(syslog-ng), dirtied inode 9445943 (SYNOSYSLOGDB__LOCALARCH.DB-shm) on dm-0
ash-4.4# dmesg -T
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154912 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154920 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154928 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154936 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154944 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154952 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154960 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:6611(md2_raid5), WRITE block 8 on sda5 (2 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:6611(md2_raid5), WRITE block 8 on sdd5 (2 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:6611(md2_raid5), WRITE block 8 on sdc5 (2 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:6611(md2_raid5), WRITE block 8 on sdb5 (2 sectors)
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16402(synonetdtool), WRITE block 1823280 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16402(synonetdtool), dirtied inode 799931 (lock_rt_tables) on tmpfs
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16402(synonetdtool), dirtied inode 23586 (multi_table_config.tmp.3zCSSq) on md0
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16402(synonetdtool), dirtied inode 23586 (multi_table_config.tmp.3zCSSq) on md0
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16402(synonetdtool), WRITE block 1823288 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154968 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154976 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154984 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2154992 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155000 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155008 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155016 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155024 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155032 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155040 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155048 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155056 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16402(synonetdtool), WRITE block 1823296 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16432(synonetdtool), dirtied inode 21267 (eth2-table-rule-v6) on md0
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16432(synonetdtool), dirtied inode 24103 (eth2-table-rule-v6.tmp.7BUaua) on md0
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16432(synonetdtool), dirtied inode 24103 (eth2-table-rule-v6.tmp.7BUaua) on md0
[Fri Aug  6 23:26:53 2021] ppid:16290(interface_polic), pid:16432(synonetdtool), WRITE block 1820888 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155064 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155072 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155080 on md0 (8 sectors)
[Fri Aug  6 23:26:53 2021] ppid:2(kthreadd), pid:4245(jbd2/md0-8), WRITE block 2155088 on md0 (8 sectors)
[... 1045 lines cut ...]
          inet addr:20.0.0.1  Bcast:20.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::2e0:4cff:fe68:63/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:852208 errors:0 dropped:0 overruns:0 frame:0
          TX packets:661515 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5265748673 (4.9 GiB)  TX bytes:4149224694 (3.8 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:20100 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20100 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:5523693 (5.2 MiB)  TX bytes:5523693 (5.2 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

Payn, 2021-08-07:
So setting MTU to default 1500 in NAS only, reduces this issue to only 1 link down/up after pc is shut down and several of them after turning pc on until login screen. Changing MTU in pc has no effect.

bb-qq, 2021-11-23:
If I detach the LAN adapter on the PC side from the USB, will the problem no longer occur?

If so, you may try changing the power management settings of the LAN adapter on the PC side from Device Manager and the power supply settings of the USB port from BIOS settings.

Payn, 2021-12-07:
Yes if i disconnect the cable on PC side, this spam stops.
I'm using Asus Strix x570-E which has onboard 2.5G LAN so no USB on PC side and i can't seem to find any LAN adapter related options in BIOS.

I tried disabling all settings in LAN adapter driver in device manager that sounded like related to power management but nothing changed. There are many more, no idea what else to change.

When PC is shut down, the LED on 2.5G port on the back is still blinking, motherboard never shuts down the adapter.

But like i said, setting MTU to 1500 at least reduce this spam, so it is somehow related to MTU setting.
Any ideas?

Payn, 2022-01-08:
```
ash-4.4# ethtool -s eth2 speed 2500 duplex full autoneg off
Cannot set new settings: Invalid argument
  not setting speed
  not setting duplex
  not setting autoneg
```

It doesn't work with speed 2500 for some reason, command accepts only speed 1000.

But it looks like it fixed this issue. Even with 9000 MTU there is only one "Link down" event in log after PC shutdown.
Thank you.

bb-qq, 2022-01-09:
This issue seems auto-negotiation issue caused by the PC side. Disabling auto-negotiation on the PC side may also fix the issue.

The problem seems to be solved, so this issue is closed.

closed 2022-01-09 by bb-qq

Payn, 2022-02-28:
I just used speed 1000 and it stayed at 2500, but i am permanently connected to my PC so the link speed is already negotiated or something and won't change.
Are you using the command while not connected through 2.5Gbit adapter?

Also if you get it to work, you need to schedule a task to run that command automatically after every reboot.

Payn, 2022-10-18:
After upgrading to DSM 7.1 and 2.16.3-2 this broke again.
`ethtool -s eth2 speed 1000 duplex full autoneg off` now sets interface speed to 1000 which wasn't happening before (it stayed at 2500)

ethtool doesn't support 2500 as speed parameter so there is no way to disable auto-negotiation anymore and interface keeps spamming those Link DOWN / Link UP events again.

Any ideas?

bb-qq, 2022-11-01:
The documentation says this command disables autonegotiation, does it work in your environment?

```
sudo ethtool -s eth2 autoneg on advertise 0x802f
```

Payn, 2022-11-01:
`ethtool eth2` still shows: `Auto-negotiation: on` with that command

Payn, 2022-11-01:
It seems like auto-negotiation setting just isn't available while using 2.5Gbit mode. See https://github.com/bb-qq/r8152/blob/master/r8152.c#L9121
SPEED_2500 case is missing in the driver so you just can't use `autoneg off` with anything else than 10/100/1000 currently, also reason why it returns "Invalid argument  not setting autoneg".

Do you think you could add support for it?

bb-qq, 2022-11-23:
Realtek says that:

```
- Example of setting speed

	2.5G before kernel v4.10
	# ethtool -s eth0 autoneg on advertise 0x802f

	2.5G for kernel v4.10 and later
	# ethtool -s eth0 autoneg on advertise 0x80000000002f
```

So I think auto-negotiation should be off internally. Is it possible to know the link speed on the peer device side?

Payn, 2022-11-23:
Those commands aren't doing anything. `ethtool eth2` outputs are same before and after and that LINK UP/DOWN spam in log didn't disappear. Those flags just contain bits for adding default advertised link modes 
To me it seems like auto-negotiation is on internally by default (ethtool always says "on") and because of the missing case in switch i posted above there is no way to disable it. 
`ethtool -s eth2 autoneg off speed 1000` <- WORKS
`ethtool -s eth2 autoneg off speed 2500` <- DOES NOT WORK (invalid argument error)

On PC i disabled auto-negotiation and set 2.5Gbit manually. But looking at ethtool output these lines are interesting:

>         Advertised link modes:  10baseT/Half 10baseT/Full
>                                 100baseT/Half 100baseT/Full
>                                 1000baseT/Full
>                                 2500baseX/Full
>         Advertised pause frame use: Symmetric Receive-only
>         Advertised auto-negotiation: Yes

>        Link partner advertised link modes:  10baseT/Half 10baseT/Full
>                                             100baseT/Half 100baseT/Full
>                                             1000baseT/Full
>        Link partner advertised pause frame use: No
>        Link partner advertised auto-negotiation: Yes

If i understand correctly Link partner is my PC, and for some reason it is not advertising 2500 but it is advertising auto-negotiation as enabled even though it's set to 2500 manually. Could it be the cause for this issue?