bb-qq/r8152 #147: Request for DS214play version

daveyeger, 2021-08-25:
Hi @bb-qq, any possibility of get a driver for the evansport chipset? 

Thank you for all of your work!

Dave

demotam, 2021-08-28:
I always need for my DS214play.

thank you so much~

k-vq, 2021-09-02:
It would be great if @bb-qq could support evansport for DSM7

bb-qq, 2021-09-24:
I uploaded [2.15.0-4 release](https://github.com/bb-qq/r8152/releases/tag/2.15.0-4) which includes the evansport support.
I would appreciate it if you report whether it works and its stability.

daveyeger, 2021-09-24:
Thank you. I can confirm it installs, works and comes up after a reboot.

Will perform performance testing next.

You are the man!

[image]

k-vq, 2021-09-25:
The 2.15.0-4 version install and works well on my DS215play

iperf3 report speed between 1.4 and 1.5 Gbps
[image]

but file downloading only able to reach max 1Gbps
[image]

demotam, 2021-09-26:
working well on my 214play!!! thank so much!!!

daveyeger, 2021-09-27:
@k-vq your drives are probably limiting throughput. If you are running DSM 7 run a disk perf test. I was running WD Red drives. Switched to Seagate. about 50% faster now.

k-vq, 2021-09-27:
> @k-vq your drives are probably limiting throughput. If you are running DSM 7 run a disk perf test. I was running WD Red drives. Switched to Seagate. about 50% faster now.

Yes, I use WD Red as well (two disks, non RAID). It's strange that DSM7 disk benchmark show 194MB/s throughput (quite close to WD official spec of 210MB/s). But disk to disk file copying only able to reach ~100MB/s

daveyeger, 2021-09-27:
> Yes, I use WD Red as well (two disks, non RAID). It's strange that DSM7 disk benchmark show 194MB/s throughput (quite close to WD official spec of 210MB/s). But disk to disk file copying only able [...]

I was getting about 50% of the benchmark the WD drives, Benchmark also had them pretty close to spec. The seagate exos I have in the ds214 now benchmarked at 240MB and it's pushing close to 1.1gb sustained and then I run out of CPU :-(

bb-qq, 2021-11-21:
Setting the MTU to 9000 may improve performance.

Also, if the file transfer speed is slower than the speed measured by iperf, changing the SMB version to v3 may improve the speed.

k-vq, 2021-12-05:
I have delayed buying a new Synology, hoping for them to release a model with built-in 2.5Gbe (similar to QNAP). But after Synology 2022 launch event with no sight of a DS222+, I decided to bite the bullet and buy a DS720+

With the same WD Red driver and the same Ugreen 2.5Gbe adapter, now I immediately see improve of file copy performance. File copy speed now about 190MB/s with Ethernet utilization about 1.7Gbps.

In conclusion, DS214play architecture is the main limitation for low speed. I am quite satisfied with the DS720+. Thank again @bb-qq for your wonderful work. Also, I really disappoint with Synology, for not only denying modern NIC for their NAS, and but also making it's much harder for the customers to use 3rd party adapters.

bb-qq, 2022-01-04:
Thank you for the performance report. This is helpful since it purely shows the difference between Synology hardware generations.

Since there seems to be no problem left, I close this issue.

closed 2022-01-04 by bb-qq