bb-qq/r8152 #148: 1621+ Compatibility?

Geddart, 2021-09-01:
Hey there, I was using this driver successfully on two 918+ Devices but I'm thinking about switching to a 1621+ with a Ryzen Processor. Would this driver work as well?

Sjekke, 2021-09-01:
If you can afford a Ds1621+, why not install a PCIE 10gb-card ... no trouble with drivers.

Geddart, 2021-09-02:
I have the 2.5Gbe Adapter but you're right - the E10G18-T1 is a better, more future-proof (10Gbe) solution.

Sjekke, 2021-09-04:
🤙

bb-qq, 2021-09-24:
Yes, the driver of v1000 can be loaded for this product and some users reported it works.
However, for stability, I would also strongly recommend using a PCIe-based interface as well.

If you want to save money, you can use AQC107 based compatible NICs such as [ASUS XG-C100C ](https://amzn.to/3lXFdNu) or [TPLINK TX401](https://amzn.to/2XN6x9r).

closed 2021-09-24 by bb-qq