bb-qq/r8152 #149: TUC-ET2G only running with USB 480MBit/s

doribell, 2021-09-03:
### Description of the problem

> I can't get a TUC-ET2G Adapter running in USB3 Mode with 5000MBit/s. It seems always to run with 480MBit/s only. I bought a  CLUB 3D CAC-1420 now and that one works fine. Tested with a DS920+ and DS218+ as well.

### Description of your products

> - DS920+
>     - Linux NAS1 4.4.180+ #41890 SMP Fri Jun 25 02:37:14 CST 2021 x86_64 GNU/Linux synology_geminilake_920+
>     - DSM 7.0-41890
> - TUC-ET2G

### Description of your environment

> - connect directly to a PC
>     - PC: NIC Realtek PCIe 2.5GbE Family Controller
> - cable category (Cat.6)

### Output of ``dmesg`` command

````
[671954.531607] r8152 2-1:1.0 eth2: carrier off
[671956.438042] xhci_hcd 0000:00:15.0: device is plugged out, empty URBs
[671956.445250] r8152 2-1:1.0 eth2: Stop submitting intr, status -71
[671956.452066] xhci_hcd 0000:00:15.0: WARN Event TRB for slot 9 ep 6 with no TDs queued?
[671956.497183] r8152 2-1:1.0 eth2: get_registers -71
[671956.506094] r8152 2-1:1.0 eth2: get_registers -71
[671956.514778] r8152 2-1:1.0 eth2: get_registers -71
[671956.523524] r8152 2-1:1.0 eth2: get_registers -71
[671957.028382] r8152 2-1:1.0 eth2: get_registers -110
[671957.533590] r8152 2-1:1.0 eth2: get_registers -110
[671958.038478] r8152 2-1:1.0 eth2: get_registers -110
[671958.546179] r8152 2-1:1.0 eth2: get_registers -110
[671959.051306] r8152 2-1:1.0 eth2: get_registers -110
[671959.267088] usb 2-1: USB disconnect, device number 4
[671959.272745] r8152 2-1:1.0 eth2: get_registers -19
[671959.278116] r8152 2-1:1.0 eth2: Get ether addr fail
[671959.556218] r8152 2-1:1.0 eth2: get_registers -110
[671959.943647] audit_printk_skb: 6 callbacks suppressed
[671959.949320] audit: type=1325 audit(1630684477.831:319): table=filter family=2 entries=43
[671959.967618] audit: type=1325 audit(1630684477.855:320): table=filter family=10 entries=11
[671959.987091] audit: type=1325 audit(1630684477.875:321): table=filter family=2 entries=34
[671960.001528] audit: type=1325 audit(1630684477.889:322): table=filter family=2 entries=38
[671960.016019] audit: type=1325 audit(1630684477.904:323): table=filter family=2 entries=39
[671960.030651] audit: type=1325 audit(1630684477.918:324): table=filter family=2 entries=40
[671960.045105] audit: type=1325 audit(1630684477.933:325): table=filter family=2 entries=41
[671960.059451] audit: type=1325 audit(1630684477.947:326): table=filter family=2 entries=42
[671960.073879] audit: type=1325 audit(1630684477.962:327): table=filter family=10 entries=4
[671960.088228] audit: type=1325 audit(1630684477.976:328): table=filter family=10 entries=8
[671964.296096] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[671964.437483] usb 1-1: not running at top speed
[671964.812254] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[671964.953376] usb 1-1: not running at top speed
[671965.327200] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[671965.468236] usb 1-1: not running at top speed
[671965.844046] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[671965.986171] usb 1-1: not running at top speed
[671966.360014] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[671966.501836] usb 1-1: not running at top speed
[671966.875890] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[671967.017031] usb 1-1: not running at top speed
[671967.149291] usb 1-1: reset high-speed USB device number 8 using xhci_hcd
[671967.291958] usb 1-1: not running at top speed
[671967.666755] usb 1-1: reset high-speed USB device number 8 using xhci_hcd
[671967.808871] usb 1-1: not running at top speed
[671968.182661] usb 1-1: reset high-speed USB device number 8 using xhci_hcd
[671968.324760] usb 1-1: not running at top speed
[671968.698613] usb 1-1: reset high-speed USB device number 8 using xhci_hcd
[671968.840699] usb 1-1: not running at top speed
[671969.214238] usb 1-1: reset high-speed USB device number 8 using xhci_hcd
[671969.355657] usb 1-1: not running at top speed
[671969.729099] usb 1-1: reset high-speed USB device number 8 using xhci_hcd
[671969.870506] usb 1-1: not running at top speed
[671969.964015] r8152 1-1:1.0 eth2: v2.15.0 (2021/04/15)
[671969.969672] r8152 1-1:1.0 eth2: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.
````

### Output of ``lsusb`` command

using: TUC-ET2G Adapter
````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         20f4:e02b:3004 00  2.10  480MBit/s 500mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 400000001)
  |__1-2         051d:0002:0106 00  1.10  1.5MBit/s 24mA 1IF  (American Power Conversion Back-UPS XS 700U   FW:924.Z3 .I USB FW:Z3  3B1830X60969  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0005FA1E13F221)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
````

using: CLUB 3D CAC-1420 Adapter
````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         051d:0002:0106 00  1.10  1.5MBit/s 24mA 1IF  (American Power Conversion Back-UPS XS 700U   FW:924.Z3 .I USB FW:Z3  3B1830X60969  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0005FA1E13F221)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
````

bb-qq, 2021-09-24:
What A to C conversion adapter are you using? One user reported that some conversion adapters have directional connectors and will not work with USB 3.0 depending on how they are connected.
https://github.com/bb-qq/r8152/issues/129#issuecomment-888814634

bb-qq, 2021-11-13:
Hmm, the only thing I come up with is a front/back direction of the A-C conversion adapter or a difference in signal quality.
But if you can connect it to your PC in the same way and have no problem, then that doesn't seem to be the cause either.

Please let me know if you find out anything after that.

Auberon3, 2022-07-25:
HI how do you get TUC-ET2G Adapter working in 480MB/s mode, I have only 2 usb3 ports on DS920+ and when I plug TUC-ET2G Adapter it shows up as 5000MB/s

Auberon3, 2022-07-25:
ok I found it, usbC-A converter has pin to avoid plugging the wrong way, if you plug it the wrong way it shows up 480MB/S but anyway for me in any case TUC-ET2G Adapter don't show up in interfaces ???

bb-qq, 2022-08-01:
> but anyway for me in any case TUC-ET2G Adapter don't show up in interfaces ???

This issue is handled in #231.

bb-qq, 2022-08-01:
I believe this issue is due to the orientation of the adapter. I am closing this issue as I have not heard back from the first issue author.

bb-qq, 2022-08-01:
Please feel free to reopen the issue if the original problem is not resolved.

closed 2022-08-01 by bb-qq