bb-qq/r8152 #151: PCIE 8125B not working

fulracoco, 2021-09-08:
asrock j3455 itx,  pcie 8125B
it's works on dsm 6.2.3, but the 2.15.0-3 DSM7 not working on dsm 7.0.1

root@DS918:~# dmesg | grep 'r81'
[    0.000000] PERCPU: Embedded 33 pages/cpu @ffff88047dc00000 s95128 r8192 d31848 u524288
[    0.000000] pcpu-alloc: s95128 r8192 d31848 u524288 alloc=1*2097152
[  573.472406] systemd[1]: Created slice r8152's slice.
[  573.478851] systemd[1]: Starting r8152's slice.
[  574.037700] r8168 Gigabit Ethernet driver 8.045.12-NAPI loaded
[  574.037734] r8168 0000:01:00.0: enabling device (0000 -> 0003)
[  574.052037] r8168: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[  574.052049] r8168  Copyright (C) 2017  Realtek NIC software team <nicfae@realtek.com> 
[  588.563567] r8168: eth0: link up 1 Gbps
[  610.203239] usbcore: registered new interface driver r8152
root@DS918:~# dmesg | grep 'net'
[    0.000000] Command line: BOOT_IMAGE=/zImage netif_num=2 earlycon=uart8250,io,0x3f8,115200n8 syno_hdd_powerup_seq=1 syno_hdd_detect=0 elevator=elevator root=/dev/md0 loglevel=15 log_buf_len=32M mac1=xxxxxxxxxx67 mac2=xxxxxxxxxx68 vid=0x0781 pid=0x5151 console=ttyS0,115200n8 sn=xxxxxxxxxxx04 earlyprintk syno_port_thaw=1 HddHotplug=0 withefi syno_hw_version=DS918+ vender_format_version=2
[    0.000000] Kernel command line: BOOT_IMAGE=/zImage netif_num=2 earlycon=uart8250,io,0x3f8,115200n8 syno_hdd_powerup_seq=1 syno_hdd_detect=0 elevator=elevator root=/dev/md0 loglevel=15 log_buf_len=32M mac1=xxxxxxxxxx67 mac2=xxxxxxxxxx68 vid=0x0781 pid=0x5151 console=ttyS0,115200n8 sn=xxxxxxxxxxx04 earlyprintk syno_port_thaw=1 HddHotplug=0 withefi syno_hw_version=DS918+ vender_format_version=2
[    0.000000] Internal netif num: 2
[    5.387025] <redpill/cmdline_delegate.c:382> Cmdline: BOOT_IMAGE=/zImage netif_num=2 earlycon=uart8250,io,0x3f8,115200n8 syno_hdd_powerup_seq=1 syno_hdd_detect=0 elevator=elevator root=/dev/md0 loglevel=15 log_buf_len=32M mac1=xxxxxxxxxx67 mac2=xxxxxxxxxx68 vid=0x0781 pid=0x5151 console=ttyS0,115200n8 sn=xxxxxxxxxxx04 earlyprintk syno_port_thaw=1 HddHotplug=0 withefi syno_hw_version=DS918+ vender_format_version=2
[    5.449763] <redpill/cmdline_delegate.c:393> Param #1: |netif_num=2|
[    5.457133] <redpill/cmdline_delegate.c:231> Declared network ifaces # as 2
[    8.957216] <redpill/sanitize_cmdline.c:107> Sanitized cmdline to: BOOT_IMAGE=/zImage netif_num=2 earlycon=uart8250,io,0x3f8,115200n8 syno_hdd_powerup_seq=1 syno_hdd_detect=0 root=/dev/md0 mac1=xxxxxxxxxx67 mac2=xxxxxxxxxx68 console=ttyS0,115200n8 sn=xxxxxxxxxxx04 HddHotplug=0 withefi syno_hw_version=DS918+ vender_format_version=2
[    9.401087] audit: initializing netlink subsys (disabled)
[  573.998297] Intel(R) Gigabit Ethernet Linux Driver - version 5.3.5.22
[  574.037700] r8168 Gigabit Ethernet driver 8.045.12-NAPI loaded
[  613.827542] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[  613.915140] IPVS: Creating netns size=2096 id=0
[  615.062541] Initializing XFRM netlink socket
root@DS918:~# lsmod | grep 'r81'
r8152                 216790  0 
r8168                 527330  0 
usbcore               201223  10 etxhci_hcd,r8152,usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,usbhid,xhci_hcd,xhci_pci
root@DS918:~# ls -la /usr/lib64/modules/r81*
-rw-r--r-- 1 root root 1720632 Sep  8 09:42 /usr/lib64/modules/r8152.ko
-rw-r--r-- 1 root root  644163 Aug  3 18:58 /usr/lib64/modules/r8168.ko

fulracoco, 2021-09-08:
DSM 7 netork setting-interface only one Lan RTL8111H, these is no 8125B

Marc176, 2021-09-10:
try as root ifconfig eth1 up and then re-check

bb-qq, 2021-09-24:
This driver does not support the chipset family of PCIe. Please see #80, #89.

closed 2021-09-24 by bb-qq