bb-qq/r8152 #152: Support for rtd1296 dropped?

Marc176, 2021-09-10:
Hi, 

have DS420j and i dont see new rtd1296 attempts ? No future for that SoC?

Marc176, 2021-09-10:
`[ 1001.464529] Synotify use 16384 event queue size
[ 1307.967713] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
[ 1307.974337] ------------[ cut here ]------------
[ 1307.979063] WARNING: at net/sched/sch_generic.c:306
[ 1307.984054] Modules linked in: btusb btrtl btintel btbcm hidp rfcomm bluetooth r8152(OE) cifs udf isofs loop tcm_loop(O) iscsi_target_mod(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) syno_hddmon(P) hid_generic usbhid usblp usb_storage uhci_hcd ds420j_synobios(PO) exfat(O) btrfs synoacl_vfs(PO) zlib_deflate hfsplus md4 hmac r8168(O) i2c_algo_bit vxlan ip6_udp_tunnel udp_tunnel fuse vfat fat aes_neon_blk aes_ce_blk gf128mul hid ablk_helper arc4 cryptd ecryptfs sha256_generic ecb authenc des_generic ansi_cprng cts md5 cbc dm_snapshot dm_bufio crc_itu_t crc_ccitt quota_v2 quota_tree psnap p8022 llc sit tunnel4 ip_tunnel ipv6 sg dwc3_rtk xhci_pci xhci_plat_hcd xhci_hcd ehci_pci ehci_hcd dwc3 usbcore usb_common
[ 1308.057353]  phy_rtk_usb3 phy_rtk_usb2 phy_rtk_rle0599 [last unloaded: bluetooth]

[ 1308.065255] CPU: 2 PID: 0 Comm: swapper/2 Tainted: P           OE   4.4.59+ #25556
[ 1308.073003] Hardware name: Realtek_RTD1296 (DT)
[ 1308.077641] task: ffffffc03c4fef80 ti: ffffffc03c504000 task.ti: ffffffc03c504000
[ 1308.085309] PC is at dev_watchdog+0x24c/0x25c
[ 1308.089769] LR is at dev_watchdog+0x24c/0x25c
[ 1308.094227] pc : [<ffffffc0007315ec>] lr : [<ffffffc0007315ec>] pstate: 60000145
[ 1308.101797] sp : ffffffc03c507c40
[ 1308.105189] x29: ffffffc03c507c40 x28: 00000000000000a0
[ 1308.110650] x27: 0000000000000008 x26: dead000000000200
[ 1308.116110] x25: ffffffc03df9d9b8 x24: ffffffc03df9d7b8
[ 1308.121570] x23: 00000000ffffffff x22: ffffffc000a91000
[ 1308.127030] x21: ffffffc000a46000 x20: ffffffc028054000
[ 1308.132489] x19: 0000000000000000 x18: 0000007fa55fdd90
[ 1308.137949] x17: 0000007fa8b2745c x16: ffffffc0002eb658
[ 1308.143409] x15: 0000007fa55fdd90 x14: 0fffffffffffffff
[ 1308.148870] x13: 0000000000000028 x12: 0101010101010101
[ 1308.154330] x11: ffffffc000a55000 x10: 0000000000000305
[ 1308.159790] x9 : ffffffc03c5079a0 x8 : 756f2064656d6974
[ 1308.165250] x7 : 0000000000000306 x6 : 0000000000000049
[ 1308.170710] x5 : 0000000000000000 x4 : 0000000000000000
[ 1308.176169] x3 : 0000000000000000 x2 : ffffffc03df9e228
[ 1308.181629] x1 : ffffffc03df9cae0 x0 : 0000000000000039

[ 1308.188614] Call trace:
[ 1308.191120] [<ffffffc0007315ec>] dev_watchdog+0x24c/0x25c
[ 1308.196651] [<ffffffc0002ea310>] call_timer_fn.isra.29+0x28/0x88
[ 1308.202800] [<ffffffc0002ea4f0>] run_timer_softirq+0x180/0x1f0
[ 1308.208774] [<ffffffc00029c8d0>] __do_softirq+0xf8/0x218
[ 1308.214212] [<ffffffc00029cc0c>] irq_exit+0x64/0x90
[ 1308.219207] [<ffffffc0002db424>] __handle_domain_irq+0x60/0xb0
[ 1308.225179] [<ffffffc000280cf8>] gic_handle_irq+0x60/0xb8
[ 1308.230706] Exception stack(0xffffffc03c507df0 to 0xffffffc03c507f20)
[ 1308.237300] 7de0:                                   ffffffc000a46000 0000008000000000
[ 1308.245317] 7e00: ffffffc03c507f50 ffffffc000284e70 0000000060000145 ffffffc0007ea000
[ 1308.253333] 7e20: 0000000000000000 0000000000000000 0000000000000001 ffffffc000a47000
[ 1308.261349] 7e40: 0000000000000001 0000000000000000 0000000000000810 00000001000189be
[ 1308.269365] 7e60: ffffffc03c4ff7f0 ffffffc03c507ec0 0000000000000810 00000000613b5866
[ 1308.277381] 7e80: 0000007f9c04e8a0 0000000000000059 0000007f9c1a9d20 0000007fa55fdd90
[ 1308.285397] 7ea0: ffffffc0002eb658 0000007fa8b2745c 0000007fa55fdd90 ffffffc000a46000
[ 1308.293413] 7ec0: ffffffc000a46a48 ffffffc000a2bac0 ffffffc000a2f048 ffffffc03c507f70
[ 1308.301429] 7ee0: ffffffc0007ea000 ffffffc000a46000 ffffffc03c504000 0000000000000000
[ 1308.309446] 7f00: 0000000000000000 ffffffc03c507f50 ffffffc000284e6c ffffffc03c507f50
[ 1308.317462] [<ffffffc000283f00>] el1_irq+0x80/0x100
[ 1308.322456] [<ffffffc0002ced60>] default_idle_call+0x1c/0x30
[ 1308.328249] [<ffffffc0002cef58>] cpu_startup_entry+0x194/0x1e4
[ 1308.334221] [<ffffffc00028cf7c>] secondary_start_kernel+0x11c/0x140
[ 1308.340637] [<00000000002810cc>] 0x2810cc
[ 1308.344740] ---[ end trace 0a1da600e80bd39d ]---
[ 1308.349495] r8152 3-1:1.0 eth1: Tx timeout
[ 1308.356073] r8152 3-1:1.0 eth1: Tx status -2
[ 1308.360472] r8152 3-1:1.0 eth1: Tx status -2
[ 1308.364870] r8152 3-1:1.0 eth1: Tx status -2
[ 1308.369283] r8152 3-1:1.0 eth1: Tx status -2
[ 1310.374654] r8152 3-1:1.0 eth1: get_registers -108
[ 1310.379632] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.384502] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.389401] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.394273] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.399159] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.404030] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.408911] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.414542] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.419431] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.424302] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.429184] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.434055] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.438936] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.443806] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.448686] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.455367] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.460299] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.465177] r8152 3-1:1.0 eth1: get_registers -71
[... cut ...] gisters -71
[ 1310.474946] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.479829] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.484701] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.489583] r8152 3-1:1.0 eth1: get_registers -71
[ 1310.524303] usb 3-1: reset SuperSpeed USB device number 2 using xhci-hcd
[ 1326.881897] r8152 3-1:1.0 eth1: carrier off
[ 1328.231854] init: dhcp-client (eth1) main process (16863) killed by TERM signal
[ 1329.225652] init: nmbd main process (22611) killed by TERM signal
`

rswiz, 2021-09-20:
> Hi,
> 
> have DS420j and i dont see new rtd1296 attempts ? No future for that SoC?

Hey Marc, check this out, I am running a 418j so I feel your pain.

https://github.com/bb-qq/r8152/issues/141#issuecomment-903207140

bb-qq, 2021-09-24:
Yes, I dropped this platform because none of the users reported the driver works.
Even if the driver works stably, the platform is so unpowerful that it will not be able to fully exploit the performance of multi-giga Ethernet devices.

nsa666666, 2021-09-24:
It worked actually fine with some version i've downloaded from the comments. But it does not launched on boot. And it did realy reached 250MB/s speed with samba. I have an DS218+.

rswiz, 2021-09-25:
> It worked actually fine with some version i've downloaded from the comments. But it does not launched on boot. And it did realy reached 250MB/s speed with samba. I have an DS218+.

I remember that too, had to downgrade for the driver to work and by work I mean communicate with my network before it died from a high CPU spike. I was using "r8152-rtd1296-2.15.0-1.spk" on DSM7, it just couldn't sustain the transmission and by that it would even while idling go off.

nsa666666, 2021-09-26:
That is strange. I could actually use it. Without crashes. And transfering files worked too. I could only not get it started on boot, and must make insmod every time.

bb-qq, 2021-11-13:
I have uploaded the driver again. However, there seem to be many reports of it not working.

closed 2021-11-13 by bb-qq