bb-qq/r8152 #156: UGreen 2.5Gbe USB Dongle is not working under Synology DS1019+ installed with DSM 7.0.1.42218

sychoi8888, 2021-10-02:
UGreen 2.5Gbe USB Dongle (Realtek R8156 chip) is not working under Synology DS1019+ installed with DSM 7.0.1.42218.  The driver has been installed but it is continuously reported to be manually disabled.

sychoi8888, 2021-10-03:
r8152-apollolake-2.15.0-4.apk has been used.

jglathe, 2021-10-12:
Do you have listing of lsmod, lsusb, dmesg and ifconfig for some debugging? I have one runing with Apollolake... DS218+.

colin19889, 2021-11-04:
I had some issues with my 2.5G adapter when I first updated to 7.0.1-42218. I reinstalled the driver (failed the first time and then I used the command via SSH). Afterwards DSM was reporting the correct IP address but I couldn't ping it. When I checked with ifconfig it was reporting the adapter but there was no static IP associated with it. I changed the static IP in DSM to something different and then back to the correct one and ifconfig reported the correct IP and it started working as it should. Just something to check if you run into issues.

bb-qq, 2021-11-13:
@sychoi8888 , could you run again and confirm this command was suceeded: `sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su`

@geraldlimw could you create a new issue using the bug report template?

geraldlimw, 2021-11-13:
@bb-qq, issue had been resolved. by using the front ports ;) thanks!

Jlozde, 2022-01-11:
I have 
UGreen 2.5Gbe USB Dongle (Realtek R8156 chip)
CAT6 short patch cable
DS920+(geminilake) 
DSM 7.0.1-42218
Used 2.15.0-4 
`sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su` command also applied
Device conencted to a tp-LINK TL-SG108-m2 2.5Gbe switch -> Negotiate 1Gbps
Device connected to my desktop pc -> Negotiate 1Gbps

Any idea what is wrong. @bb-qq

closed 2022-01-12 by bb-qq