bb-qq/r8152 #163: Asus USB-c2500 + DSM 7.0.1 on  DS 1520+ : Rcognized, and works. but the speed is low

aka80a, 2021-11-04:
### Description of the problem

The adapter is recognized, and works. But the speed is low, barely 190 Mbps. On my mac, the speed goes up to 1.2 Gigabits. I do not understand why I do not reach the advertised rates?

### Description of my products

> - Nas Synology DS 1520+
>     - DSM 7.0.1
      - NTEL Celeron J4125
 - Ethernet dongle: Asus USB-C2500
   - Drivers;  RTL8152/RTL8153/RTL8156 driver package for Synology NASes.
 Ethernet cable Cat8 
https://www.amazon.fr/gp/product/B0843R4812/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&th=1

aka80a, 2021-11-07:
> ### Description of the problem
> The adapter is recognized, and works. But the speed is low, barely 190 Mbps. On my mac, the speed goes up to 1.2 Gigabits. I do not understand why I do not reach the [...]

Well, after several tests, for several days, I have the same results. I think the drivers are not optimized enough. Can you review for an update after investigation and optimization? thank you

bb-qq, 2021-11-21:
Could you paste outputs of `demsg`, `lsusb` and `ifconfig -a` ?

fifolifo, 2021-12-04:
While I didn't create this thread, I do have this Asus USB-C2500 adapter connected to the front USB port of my apollolake DS1019+ running DSM 7.0.1-42218 and I can provide the requested outputs.  From my managed switch, I've verfied that the port this adapter is connected to is showing a link speed of 2.5Gb.  I have noticed it being laggy.  I assume that this is due to the retransmits errors that I see when performing an iperf3 test.  Let me know if you need any other outputs and thanks for your work.

**iperf3 output**
```
 % iperf3 -c 192.168.168.4
Connecting to host 192.168.168.4, port 5201
[  5] local 192.168.168.8 port 63619 connected to 192.168.168.4 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   216 MBytes  1.81 Gbits/sec   40   75.5 KBytes
[  5]   1.00-2.00   sec   177 MBytes  1.48 Gbits/sec   14    108 KBytes
[  5]   2.00-3.00   sec   220 MBytes  1.84 Gbits/sec   41    106 KBytes
[  5]   3.00-4.00   sec   236 MBytes  1.98 Gbits/sec   41   80.7 KBytes
[  5]   4.00-5.00   sec   236 MBytes  1.98 Gbits/sec   26    116 KBytes
[  5]   5.00-6.00   sec  70.2 MBytes   589 Mbits/sec   34   69.9 KBytes
[  5]   6.00-7.00   sec   166 MBytes  1.39 Gbits/sec   27   25.6 KBytes
[  5]   7.00-8.00   sec   235 MBytes  1.97 Gbits/sec   28    115 KBytes
[  5]   8.00-9.00   sec   190 MBytes  1.59 Gbits/sec   24    116 KBytes
[  5]   9.00-10.00  sec   215 MBytes  1.81 Gbits/sec   19   65.5 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.91 GBytes  1.64 Gbits/sec  294             sender
[  5]   0.00-10.01  sec  1.91 GBytes  1.64 Gbits/sec                  receiver

iperf Done.
% iperf3 -c 192.168.168.4 -R
Connecting to host 192.168.168.4, port 5201
Reverse mode, remote host 192.168.168.4 is sending
[  5] local 192.168.168.8 port 63621 connected to 192.168.168.4 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   194 MBytes  1.63 Gbits/sec
[  5]   1.00-2.00   sec   187 MBytes  1.57 Gbits/sec
[  5]   2.00-3.00   sec   180 MBytes  1.51 Gbits/sec
[  5]   3.00-4.00   sec   187 MBytes  1.57 Gbits/sec
[  5]   4.00-5.00   sec   191 MBytes  1.60 Gbits/sec
[  5]   5.00-6.00   sec   189 MBytes  1.58 Gbits/sec
[  5]   6.00-7.00   sec   196 MBytes  1.65 Gbits/sec
[  5]   7.00-8.00   sec   200 MBytes  1.68 Gbits/sec
[  5]   8.00-9.00   sec   195 MBytes  1.64 Gbits/sec
[  5]   9.00-10.00  sec   199 MBytes  1.67 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  1.88 GBytes  1.61 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.87 GBytes  1.61 Gbits/sec                  receiver

iperf Done.
% iperf3 -c 192.168.168.4
Connecting to host 192.168.168.4, port 5201
[  5] local 192.168.168.8 port 47163 connected to 192.168.168.4 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   113 MBytes   951 Mbits/sec   25    131 KBytes
[  5]   1.00-2.00   sec   254 MBytes  2.13 Gbits/sec   31    119 KBytes
[  5]   2.00-3.00   sec   243 MBytes  2.04 Gbits/sec   24    126 KBytes
[  5]   3.00-4.00   sec   243 MBytes  2.03 Gbits/sec   29    115 KBytes
[  5]   4.00-5.00   sec   231 MBytes  1.94 Gbits/sec   33   69.5 KBytes
[  5]   5.00-6.00   sec   171 MBytes  1.44 Gbits/sec   17   1.41 KBytes
[  5]   6.00-7.00   sec   150 MBytes  1.26 Gbits/sec   12    109 KBytes
[  5]   7.00-8.00   sec   228 MBytes  1.91 Gbits/sec   37   93.6 KBytes
[  5]   8.00-9.00   sec   238 MBytes  1.99 Gbits/sec   20    104 KBytes
[  5]   9.00-10.00  sec   233 MBytes  1.96 Gbits/sec   35    105 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.06 GBytes  1.77 Gbits/sec  263             sender
[  5]   0.00-10.01  sec  2.06 GBytes  1.76 Gbits/sec                  receiver

iperf Done.
% iperf3 -c 192.168.168.4 -R
Connecting to host 192.168.168.4, port 5201
Reverse mode, remote host 192.168.168.4 is sending
[  5] local 192.168.168.8 port 16437 connected to 192.168.168.4 port 5201
[ ID] Interval           Transfer     Bitrate
[... 763 lines cut ...]
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2681162 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2681162 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:262648079 (250.4 MiB)  TX bytes:262648079 (250.4 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun1000   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:169.254.207.97  P-t-P:169.254.207.97  Mask:255.255.248.0
          inet6 addr: fec0:a2b2:11::761/112 Scope:Site
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1400  Metric:1
          RX packets:188047 errors:0 dropped:0 overruns:0 frame:0
          TX packets:210184 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:58964008 (56.2 MiB)  TX bytes:30907441 (29.4 MiB)
```

bb-qq, 2022-01-04:
There have been previous reports of similar problems occurring when the power supply of the USB port is insufficient.

We don't know if the cause is the same, but using a self-powered hub that can be connected to an AC adapter may improve the situation.

fifolifo, 2022-01-04:
Thanks for the tip. I ended up replacing the Asus with a Club 3D adapter and it’s been working great ever since. Thanks for creating this project and for your development work on it.

avielc, 2022-05-22:
Hi @bb-qq -
Firstly, thanks for all your work on this
Secondly, sorry for reviving this topic for a very short time. 
I have the Asus-C2500 as well, the copy speed to the DS918+ (connected in the back) is only 320MBps faster (instead of 111mbps I get 148mbps) 
You think it's the power consumpter of this card? rather than some driver set up? 

Thanks, and sorry if I nudge again.

bb-qq, 2022-05-28:
Regarding the ASUS-C2500, you may be using the wrong side of the A-C conversion adapter. If that does not seem to be the case, I cannot say anything without more information. Could you please create a new Issue using the bug report template and fill in the required information?

avielc, 2022-05-28:
It'll take me a while to open an issue (not infront of the nas right now) 
could you elaborate on the " wrong side of the A-C conversion adapter"?
Thanks

bb-qq, 2022-05-28:
Sorry, I confused ASUS-C2500(Type-A) with ASUSTOR AS-U2.5G(Type-C). Please ignore the comment about the A-C conversion adapter.

avielc, 2022-05-28:
Ha! that makes sense :)
Thanks for that, I think I saw a post with comparison of network cards... can't find it for some reason (someone posted there a 200mbps transfer rate - I think I'll look it up and simply return this one to the seller