bb-qq/r8152 #165: DiskStation 1812+ with a Sabrent NT-S25G doesn't work at 2.5Gbe

dbistriceanu, 2021-11-14:
### Description of the problem

I have an old Synology DiskStation and it seems that it doesn't actually work at 2.5Gbe with a  Sabrent NT-S25G USB adapter, even though both it and the switch indicate that was the auto negotiated speed.

I've included below the iperf3 results between my workstation and the same adapter running on Synology first and then on my laptop.

```
❯ iperf3 -c 192.168.107.200
Connecting to host 192.168.107.200, port 5201
[  5] local 192.168.107.64 port 54260 connected to 192.168.107.200 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   130 MBytes  1.09 Gbits/sec    0   1010 KBytes
[  5]   1.00-2.00   sec   128 MBytes  1.07 Gbits/sec    0   1.14 MBytes
[  5]   2.00-3.00   sec   124 MBytes  1.04 Gbits/sec    0   1.45 MBytes
[  5]   3.00-4.00   sec   126 MBytes  1.06 Gbits/sec    0   2.40 MBytes
[  5]   4.00-5.00   sec   129 MBytes  1.08 Gbits/sec    0   2.40 MBytes
[  5]   5.00-6.00   sec   128 MBytes  1.07 Gbits/sec    0   2.40 MBytes
[  5]   6.00-7.00   sec   129 MBytes  1.08 Gbits/sec    0   2.40 MBytes
[  5]   7.00-8.00   sec   129 MBytes  1.08 Gbits/sec    0   2.40 MBytes
[  5]   8.00-9.00   sec   128 MBytes  1.07 Gbits/sec    0   2.40 MBytes
[  5]   9.00-10.00  sec   128 MBytes  1.07 Gbits/sec    0   2.40 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.25 GBytes  1.07 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.24 GBytes  1.07 Gbits/sec                  receiver

iperf Done.
```
vs
```
❯ iperf3 -c 192.168.107.178
Connecting to host 192.168.107.178, port 5201
[  5] local 192.168.107.64 port 41470 connected to 192.168.107.178 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   274 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   1.00-2.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   2.00-3.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   3.00-4.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   4.00-5.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   5.00-6.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   6.00-7.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   7.00-8.00   sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
[  5]   8.00-9.00   sec   272 MBytes  2.28 Gbits/sec    0    220 KBytes
[  5]   9.00-10.00  sec   273 MBytes  2.29 Gbits/sec    0    220 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.66 GBytes  2.29 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  2.66 GBytes  2.29 Gbits/sec                  receiver

iperf Done.
```
So the same adapter works just fine at 2.5Gbe if I plug it in on my laptop, using the same cable and switch port.

### Description of your products

Synology DiskStation 1812+ using the latest available DSM 6.2.4-25556 Update 2 (no support for DSM 7):
```
admin@hds1812:~$ uname -a
Linux hds1812 3.10.105 #25556 SMP Sat Aug 28 02:15:10 CST 2021 x86_64 GNU/Linux synology_cedarview_1812+
```

[Sabrent NT-S25G](https://www.sabrent.com/product/NT-S25G/usb-type-a-or-type-c-to-2-5-gigabit-ethernet-adapter-10-100-1000-2500-nt-s25g/)

### Description of your environment

[Netgear GS110MX switch](https://www.netgear.com/business/wired/switches/unmanaged/gs110mx/)
[Intel X550T ](https://ark.intel.com/content/www/us/en/ark/products/88208/intel-ethernet-converged-network-adapter-x550t1.html)

### Output of ``dmesg`` command

````
[ 5219.768866] usb 8-2: new SuperSpeed USB device number 3 using xhci_hcd
[ 5219.787565] usb 8-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 5219.944635] usb 8-2: reset SuperSpeed USB device number 3 using xhci_hcd
[ 5219.963729] usb 8-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 5220.002541] xhci_hcd 0000:05:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800aa127cc0
[ 5220.011679] xhci_hcd 0000:05:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800aa127d00
[ 5220.020834] xhci_hcd 0000:05:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800aa127d40
[ 5220.587292] r8152 8-2:1.0 eth2: v2.14.0 (2020/09/24)
[ 5220.592375] r8152 8-2:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 5220.608182] r8152 8-2:1.0 eth2: chip rev 12
````

### Output of ``lsusb`` command

````
admin@hds1812:~$ lsusb | grep -i realtek
  |__8-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
````

### Output of ``ifconfig -a`` command

````
admin@hds1812:~$ ifconfig -a eth2
eth2      Link encap:Ethernet  HWaddr 00:24:27:88:0C:51
          inet addr:192.168.107.200  Bcast:192.168.107.255  Mask:255.255.255.0
          inet6 addr: fd66:f97d:36bb:1:224:27ff:fe88:c51/64 Scope:Global
          inet6 addr: fe80::224:27ff:fe88:c51/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4764318 errors:0 dropped:0 overruns:0 frame:0
          TX packets:899425 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6919539023 (6.4 GiB)  TX bytes:958405940 (914.0 MiB)
````

### Output of ``ethtool`` command

````
admin@hds1812:~$ sudo ethtool eth2
Settings for eth2:
	Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseX/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseX/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Link partner advertised link modes:  100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Speed: 2500Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
admin@hds1812:~$
````

### Output of ``usb-devices`` command

````
❯ usb-devices | grep '2.5G' -C5

T:  Bus=04 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  9 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  3
P:  Vendor=0bda ProdID=8156 Rev=30.00
S:  Manufacturer=Realtek
S:  Product=USB 10/100/1G/2.5G LAN
S:  SerialNumber=000000001
C:  #Ifs= 2 Cfg#= 2 Atr=a0 MxPwr=512mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=02(commc) Sub=0d Prot=00 Driver=cdc_ncm
E:  Ad=83(I) Atr=03(Int.) MxPS=  16 Ivl=16ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm

````

Sjekke, 2021-11-14:
My 50 cents : Your Synology only supports SATA-300 so your throughput is lower then your Desktop or Laptop (I think).

dbistriceanu, 2021-11-14:
SATA-300 is 3Gb/s, which should still allow for more than the hard 1Gb/s cap seen in iperf3. To me, it seems it doesn't actually connect at 2.5Gbe but 1Gbe, which it is able to saturate with iperf3.

dbistriceanu, 2021-11-16:
I suspect this is a driver issue because for the same adapter on a PC connected to the same switch `ethtool` reports
```
❯ sudo ethtool enp0s20u1
Settings for enp0s20u1:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  100baseT/Full
	                                     1000baseT/Half 1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Supports Wake-on: pumbg
	Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
```

Notice that the `Supported link modes` and `Advertised link modes` both list `2500baseT/Full`, and that's also a `Link partner advertised link mode`, which I guess would explain why it does work at 2.5Gbe in that case.

While on Synology the driver supports and advertises `2500baseX/Full`, which apparently is not advertised by the switch, and thus the link is not established at 2.5Gbe? Nor is `2500baseX/Full` supported and advertised by the driver included in the latest kernel used on that PC, as it can be seen above. Presumably because according to the `ethtool` man page it is _not supported by IEEE standards_?

bb-qq, 2021-11-21:
The kernel version used by the DS1812+ is older, so Multi-Giga support may be incomplete, but the link speed itself seems to be 2.5gbps. Because 1.07gbps of performance should not be obtainable on a 1000base link.

Can you try increasing the MTU to 9000?

dbistriceanu, 2021-11-21:
iperf3 seems to take a complete nosedive at 9000 MTU
```admin@hds1812:~$ ip a s eth2
6: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc pfifo_fast state UP qlen 1000
    link/ether 00:24:27:88:0c:51 brd ff:ff:ff:ff:ff:ff
    inet 192.168.107.200/24 brd 192.168.107.255 scope global eth2
       valid_lft forever preferred_lft forever
    inet6 fd66:f97d:36bb:1:224:27ff:fe88:c51/64 scope global dynamic
       valid_lft 1694sec preferred_lft 1694sec
    inet6 fe80::224:27ff:fe88:c51/64 scope link
       valid_lft forever preferred_lft forever
   ```
   ```~
❯ iperf3 -c 192.168.107.200
Connecting to host 192.168.107.200, port 5201
[  5] local 192.168.107.64 port 56272 connected to 192.168.107.200 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  5.50 MBytes  46.1 Mbits/sec    0    318 KBytes
[  5]   1.00-2.00   sec  1.25 MBytes  10.5 Mbits/sec    0    318 KBytes
[  5]   2.00-3.00   sec  1.25 MBytes  10.5 Mbits/sec    0    318 KBytes
[  5]   3.00-4.00   sec  2.50 MBytes  21.0 Mbits/sec    0    318 KBytes
[  5]   4.00-5.00   sec  1.25 MBytes  10.5 Mbits/sec    0    318 KBytes
[  5]   5.00-6.00   sec  1.25 MBytes  10.5 Mbits/sec    0    318 KBytes
[  5]   6.00-7.00   sec  2.50 MBytes  21.0 Mbits/sec    0    318 KBytes
[  5]   7.00-8.00   sec  1.25 MBytes  10.5 Mbits/sec    0    318 KBytes
[  5]   8.00-9.00   sec  2.50 MBytes  21.0 Mbits/sec    0    318 KBytes
[  5]   9.00-10.00  sec  1.25 MBytes  10.5 Mbits/sec    0    318 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  20.5 MBytes  17.2 Mbits/sec    0             sender
[  5]   0.00-10.10  sec  17.5 MBytes  14.5 Mbits/sec                  receiver

iperf Done.
```

bb-qq, 2021-11-21:
When connecting directly to the PC without through the switch, and setting both to MTU 9000, will you get the same result?

dbistriceanu, 2021-11-21:
Similar, yes
```
~
❯ ip a | grep 'mtu 9000'
3: enp0s20u1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc fq_codel state UP group default qlen 1000
```
```
admin@hds1812:~$ ip a | grep 'mtu 9000'
6: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc pfifo_fast state UP qlen 1000
```
```~
❯ iperf3 -c 192.168.107.200
Connecting to host 192.168.107.200, port 5201
[  5] local 192.168.107.64 port 38762 connected to 192.168.107.200 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.56 MBytes  21.5 Mbits/sec   14   43.7 KBytes
[  5]   1.00-2.00   sec  2.21 MBytes  18.5 Mbits/sec   15   35.0 KBytes
[  5]   2.00-3.00   sec  2.15 MBytes  18.0 Mbits/sec   12   26.2 KBytes
[  5]   3.00-4.00   sec  2.33 MBytes  19.5 Mbits/sec   15   35.0 KBytes
[  5]   4.00-5.00   sec  2.15 MBytes  18.0 Mbits/sec   16   35.0 KBytes
[  5]   5.00-6.00   sec  2.27 MBytes  19.0 Mbits/sec   14   35.0 KBytes
[  5]   6.00-7.00   sec  2.27 MBytes  19.0 Mbits/sec   11   43.7 KBytes
[  5]   7.00-8.00   sec  2.15 MBytes  18.0 Mbits/sec   12   43.7 KBytes
[  5]   8.00-9.00   sec  2.33 MBytes  19.5 Mbits/sec   13   26.2 KBytes
[  5]   9.00-10.00  sec  1.91 MBytes  16.0 Mbits/sec   12   26.2 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  22.3 MBytes  18.7 Mbits/sec  134             sender
[  5]   0.00-10.00  sec  22.1 MBytes  18.5 Mbits/sec                  receiver

iperf Done.
```

bb-qq, 2021-11-21:
Hmm..., the last question, how about lower values such as 2500~4000?

dbistriceanu, 2021-11-21:
Direct connection still, same MTU on both ends:
```
admin@hds1812:~$ ip a | grep 'mtu 4000'
6: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 4000 qdisc pfifo_fast state UP qlen 1000
```
```
~
❯ iperf3 -c 192.168.107.200
Connecting to host 192.168.107.200, port 5201
[  5] local 192.168.107.64 port 38782 connected to 192.168.107.200 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  77.1 MBytes   647 Mbits/sec  137   42.4 KBytes
[  5]   1.00-2.00   sec  73.7 MBytes   619 Mbits/sec  132    158 KBytes
[  5]   2.00-3.00   sec  75.7 MBytes   635 Mbits/sec  146   73.3 KBytes
[  5]   3.00-4.00   sec  73.1 MBytes   613 Mbits/sec  128   38.6 KBytes
[  5]   4.00-5.00   sec  72.4 MBytes   607 Mbits/sec  130   81.0 KBytes
[  5]   5.00-6.00   sec  68.8 MBytes   577 Mbits/sec  142   73.3 KBytes
[  5]   6.00-7.00   sec  71.3 MBytes   598 Mbits/sec  134   50.1 KBytes
[  5]   7.00-8.00   sec  74.1 MBytes   622 Mbits/sec  122   42.4 KBytes
[  5]   8.00-9.00   sec  77.3 MBytes   648 Mbits/sec  131   42.4 KBytes
[  5]   9.00-10.00  sec  75.8 MBytes   636 Mbits/sec  146   50.1 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   739 MBytes   620 Mbits/sec  1348             sender
[  5]   0.00-10.00  sec   738 MBytes   619 Mbits/sec                  receiver

iperf Done.
```
```
admin@hds1812:~$ ip a | grep 'mtu 2500'
6: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2500 qdisc pfifo_fast state UP qlen 1000
```
```
~
❯ iperf3 -c 192.168.107.200
Connecting to host 192.168.107.200, port 5201
[  5] local 192.168.107.64 port 38796 connected to 192.168.107.200 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  74.8 MBytes   627 Mbits/sec  111   74.1 KBytes
[  5]   1.00-2.00   sec  72.7 MBytes   610 Mbits/sec  114    172 KBytes
[  5]   2.00-3.00   sec  79.6 MBytes   668 Mbits/sec  120    108 KBytes
[  5]   3.00-4.00   sec  73.5 MBytes   617 Mbits/sec   98   45.4 KBytes
[  5]   4.00-5.00   sec  70.5 MBytes   592 Mbits/sec  129   52.6 KBytes
[  5]   5.00-6.00   sec  69.8 MBytes   586 Mbits/sec  149   47.8 KBytes
[  5]   6.00-7.00   sec  76.9 MBytes   645 Mbits/sec  145   52.6 KBytes
[  5]   7.00-8.00   sec  80.6 MBytes   676 Mbits/sec  166   52.6 KBytes
[  5]   8.00-9.00   sec  74.8 MBytes   627 Mbits/sec  153    100 KBytes
[  5]   9.00-10.00  sec  73.0 MBytes   613 Mbits/sec  191   57.4 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   746 MBytes   626 Mbits/sec  1376             sender
[  5]   0.00-10.00  sec   745 MBytes   625 Mbits/sec                  receiver

iperf Done.
```