bb-qq/r8152 #169: DS920+ not seeing nic in virtual machine manager.

Oq6Jw1Oi4jHR1L, 2021-12-12:
### Description of the problem

> I installed an USB 3.0 2.5G NIC and it shows up in the main networking but I am not able to select it from the list of available adapters in virtual machine manager.

[image]

[image]

### Description of your products

> - DS920+
>     - Linux daemonx 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_920+
>     - DSM version  DSM 7.0.1-42218
> - EDUP 2.5G USB C & USB 3.0 to Ethernet Adapter RJ45 Gigabit LAN

### Description of your environment

### Output of ``dmesg`` command

````

[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.180+ (root@build13) (gcc version 7.5.0 (GCC) ) #42218 SMP Mon Oct 18 19:16:55 CST 2021
[    0.000000] Command line: root=/dev/md0 intel_iommu=igfx_off netif_num=2 HddEnableDynamicPower=1 syno_hw_version=DS920+ vender_format_version=2 synoboot2 sn=2070SBRMSMVES macs=001132d50b97,001132d50b98 syno_ttyS0=pciserial,0x0:0x18.0x2,115200 syno_ttyS1=pciserial,0x0:0x18.0x0,115200
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000070f9dfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000070f9e000-0x000000007149dfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000007149e000-0x000000007789dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007789e000-0x0000000077a9dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000077a9e000-0x0000000077addfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000077ade000-0x0000000078d5efff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078d5f000-0x000000007965efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007965f000-0x0000000079bfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079c00000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d3709fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000057fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x01de9fb8-0x01de9fc7] usable ==> usable
[    0.000000] e820: update [mem 0x59b62040-0x59b62989] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001de9fb7] usable
[    0.000000] reserve setup_data: [mem 0x0000000001de9fb8-0x0000000001de9fc7] usable
[    0.000000] reserve setup_data: [mem 0x0000000001de9fc8-0x000000000fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000012151000-0x0000000059b6203f] usable
[    0.000000] reserve setup_data: [mem 0x0000000059b62040-0x0000000059b62989] usable
[... 1288 lines cut ...]
tap021132 Link encap:Ethernet  HWaddr 2E:4C:27:C2:26:A6
          inet6 addr: fe80::2c4c:27ff:fec2:26a6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:32 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7168 (7.0 KiB)  TX bytes:5611 (5.4 KiB)

tap021132 Link encap:Ethernet  HWaddr 76:EE:BD:D5:65:F2
          inet6 addr: fe80::74ee:bdff:fed5:65f2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:32 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7114 (6.9 KiB)  TX bytes:5611 (5.4 KiB)

tap021132 Link encap:Ethernet  HWaddr 36:29:3C:AD:1F:76
          inet6 addr: fe80::3429:3cff:fead:1f76/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1133 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:53338 (52.0 KiB)  TX bytes:5611 (5.4 KiB)

````

Oq6Jw1Oi4jHR1L, 2021-12-12:
Also wanted to add that I thought it was the bond I had, so I had removed them along with the vswitches in virtual machine manager. I notied that I'm not able to recreate the bond.

[image]

bb-qq, 2022-01-04:
The current release does not support open vSwitch. I will release a supported version in near future.
If you can't wait for that, PR #134 will help.

bb-qq, 2022-01-09:
I released a new version that includes PR #134.
Please feel free to re-open this issue if you still have a problem with that new driver.

closed 2022-01-09 by bb-qq

kebot, 2023-02-11:
I still have the same problem with DS920+ DSM 7.2 with latest version(2.16.3) of this driver

ampersandru, 2024-10-10:
@bb-qq  Just installed this driver and the UGreen 2.5gb USB adapter works in DSM, but it still doesn't show up in VM network settings