bb-qq/r8152 #171: Can

CaulyKan, 2021-12-14:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

CaulyKan, 2021-12-14:
Any diagnose ideas are welcome, thanks a lot!

bb-qq, 2022-01-04:
So far I have not come up with a good way to investigate this problem.

There have been many other reports of problems with the rtd1296 platform, so there may be a problem at the hardware level.

CaulyKan, 2022-01-04:
Thanks a lot for the reply, I'll close this issue and open a new one when new progress is made.

closed 2022-01-04 by CaulyKan