bb-qq/r8152 #172: Connection stops after some time! Driver Bug ?

DJGIG83, 2021-12-16:
I`ve installed the 2.15.0-1 Realtek RTL8152/Rtl8153 driver for my RTL8153 Gigabit Thernet Adapter on my Xpenology 3617xs.
Everything works fine till, the second  Lan (LAN 2) is not visible anymore after some hours or sometimes minutes.
After i stop the driver in the Application Center and re-run the driver again its there again and works .
What can i do to repair this fault ?
I tried it with fix ip and dhcp ( same fault )
I think it`s a driver problem .
or witch version (even older) works for my setup ?
B.r.

HebeROF, 2021-12-19:
Have had the same problem this morning. 

Used the 2.15.0.-3 on my DS718+ running DSM 7
2.5 Gbit USB is: https://www.amazon.de/gp/product/B07Q626XK2/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

NAS wasn't in my network table anymore. 
NAS shutdown and restart failed. Rebooted all network routers and so one.

had to connect my NAS directly to my Macbook to stop the driver in the Application Center and did a restart. from this point everything was fine again.

DJGIG83, 2021-12-21:
i solved it i just used a other usb port :)

bb-qq, 2022-01-04:
Synology devices seem to have various problems with USB3, but in most cases using the front port works relatively stably. I'm not sure why, but I assume it's because the front USB ports are directly soldered to the motherboard.

I close this issue.

closed 2022-01-04 by bb-qq

Sjekke, 2022-01-04:
I use the back usb on my 916+ and don't have issues