bb-qq/r8152 #173: DS716+ not working with any RTL-based adapter

jamesu, 2021-12-18:
### Description of the problem

Network interface doesn't appear despite ethernet dongle lighting up and aqc111 device driver loading.

This is using version 1.3.3.0-4.

### Description of your products

DS716+
(also have a DX513-1 expansion unit plugged into the esata port)

Kernel: Linux 3.10.108 #42218 SMP Mon Oct 18 19:15:17 CST 2021 x86_64 GNU/Linux synology_braswell_716+
DSM: 7.0.1-42218.

Two products tested:

* https://www.amazon.co.uk/gp/product/B084L4JL9K
* https://www.amazon.co.uk/D-Link-DUB-E250-Ethernet-Compatible-Thunderbolt/dp/B08M9GXDY8

(both reportedly use RTL8156B chips)

### Description of your environment

For testing, plugged into a 1gbit dlink switch using cat.5e. Devices are being plugged into the USB3 ports on the back.

### Output of ``dmesg`` command

(confidential info omitted)

````

[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.108 (root@build17) (gcc version 7.5.0 (GCC) ) #42218 SMP Mon Oct 18 19:15:17 CST 2021
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007a4befff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] e820: update [mem 0x01d87120-0x01d8712f] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001d8711f] usable
[    0.000000] reserve setup_data: [mem 0x0000000001d87120-0x0000000001d8712f] usable
[    0.000000] reserve setup_data: [mem 0x0000000001d87130-0x000000007a4befff] usable
[    0.000000] reserve setup_data: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x7bffe014  SMBIOS=0x7a850000 
[    0.000000] efi: mem00: type=3, attr=0xf, range=[0x0000000000000000-0x0000000000001000) (0MB)
[... 782 lines cut ...]

eth1      Link encap:Ethernet  HWaddr xxxx  
          inet addr:169.254.86.56  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3536 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3536 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:481261 (469.9 KiB)  TX bytes:481261 (469.9 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

Sjekke, 2021-12-18:
I think you are using the wrong driver ...

https://github.com/bb-qq/r8152/releases/tag/2.15.0-4

jamesu, 2021-12-18:
Well spotted, I must have followed the wrong link at some stage and gotten confused.

Installed the right driver and eth2 now appears with Device 1 plugged in. 

````
[  133.760619] r8152 4-2:1.0 eth2: v2.15.0 (2021/04/15)
[  133.766263] r8152 4-2:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.
[  133.783709] r8152 4-2:1.0 eth2: chip rev 14
[  136.553439] r8152 4-2:1.0 eth2: carrier on
```` 

Device 2 (the dlink adapter) however still doesn't work. Driver is still present as evidenced by dmesg:

```
[  121.685278] r8152: module verification failed: signature and/or required key missing - tainting kernel
[  121.696625] usbcore: registered new interface driver r8152
```

The only info I can seem to find about the dlink adapter on my linux box is: VendorId=2001, idProduct=b301. I'm beginning to suspect either its not "Based on Realtek" or its using some obscure chipset not supported by the driver. I'm not sure how to diagnose this further without opening up the device.

In any event, the fact that Device 1 works is sufficient for me, the only reason I checked Device 2 was because it was supposedly "Based on Realtek".

But still, i'm happy to diagnose Device 2 further assuming its actually relevant in terms of this project.

Sjekke, 2021-12-18:
👍

bb-qq, 2022-01-04:
Currently, using multiple devices is not supported. So I merge this issue to #140.

closed 2022-01-04 by bb-qq