bb-qq/r8152 #174: Traffic issues when using the internal port bonding

Joe-LAI24, 2021-12-18:
Machine:DS920+
Network card chipset : RTL8156

When i enable the bonding at the DS920+ internal lan port. the SMB and iscsi upload traffic will be using the bonding port,the download traffic will be use the usb network card.

It means when i using my PC to access the NAS,the upload speed can up to 2.5Gbps,the download speed will be limited by an gigabit speed.

I try to change the following file via SSH
/etc/synoinfo.conf
vi /etc.defaults/synoinfo.conf
Change the maxlanport=“2” to maxlanport=“3”.
But nothing happend.

And i can't see the usb network card at the Control Panel - Network - Network Interface - Services order, and the VMM.

This issues maybe happend after update to the DSM 7.0.1-42218,i rember everything should be works fines when using the DSM 7.0

When i disable the internal port bonding ,the usb network card works properlly again.

bb-qq, 2022-01-04:
> It means when i using my PC to access the NAS,the upload speed can up to 2.5Gbps,the download speed will be limited by an gigabit speed.

I guess this is a routing problem. When we use two physical/bonded adapters, we must separate both subnets.
Could you illustrate your entire network configuration?

> And i can't see the usb network card at the Control Panel - Network - Network Interface - Services order, and the VMM.

The current version of drivers does not support open vSwitch. I will release a supported version in near future.

bb-qq, 2022-01-23:
> The current version of drivers does not support open vSwitch. I will release a supported version in near future.

The Open vSwitch support was added in the 2.15.0-5 release.

Since I haven't heard back from you, I'm closing this issue, but feel free to reopen it if you still have any problems.

closed 2022-01-23 by bb-qq