bb-qq/r8152 #175: DS416slim DSM 7.0.1-42218 Adapter visible in OS, but "ifconfig eth2 up" results in Error "SIOCSIFFLAGS: Cannot allocate memory"

Steff-OG, 2021-12-23:
### Description of the problem

 Adapter visible in OS, but "ifconfig eth2 up" results in Error "SIOCSIFFLAGS: Cannot allocate memory"

root@SAN02:~# ifconfig eth2 up
SIOCSIFFLAGS: Cannot allocate memory

### Description of your products

Linux SAN02 3.10.108 #42218 SMP Mon Oct 18 19:14:53 CST 2021 armv7l GNU/Linux synology_armada38x_ds416slim

DSM 7.0.1-42218

### Description of your environment

### Output of ``dmesg`` command

[  646.129601] ifconfig: page allocation failure: order:4, mode:0x40d0
[  646.136030] CPU: 0 PID: 17571 Comm: ifconfig Tainted: PF          O 3.10.108 #42218
[  646.143808] [<80014e48>] (unwind_backtrace+0x0/0xe0) from [<80011e8c>] (show_stack+0x10/0x14)
[  646.152402] [<80011e8c>] (show_stack+0x10/0x14) from [<800a0574>] (warn_alloc_failed+0xd8/0x110)
[  646.161285] [<800a0574>] (warn_alloc_failed+0xd8/0x110) from [<800a289c>] (__alloc_pages_nodemask+0x5a8/0x7a8)
[  646.171373] [<800a289c>] (__alloc_pages_nodemask+0x5a8/0x7a8) from [<7f69fc80>] (alloc_rx_agg+0x4c/0x124 [r8152])
[  646.181745] [<7f69fc80>] (alloc_rx_agg+0x4c/0x124 [r8152]) from [<7f6ab360>] (rtl8152_open+0xac/0x3dc [r8152])
[  646.191877] [<7f6ab360>] (rtl8152_open+0xac/0x3dc [r8152]) from [<80367598>] (__dev_open+0x9c/0x100)
[  646.201088] [<80367598>] (__dev_open+0x9c/0x100) from [<803677d8>] (__dev_change_flags+0x94/0x128)
[  646.210306] [<803677d8>] (__dev_change_flags+0x94/0x128) from [<803678e0>] (dev_change_flags+0x10/0x4c)
[  646.221965] [<803678e0>] (dev_change_flags+0x10/0x4c) from [<803bee44>] (devinet_ioctl+0x604/0x70c)
[  646.233711] [<803bee44>] (devinet_ioctl+0x604/0x70c) from [<8034f474>] (sock_ioctl+0x128/0x288)
[  646.242875] [<8034f474>] (sock_ioctl+0x128/0x288) from [<800eeed8>] (do_vfs_ioctl+0x564/0xa74)
[  646.251674] [<800eeed8>] (do_vfs_ioctl+0x564/0xa74) from [<800ef488>] (SyS_ioctl+0xa0/0xb0)
[  646.262176] [<800ef488>] (SyS_ioctl+0xa0/0xb0) from [<8000e1e0>] (ret_fast_syscall+0x0/0x38)
[  646.270917] Mem-info:
[  646.273421] Normal per-cpu:
[  646.276289] CPU    0: hi:  186, btch:  31 usd:  31
[  646.281255] CPU    1: hi:  186, btch:  31 usd: 171
[  646.288032] active_anon:19815 inactive_anon:19842 isolated_anon:0
                active_file:7745 inactive_file:7447 isolated_file:0
                unevictable:265 dirty:35 writeback:95 unstable:0
                free:36947 slab_reclaimable:4452 slab_unreclaimable:22244
                mapped:14551 shmem:7386 pagetables:3865 bounce:0
                free_cma:0
[  646.321449] Normal free:150388kB min:8192kB low:10240kB high:12288kB active_anon:79080kB inactive_anon:79124kB active_file:29256kB inactive_file:29224kB unevictable:1060kB isolated(anon):0kB isolated(file):128kB present:524288kB managed:509048kB mlocked:1060kB dirty:140kB writeback:0kB mapped:57748kB shmem:29544kB slab_reclaimable:17808kB slab_unreclaimable:88976kB kernel_stack:3984kB pagetables:15460kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:88 all_unreclaimable? no
[  646.365549] lowmem_reserve[]: 0 0
[  646.369043] Normal: 7189*4kB (UEM) 8986*8kB (UEM) 2923*16kB (UEMR) 78*32kB (MR) 0*64kB 0*128kB 0*256kB 1*512kB (R) 0*1024kB 0*2048kB 0*4096kB = 150420kB
[  646.383214] 25780 total pagecache pages
[  646.387111] 4252 pages in swap cache
[  646.390703] Swap cache stats: add 57183, delete 52975, find 36306/37325
[  646.397491] Free swap  = 1914112kB
[  646.401779] Total swap = 2097084kB
[  646.410560] 131072 pages of RAM
[  646.414404] 40377 free pages
[  646.417391] 2911 reserved pages
[  646.420901] 26700 slab pages
[  646.424159] 594625 pages shared
[  646.427556] 4223 pages swap cached

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
  |__1-1         05e3:0610:9306 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.1 Hub d6a3ebcbfad4e6a4f7a2) hub
    |__1-1.4     05e3:0610:9306 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.1 Hub d6a3ebcbfad4e6a4f7a0) hub
      |__1-1.4.4 05e3:0610:9306 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.1 Hub d6a3ebcbfad4e6a4f7a1) hub
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
  |__2-1         05e3:0620:9306 09  3.10 5000MBit/s 0mA 1IF  (GenesysLogic USB3.1 Hub d6a3ebcbfad4e7a5f6a3) hub
    |__2-1.4     05e3:0620:9306 09  3.10 5000MBit/s 0mA 1IF  (GenesysLogic USB3.1 Hub d6a3ebcbfad4e7a5f6a0) hub
      |__2-1.4.4 05e3:0620:9306 09  3.10 5000MBit/s 0mA 1IF  (GenesysLogic USB3.1 Hub d6a3ebcbfad4e7a5f6a1) hub
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
  **|__4-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000000)**
|__usb5          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd f1058000.usb) hub

### Output of ``ifconfig -a`` command

bond0     Link encap:Ethernet  HWaddr 00:11:32:8E:5D:2B
          inet addr:192.168.178.230  Bcast:192.168.178.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:48863 errors:0 dropped:0 overruns:0 frame:0
          TX packets:36490 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:6780956 (6.4 MiB)  TX bytes:7889977 (7.5 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:8E:5D:2B
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:35135 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25455 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:4748738 (4.5 MiB)  TX bytes:6618963 (6.3 MiB)
          Interrupt:193

eth1      Link encap:Ethernet  HWaddr 00:11:32:8E:5D:2B
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:13728 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11035 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:2032218 (1.9 MiB)  TX bytes:1271014 (1.2 MiB)
          Interrupt:194

**eth2      Link encap:Ethernet  HWaddr 1C:BF:CE:D4:01:39
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)**

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:11386 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11386 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:3099576 (2.9 MiB)  TX bytes:3099576 (2.9 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2022-01-04:
The DS416slim has only 512MB of memory and seems to be running out of memory. You may be able to get the driver to work by reducing the number of services running.

bb-qq, 2022-02-05:
I created an experimental driver with low memory consumption. Could you try that?
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-armada38x-2.15.0-6_rxbuf-test.spk

Steff-OG, 2022-02-05:
Great job the new driver works 

ifconfig -a
eth2      Link encap:Ethernet  HWaddr 1C:BF:CE:D4:01:39
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

dmesg
[2166768.257149] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
[2166768.277218] usb 4-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[2166768.437199] usb 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
[2166768.457238] usb 4-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[2166768.466579] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9d9ff300
[2166768.475295] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9d9ff32c
[2166768.483976] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9d9ff358
[2166783.557452] netif_napi_add() called with weight 256 on device eth%d
[2166783.597210] r8152 4-1:1.0 eth2: v2.15.0 (2021/04/15)
[2166783.602455] r8152 4-1:1.0 eth2: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.

[2166783.619131] r8152 4-1:1.0 eth2: chip rev 14
[2166783.623550] r8152 4-1:1.0 eth2: rx_buf_sz will be overridden to 8192
[2166936.249477] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[2166936.495493] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[2166936.502766] r8152 4-1:1.0 eth2: carrier on

free -m
              total        used        free      shared  buff/cache   available
Mem:            500         281          33          46         185         141
Swap:          2047         308        1738

bb-qq, 2022-03-06:
I included this change in the latest releases.

closed 2022-03-06 by bb-qq