bb-qq/r8152 #176: Failed to enable MTU 15K on RTL8156

Ngakt, 2021-12-27:
I have updated the driver to newest version 2.15.0.1 but I failed to enable MTU over 9000 anyway. The option of MTU over 9K doesn't show up.
[image]

> 蜗牛星际 C双 J1900@2.0GHz
> Linux GnyuehBase 4.4.59+ #23824 SMP PREEMPT Tue Feb 12 16:52:55 CST 2019 x86_64 GNU/Linux synology_apollolake_918+
>  DSM 6.2.1-23824 Update 6
> type C 8156 dongle with no label
> using a hub(check lsusb result for detail)
> - PC: doesn't matter
> - Hub: model name (check lsusb result for detail)
> - cable category (Cat.5e or Cat.6 or Cat.7):doesn't matter

### Output of ``lsusb`` command
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-1         05e3:0610:9224 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.0 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
  |__1-2         05e3:0608:8536 09  2.00  480MBit/s 100mA 1IF  ( ffffffd6ffffffa3ffffffebffffffc1) hub
    |__1-2.4     0951:1665:0100 00  2.00  480MBit/s 200mA 1IF  (Kingston DataTraveler 2.0 50E5495137E5BD6189B00153)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-1         05e3:0612:9224 09  3.00 5000MBit/s 0mA 1IF  (GenesysLogic USB3.0 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
    |__2-1.1     0bda:8156:3100 00  3.20 5000MBit/s 304mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
    |__2-1.3     1058:25ee:4004 00  3.00 5000MBit/s 8mA 1IF  (Western Digital My Book 25EE 3259484C4D4E3044)

### Output of ``ifconfig -a`` command

docker0   Link encap:Ethernet  HWaddr 02:42:B9:6E:28:AD  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:b9ff:fe6e:28ad/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

docker570 Link encap:Ethernet  HWaddr B6:6B:2D:04:56:8F  
          inet6 addr: fe80::b46b:2dff:fe04:568f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:1086 (1.0 KiB)

dockerd25 Link encap:Ethernet  HWaddr 8E:83:57:79:FF:0C  
          inet6 addr: fe80::8c83:57ff:fe79:ff0c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:738 (738.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:12:34:56  
          inet addr:192.168.50.100  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe12:3456/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:85209 errors:2 dropped:0 overruns:0 frame:2
          TX packets:71502 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:15465395 (14.7 MiB)  TX bytes:78335631 (74.7 MiB)
memory 0xd0700000-d071ffff  

eth1      Link encap:Ethernet  HWaddr 00:E0:67:0E:E2:8F  
          inet addr:192.168.50.29  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:67ff:fe0e:e28f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:381736 errors:0 dropped:0 overruns:0 frame:0
          TX packets:55985 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:469884942 (448.1 MiB)  TX bytes:4009484 (3.8 MiB)
memory 0xd0600000-d061ffff  

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:00:24  
          inet addr:192.168.212.2  Bcast:192.168.212.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:24/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:5495467 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2323432 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8112797827 (7.5 GiB)  TX bytes:2984140813 (2.7 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:572834 errors:0 dropped:0 overruns:0 frame:0
          TX packets:572834 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:13357051950 (12.4 GiB)  TX bytes:13357051950 (12.4 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1

Ngakt, 2021-12-27:
I checked the code of the driver and it seems for both R8156(RTL_VER_10) and R8156b(RTL_VER_12 13 15), option of MTU over 9k is available. So I suppose my 2.5GbE NIC support MTU over 9k.

ethtool

gnyueh@GnyuehBase:/$ ethtool --driver eth2
driver: r8152
version: v2.15.0 (2021/04/15)
firmware-version: 
bus-info: usb-0000:00:14.0-1.1
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

bb-qq, 2022-01-04:
This is probably a limitation on the DSM's UI. Could you configure that on the previous releases?

But I think setting MTU to over 9k is meaningless from a performance improvement point of view.

Ngakt, 2022-01-04:
Hi, 
thanks a lot to bring out limitation of DSM‘s UI!
I have set MTU to 15K through `ip link set dev eth2 mtu 15000` successfully on current release.
`eth2      Link encap:Ethernet  HWaddr [XXXXXXXXXXXXX]
    inet addr:192.168.212.2  Bcast:192.168.212.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:24/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:15000  Metric:1
          RX packets:64 errors:0 dropped:0 overruns:0 frame:0
          TX packets:352 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3344 (3.2 KiB)  TX bytes:81509 (79.5 KiB)`
it seems true that DSM's UI is limiting the MTU.

On the performance, I agree that setting MTU to 15K is not likely to bring any significant improvent compared to MTU 9K. But I think it is still worth a try since the speed of rtl8156 on my J1900 NAS is not quite stable(both single-thread iperf3 and smb3, the speed is okay at first few seconds, then drops and fluctuates), probably because of the poor single thread performance of J1900. I have tried some possible optimizations such as enabling RPS but the result is not very satisfying anyway. That's the reason why I want to set MTU larger to see If there could be any improvement.
I will update the result soon in this thread.

Ngakt, 2022-01-09:
Finally have some time to test the performance of MTU over 9K.
I will put my conclusion first: I would not suggest enabling MTU over 9K and MTU 9K seems to be a better option.
1. Details on the support of MTU over 9K: sure the new r8152 driver supports MTU over 9K and max supported MTU is 16362 on my 2.5G USB NIC Dongle, which suggests this NIC is probably RTL8156B (For RTL8156 the max supported MTU is 15K according to the driver source code). But the **Synology GUI doesn't support MTU over 9K**. Though it is possible to set MTU over 9K through SSH `ip link set dev eth2 mtu 16362`, **the system will set the MTU back to the value shown in the GUI** somtimes probably because of the conflict.
(By the way, does anybody have any idea on why there are so many similar MTU value such as 9000(max for i211 on synology) and 9014(max for i211 on windows), 16128(max for r8125 on windows) and 16362(max for r8156b on windows) ?)
2. The result shows enabling MTU over 9K is not significantly beneficial but can be harmful to RX performance due to unknown reason. 
Test Environment: 
	NAS: As described above 
	Computer:3950X + X570 board with on board RTL8125B, driver has updated to the newest(10.51.811.2021), set to MTU 16128 throughout the test for simplicity 
	Connection: Direct Connection between RTL8156B and RTL8125B, Xiaomi flat Cat.6 cable
Here I will give the iperf3 result of MTU 1500 9000 15000 and 16128(largest supported by RTL8125B, 16362 will trigger error on RTL8125B) for example. iperf3 server is set on the nas (to put more load on the nas side which is a realistic scenario(like an smb3 server))
MTU 1500 RX 2.33Gbps TX 2.35Gbps
[image]
MTU 9000 RX 2.41Gbps TX 2.46Gbps
[image]
MTU 15000 RX 2.36Gbps TX 2.47Gbps
[image]
MTU 16128 RX 2.36Gbps TX 2.47Gbps
[image]

When MTU over 9K is enabled there is a drop in RX speed for unknown reason(probably because of the cpu and usb controller). 

So sure enabling MTU over 9K is meaningless considering the iperf3 performance (at least on my J1900 )and the conflict with Synology system.

I will update some results of realworld workload such as smb3 soon. The overall impression is USB NICs are not as stable as PCIe NICs. Use a 2.5G PCIe NIC if possible.

bb-qq, 2022-01-23:
I close this issue since this was DSM UI issue and there's nothing I can do.
Thank you for all your experiments!

closed 2022-01-23 by bb-qq