bb-qq/r8152 #177: SAMZHE type-c to 2.5Gbps RTL8156B The negotiation speed is only 1Gbps

Audi-dask, 2022-01-02:
### Description of the problem

> SAMZHE type-c to 2.5Gbps RTL8156B The negotiation speed is only 1Gbps

### Description of your products

>     DS220+
>     Linux Diskstation 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_220+
>     DSM version 7.0.1 42218
>     SAMZHE type-c to 2.5Gbps RTL8156B 

### Description of your environment

> - connect directly to a PC or using a hub
>     - MAC 2.5Gbps to NAS 2.5Gbps
>     - cable category ( Cat.6 )

### Output of ``dmesg`` command

````
[  656.831251] r8152: module verification failed: signature and/or required key missing - tainting kernel
[  656.842676] usbcore: registered new interface driver r8152
[  656.972997] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  657.113907] usb 1-1: not running at top speed
[  657.488989] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  657.630944] usb 1-1: not running at top speed
[  658.005069] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  658.147314] usb 1-1: not running at top speed
[  658.521079] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  658.662181] usb 1-1: not running at top speed
[  659.036136] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  659.177405] usb 1-1: not running at top speed
[  659.551265] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  659.693438] usb 1-1: not running at top speed
[  670.277184] netif_napi_add() called with weight 256 on device eth%d
[  670.284509] r8152 1-1:1.0 eth2: v2.15.0 (2021/04/15)
[  670.290066] r8152 1-1:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[  670.307475] r8152 1-1:1.0 eth2: chip rev 14
[  671.908970] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  672.812549] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[  672.820451] r8152 1-1:1.0 eth2: carrier on
[  675.464481] audit_printk_skb: 18 callbacks suppressed
[  675.470152] audit: type=1325 audit(1641106027.066:103): table=filter family=2 entries=35
````

### Output of ``lsusb`` command

````
root@Diskstation:~# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         0bda:8156:3100 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
````

### Output of ``ifconfig -a`` command

````
root@Diskstation:~# ifconfig eth2 
eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:01:C1  
          inet addr:192.168.1.106  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:1c1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1863 errors:0 dropped:0 overruns:0 frame:0
          TX packets:134 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:277751 (271.2 KiB)  TX bytes:21222 (20.7 KiB)

root@Diskstation:~# ethtool eth2
Settings for eth2:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseX/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
                                             1000baseT/Full 
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes
````

[image]

Sjekke, 2022-01-02:
Are using the latest version of the driver (released in september)?

bb-qq, 2022-01-04:
```
  |__1-1         0bda:8156:3100 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
```

This line indicates that the dongle is recognized as a USB 2.0 device. So please check the USB cable. (FYI; some cheap type-c cables support USB3.0 signal on one side only)

Audi-dask, 2022-01-07:
> Are using the latest version of the driver (released in september)?

OK Is the latest

Audi-dask, 2022-01-07:
> ```
>   |__1-1         0bda:8156:3100 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
> ```
> 
> This line indicates that the dongle is recognized as a USB 2.0 device. So please check the USB cable. (FYI; some cheap type-c cables support USB3.0 signal on one side only)

Thank you. I'm just checking

closed 2022-01-07 by Audi-dask