bb-qq/r8152 #179: llano 2.5Gbps USB Ethernet Adapter can't start on Synology 918+

morongenius, 2022-01-07:
I have this llano adapter connected to the front usb port of 918+. It works with 2.5Gbps link speed but does not starts every time the 918+ restarts (it's on daily power on/off schedule), have to manually stop and run the driver in package centre. The installed driver is 2.15.0-1 and 918+ is on DSM 6.2.4. Pls help. Thanks.

bb-qq, 2022-01-09:
Could you try to replace the startup script with to latest one?

target script: `/var/packages/r8152/scripts/start-stop-status`
latest file: https://github.com/bb-qq/r8152/blob/master/scripts/start-stop-status

morongenius, 2022-01-10:
apologies but i do not have the skills to do that.

bb-qq, 2022-01-10:
Try this command:
`curl -o /var/packages/r8152/scripts/start-stop-status https://raw.githubusercontent.com/bb-qq/r8152/master/scripts/start-stop-status`

morongenius, 2022-01-10:
sorry all these command line is beyond me. not skilled in IT, any other way? thanks.

bb-qq, 2022-01-11:
Hmm, the easiest way is that update DSM to 7 and use the latest driver...

morongenius, 2022-01-12:
i have updated to DSM 7 but failed to run and i still need to do the command line thing... how to do the ssh thing? would appreciate any step by step instructions. thanks a million!

morongenius, 2022-01-12:
I have updated to DSM7 and successfully installed driver 2.15.0-4 but the issue is still not resolved, so I tried the command above but got an error "-sh: sodo: command not found"....

Sjekke, 2022-01-12:
It's SUDO ...

morongenius, 2022-01-12:
Thanks all for your help. successfully replaced startup script and now the adapter is active upon NAS restart. Will monitor for a few days and hopefully the issue is resolved. thanks again

bb-qq, 2022-01-12:
Oh, sorry, that is a typo. I fixed the comment.

Also, the latest version of the driver is 2.15.0-*5* that includes the latest startup script that you downloaded by the curl command. You don't need such a command in the future release.

morongenius, 2022-01-21:
So far the adapter has been working well now so issue is resolved. Thanks for the help!

closed 2022-01-21 by morongenius